linux-surface/surface-aggregator-module #2: Surface Pro 2017 Module loaded but can't find anything related battery.

pijulius, 2018-11-13:
Hi qzed,

I just compiled your module with the 4.18.11 kernel (and of course all patches applied) and I can load the module just fine just as far as I can see there is nothing this module "does".

Checking dmesg for surface this is what I get:
[    4.075932] surface_acpi: Microsoft Surface ACPI Driver version 0.1
[    4.075980] surface_acpi: Attaching device MSHW0084
[    4.076077] surface_acpi: Attaching device MSHW0091
[    4.077009] surface_acpi: probe of INT3403:03 failed with error -22
[    4.077013] surface_acpi: probe of INT3403:04 failed with error -22
[    4.077021] surface_acpi: probe of INT3403:08 failed with error -22
[    4.101880] surfacegen5_acpi_notify: loading out-of-tree module taints kernel.
[    4.206001] surfacegen5_acpi_notify_ssh serial0-0: surfacegen5_acpi_notify_ssh_probe
[    4.206340] surfacegen5_acpi_notify_ssh serial0-0: surfacegen5_ssh_setup_from_resource

Can you please let me know what I do wrong or if Surface Pro 2017 is not supported and that's why I don't see anything?

Thanks in advance!
Julius

qzed, 2018-11-13:
This line
```
[ 4.075980] surface_acpi: Attaching device MSHW0091
````
indicates that you have Jake's ACPI patches applied. This conflicts with the module (2 drivers wanting to load against the same device), so you have two options: Manually unload/blacklist the `surface_acpi` module (for other readers: I'm not sure if this is a good idea on the Surface Laptop, I'm not informed on what of the relevant stuff is already integrated there) or build the kernel again without the ACPI patches ([e.g. from this fork](https://github.com/qzed/linux-surface)). (Or of course wait until it's in Jake's kernel, which hopefully happens soon, once the final major issues are addressed :-) ).

pijulius, 2018-11-13:
Thanks for writing me back so quickly, just for testing I did blacklist the surface_acpi module and now it seems yours is loading just fine:

[    4.455098] surfacegen5_acpi_notify: loading out-of-tree module taints kernel.
[    4.546343] surfacegen5_acpi_notify_ssh serial0-0: surfacegen5_acpi_notify_ssh_probe
[    4.672128] surfacegen5_acpi_notify_ssh serial0-0: surfacegen5_ssh_setup_from_resource

but for some reason still nada :)

In any case nw, just thought give it a try, will cross my fingers for Jake's updates, hopefully he will find a way to implement the missing parts, especially the wifi drop out problem is pretty crazy as it forces me to reboot 5-10 times a day but the rest is perfectly working so far.

qzed, 2018-11-13:
Hmm, with the other module blacklisted it should work, can you see any `surfacegen5_acpi_notify_san_probe` message in the logs? If this is missing it indicates that the driver for the actual "Surface ACPI Notify" platform device (`_SAN`) is not loading. According to [this comment](https://github.com/jakeday/linux-surface/issues/28#issuecomment-436645069) you might also need to blacklist/unload the `surface_platform` module. I haven't tried it with Jake's ACPI patches myself, so sorry for the not-so-precise answers in this regard.

With regards to the wifi issue (although this is not the right issue here): I've only ever encountered that when I was playing around with tlp and powertop. Also make sure you've copied all the relevant config files, if I remember correctly there was one addressing some power-management issues with the wifi controller.

calvous, 2018-11-14:
@pijulius @qzed 
I modified the sleep script which jakeday provided like this
```
cat: /usr/lib/systemd/system-sleep: Is a directory
#!/bin/sh
# unload/load kernel modules sleep/resume
# depending on the kernel version we need different modules
if [ `uname -r | grep surface` ]
then
    modules="intel_ipts mei_me mwifiex_pcie surfacegen5_acpi_notify"
else
    modules="mwifiex_pcie surfacegen5_acpi_notify"
fi
case $1 in
  pre)
    # logger "[user script] Entering $1 mode..."
    # unload the modules before going to sleep
    # logger "[user script] turning wifi & network off..."
    /usr/bin/nmcli radio wifi off
    /usr/bin/nmcli network off
    # logger "[user script] shutting down NetworkManager..."
    systemctl stop NetworkManager.service
    for mod in $modules
    do
        logger "[user script] unloading $mod..."
        /usr/bin/modprobe -v -r $mod
    done
    ;;
  post)
    # need to cycle the modules on a resume and after the reset is called, so unload...
    # logger "[user script] Entering $1 mode..."
    # for mod in $modules
    # do
    #     # logger "[user script] unloading $mod..."
    #     /usr/bin/modprobe -v -r $mod
    # done
    # and reload
    for mod in $modules
    do
        logger "[user script] loading $mod..."
        /usr/bin/modprobe -v -i $mod
    done
    # echo 1 > /sys/bus/pci/rescan
    # systemctl daemon-reload
    # logger "[user script] starting NetworkManager..."
    systemctl start NetworkManager.service
    # logger "[user script] turning wifi & network on..."
    /usr/bin/nmcli network on
    /usr/bin/nmcli radio wifi on
    ;;
esac
```
It might need some adjustment for different setups. But this way I have no problems with waking up after suspend. The "surfacegen5_acpi_notify" module must be unloaded before sleep. Otherwise it will take up to 3 minutes before the computer gets responsive. Wakeup is not possible with the keyboard. Only by pushing the ArchMouse or pressing the power button (surfacegen5_button module needed). I advice to blacklist the "surfacepro3_button" module. I could be helpful to uncomment this lines:
```
    # echo 1 > /sys/bus/pci/rescan
    # systemctl daemon-reload
```
but on my system (Surface Pro 1796) there is no need for this.

calvous, 2018-11-14:
I just found out of what I wrote above. I also catch the "communication timed out" and the corresponding "trying again" message when the system goes to suspend mode. Not always but most of the time. It is only one message, not three of them. The system can only be awakened by pushing the mouse or depressing the power button. The latter works only with your button module.

pijulius, 2018-11-14:
Indeed blacklisting surface_platform and surface_acpi made it work right away! Thanks again for your quick help and the work you did on this Qzed!

Will close this case now as everything is working perfectly now.

closed 2018-11-14 by pijulius

pijulius, 2018-11-15:
Hi Calvous,

Can confirm that unloading and reloading surfacegen5_acpi_notify fixes the sleep command, it now goes to sleep within 5 sec and starts around 15 and everything works perfectly so far.

I haven't used your sleep code from above but just simply added modprobe -r and -i for surfacegen5_acpi_notify to the existing sleep code and that did it.

Also I now rebuild the kernel 4.18.11 using qzed's linux-surface repo and can confirm everything built just fine and no need for any blacklist module or any other change except the sleep code update above and everything works just perfectly fine, will keep an eye on wifi too but all good for the moment.

Thanks again guys for all your work/help on this!
Julius