linux-surface/surface-aggregator-module #3: Hello qzed i'm wondering if it's possible to port this to 4.9 kernel

thamerx, 2018-11-28:
Hello,

 I have 4.9 kernel that works really well on android x86 ([here](https://github.com/BlissRoms-x86/kernel_common/tree/kernel-4.9-IPTS)) (I'm not affiliated with TeamBilss)
 I'm just wondering if it possible to port acpi_notify to 4.9. I was able to run get the acpi_notify on androidx86 4.18.4 but without touch, screen rotation or vol-pwr buttons.

Sorry for interrupting

-Thamer

qzed, 2018-11-28:
> Sorry for interrupting

Don't be, issues, questions and the like are always welcome!

Unfortunately the module requires the `serdev` subsystem, which is only available on 4.12 upwards (some ACPI related additions that might also be needed are only available from 4.15 upwards). So back-porting this module would require back-porting `serdev`. I don't know how different the android kernel is, but maybe you have better luck applying the relevant patches (buttons, IPTS) from Jake's repo.

thamerx, 2018-11-28:
if 4.9 is quite hard to backport to then how about 4.16 check this one [here](https://github.com/BlissRoms-x86/kernel_common/tree/k4.16.14-bliss-ipts-ga) is it possible to backport to this kernel? cause this kernel works great as well 

I did try to apply jakeday patches on 4.18.4 but they didn't work for some reason

qzed, 2018-11-28:
4.16 should be possible, I'll try to have a look at it on the weekend.

thamerx, 2018-11-28:
I'll be thankful

qzed, 2018-11-30:
Do you still need the patches? If so, let's keep this issue open.

thamerx, 2018-12-03:
The reason why I said 4.9 rather than 4.16 is that 4.9 auto-rotation works and the accelerometer works great on portrait and landscape modes while 4.16 does not autorotate for some reason and the accelerometer gives incorrect results if the display was set to portrait mode

thamerx, 2018-12-03:
I tried to copy /drivers/iio/orientation/ (hid-sensor-incl-3d.c+hid-sensor-rotation.c) drivers from 4.9 to 4.16 and removing ".driver_module = THIS_MODULE," and compile the kernel
I did build correctly but no luck getting the rotation sensor working 

I have so little knowledge in kernels

qzed, 2018-12-03:
I think you might have better luck by comparing the kernel you're using against the stock 4.16 kernel and check if there are any (significant) changes. I can't vouch for 4.16 specifically (on.y 4.18), but I can't remember any auto-rotation problems.

qzed, 2018-12-03:
Ok, so I haven't tried to build the module on 4.16 yet, but the kernel you linked to already contains Jake's patches and the patches required for the module apply without error. The thing is, with Jake's ACPI patch applied, the drivers provided in this module won't load. The easiest way to fix this for you is to just blacklist the `surface_acpi` module.

thamerx, 2018-12-08:
sry I no longer want to deal with kernels since I lack so much knowledge and I don't have much time to spare and learn

closed 2018-12-08 by thamerx

qzed, 2018-12-08:
Nothing to be sorry for, I completely understand. In case you ever want to have a look at this again, I think everything should work on 4.16, but I haven't tried to build the module on that version yet. If you, or someone else for that matter, encounter any issues with compiling, just send me a log and I'll look into it.