linux-surface/surface-aggregator-module #5: [SB2 15] Module loading, but no batteries appear

xanf, 2018-12-07:
Hi! I'm running SurfaceBook2 15" and ecently built 4.18.20 kernel (on Archlinux) using your patchset from qzed/linux-surface repo and your config (you've shared it somewhere). Unfortunately, no battery appears. Module loads automatically, previous version spammed with:
```
[    6.883302] surfacegen5_acpi_notify_san MSHW0091:00: surfacegen5_acpi_notify_san_probe
```
for like 40 times :)
(no other lines with surfacegen in dmesg)
New version keeps silence at all.

Module is loaded according to lsmod

xanf, 2018-12-07:
I do not have jakeday things involved:
```
# lsmod | grep surface
surfacegen5_acpi_notify    24576  0
```

qzed, 2018-12-07:
Seems like the serdev device for the UART isn't being created. Can you check that with
```
ls /sys/bus/acpi/devices/MSHW0084:00/physical_node/subsystem/devices/
```
?

xanf, 2018-12-07:
No such file or directory

qzed, 2018-12-07:
Ok, then what about
```
/sys/bus/pci/devices/0000:00:1e.0/dw-apb-uart.4/
```
?

xanf, 2018-12-07:
```
#LC_ALL=C ls -la /sys/bus/pci/devices/0000:00:1e.0/dw-apb-uart.4/

total 0
drwxr-xr-x 4 root root    0 Dec  7 15:36 .
drwxr-xr-x 5 root root    0 Dec  7  2018 ..
lrwxrwxrwx 1 root root    0 Dec  7 15:36 driver -> ../../../../bus/platform/drivers/dw-apb-uart
-rw-r--r-- 1 root root 4096 Dec  7 17:06 driver_override
lrwxrwxrwx 1 root root    0 Dec  7 17:06 firmware_node -> ../../../LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:76
-r--r--r-- 1 root root 4096 Dec  7 17:06 modalias
drwxr-xr-x 2 root root    0 Dec  7 17:06 power
lrwxrwxrwx 1 root root    0 Dec  7 15:36 subsystem -> ../../../../bus/platform
drwxr-xr-x 3 root root    0 Dec  7 15:36 tty
-rw-r--r-- 1 root root 4096 Dec  7 15:36 uevent
```

qzed, 2018-12-07:
Ok, `tty` there indicates that the serdev subsystem does not take over the UART device (if it works you should have something like `serial0` there). Usually this is a problem with the kernel config or the patches, but since you said you applied them I'm a bit confused here. Is there anything regarding `serdev` in the log? And altough it's probably not going to help us: what about `tty`?

xanf, 2018-12-07:
`dmesg | grep serdev` empty
The only thing about tty:
`[    4.896453] dw-apb-uart.4: ttyS4 at MMIO 0xd4748000 (irq = 20, base_baud = 3000000) is a 16550A`

xanf, 2018-12-07:
Just to make sure - here is the config I've used? Maybe I'm using something wrong?
[config.txt](https://github.com/qzed/linux-surfacegen5-acpi-notify/files/2658194/config.txt)

qzed, 2018-12-07:
Hmm, the message with `tty` is another indicator of what we already know: Instead of the serdev subsystem, the tty system takes care of the UART device.

From what I can see in the config file, this is for version `4.18.16`. How did you configure the kernel? What I mean is: Did you run `make olddefconfig` with the that config file as `.config`? (I had to upload the config as `config.txt` as GitHub did not accept the `.config` filename, maybe that's caused some confusion.)

xanf, 2018-12-07:
yes, I've done make olddefconfig.
this file is named `config` in my filder
from my PKGBUILD (build system for Arch):
```
  msg2 "Setting config..."
  cp ../config .config
  make olddefconfig

```

qzed, 2018-12-07:
Okay, that looks like it should work then. To completely rule configuration out: can you upload the output of `zcat /proc/config.gz`?

qzed, 2018-12-07:
If that doesn't work we might have to add some debug output for the serdev core, I think we should start
[here](https://elixir.bootlin.com/linux/v4.18.20/source/drivers/tty/tty_port.c#L160), just to make sure the entry to the serdev system gets called at all. Just add the line
```c
printk(KERN_ERR, "sg5_debug: tty_port_register_device_attr_serdev\n");
```
(or something like that, so that we can later easily grep for this). Ultimately, we want to make sure that [this](https://elixir.bootlin.com/linux/v4.18.20/source/drivers/tty/serdev/serdev-ttyport.c#L291) call succeeds, so also add a line directly after that call:
```c
printk(KERN_ERR, "sg5_debug: serdev_controller_add returned %d\n", ret);
```

xanf, 2018-12-07:
> Okay, that looks like it should work then. To completely rule configuration out: can you upload the output of `zcat /proc/config.gz`?

Sorry, does not have relevant config options to dump /proc/config.gz in your config :)
Will add debugs and let you know

qzed, 2018-12-07:
Ah, sorry, okay.

xanf, 2018-12-08:
Absolute silence in dmesg :(
I've thought I maybe totally wrong with compiling my kernel, so I've done following to check that my debug printk got into kernel and they are there:
```
src/linux-4.18.20/scripts # ./extract-vmlinux /boot/vmlinuz-linux-surface-qzed | strings | grep sg5
sg5_debug: tty_port_register_device_attr_serdev
sg5_debug: serdev_controller_add returned %d
```

Of course I've rebooted laptop after installing new kernel :)

xanf, 2018-12-08:
I've rebuilt kernel, so can now share /proc/config.gz and dmesg (maybe will give an idea) with you
[config.txt](https://github.com/qzed/linux-surfacegen5-acpi-notify/files/2659726/config.txt)
[dmesg.txt](https://github.com/qzed/linux-surfacegen5-acpi-notify/files/2659727/dmesg.txt)

xanf, 2018-12-08:
Sorry, it seems that my patches were fetched and applied in unexpected way. Everything seems working right now, I'm sorry for wrong bur reports

closed 2018-12-08 by xanf

qzed, 2018-12-08:
That would explain it, nice that you got it working now, no need to be sorry!