linux-surface/surface-aggregator-module #8: [SL] Caps lock key doesn't illuminate on press

zlrenner, 2018-12-16:
From https://github.com/jakeday/linux-surface/issues/5:
The capslock key does not illuminate on keypress.
>From the HID spec:
>Synchronization between LED states and CAPS LOCK, NUM LOCK, SCROLL LOCK, COMPOSE, and KANA >events is maintained by the host and NOT the keyboard. If using the keyboard descriptor in Appendix B, >LED states are set by sending a 5-bit absolute report to the keyboard via a Set_Report(Output) request.

@qzed I left the laptop at work over the weekend so I unfortunately can't get you the logs quite yet. I can do so early in the week.

qzed, 2018-12-16:
@zlrenner Thanks!

> I unfortunately can't get you the logs quite yet. I can do so early in the week.

No pressure, it may take some time until I can have a look at this issue, so there's no need to hurry.

qzed, 2019-03-09:
@zlrenner I've added some instructions on how to get the required log files [here](https://github.com/qzed/linux-surfacegen5-acpi/blob/master/README.md#getting-windows-logs-for-reverse-engineering). Can you upload a log from a few (two or so) caps-lock presses?

dsi3, 2019-07-20:
I have a Surface Laptop 2 with Windows 10 Pro still installed. Do you still need these log files? I would love to help. I tried putting Ubuntu on my device a few months ago and I found the only thing stopping me from keeping it installed was the touchpad 'tap to click' problem, where tapping only seems to work for a few minutes after logging in. It'd be great to have the caps lock light working too, of course!

qzed, 2019-07-20:
@dsi3 Still need the files. I don't have any clue what's causing the touchpad issue though.

dsi3, 2019-07-21:
@qzed I followed your steps very closely but didn't get any messages displaying in DebugView view in the last step while I was pressing the caps lock key (or any key or action). I'm not very familiar with this kind of work in general so I'm not sure what I might've done wrong. More than happy to have another try.

qzed, 2019-07-21:
@dsi3 Getting no output might mean something like the component filter mask not set properly. Can you try to set this after step 4 (e.g. using the [WinDbg preview](https://www.microsoft.com/en-us/p/windbg-preview/9pgjgd53tn86?activetab=pivot:overviewtab), second bullet point [here](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/reading-and-filtering-debugging-messages#setting-the-component-filter-mask)). You need to start a local debugging session for this (in WinDbg Preview: File -> Start Debugging -> Attach to kernel -> Local).

Also make sure you run everything as Administrator, specifically `Dbgview`.

dsi3, 2019-07-26:
Planning on trying again this weekend.

qzed, 2019-07-26:
@dsi3 Awesome, thanks! Let me know if you have any questions/problems.

dsi3, 2019-07-27:
@qzed Had another go and still not getting output.

The specific steps I performed were:
1. Download the required software to `C:\Users\<username>\Downloads` (also where I ran the two programs from).
2. Download and install WinDbg from the Microsoft Store.
3. Restart and disable secure boot from the bootloader menu (had to do this in order to successfully enable kernel debugging).
4. Disable driver signature verification (after restarting by holding shift and clicking the restart button, pressing `7`, etc.).
5. Start Powershell as administrator and enter `bcdedit /debug on` to enable kernel debugging. Close window.
5. Start `WinDbg` as administrator, start a local debugging session, and enter the command `>lkd ed Kd_DEFAULT_Mask 0xffffffff`. Leave window open.
6. Start `IRPMon` as administrator, right-click `iaLPSS2_UART2` and select `Hooked`, then check `Capture Events`. Leave window open.
7. Start `Dbgview.exe` as administrator, add `HOOK_DATA` next to `Include` under `Filter/Highlight...`, and ensure `Capture Kernel` is checked. Leave window open.
8. Turn caps lock on and off; no output in Dbgview.

qzed, 2019-07-27:
@dsi3 The steps look correct, however I had to restart after enabling kernel debugging. If I remember correctly, you should get an error in WinDbg though if kernel debugging is disabled.

I assume `iaLPSS2_UART2` is the only UART device? If not maybe it's named differently (you could just hook all UART devices in that case). You can also try to invoke another action that should trigger some communication, for example move the performance slider (opens when you click on the battery icon) or suspending and resuming.

Also are there any errors or warnings during those steps?

dsi3, 2019-07-28:
@qzed Unfortunately still now luck. I tried again and restarted _after_ enabling kernel debugging, but still didn't receive any output. I don't recall getting that error in WinDbg at any point during my attempts.

Yes, it's the only UART device in the list that I can see. I tried hooking the parent option as well as the child, with the same results. Pressing the power button to suspend the laptop caused a system crash during the tests.

No errors, aside from the system crash.

qzed, 2019-07-28:
@dsi3 Do you also experience a crash when you change the battery/performance slider? I've tried this today and I also get a system crash when I try to log something (happens as soon as a message would normally appear, e.g. when I change the slider). I hoped this would be due to the Windows Insider preview version I have installed, but if you get the crash too it looks like we need to re-compile the kernel module part. Unfortunately I don't have much experience in this, so it might take me a while.

Interesting that it doesn't crash on the keyboard though. If I'm right, this would indicate that the keyboard doesn't send anything, so I'm wondering if it has a secondary interface.

dsi3, 2019-07-30:
@ Tried again this morning, using the same steps as described previously,
and adjusting the battery/performance slider did not cause a crash, nor did
it produce any log results. Pressing the power button to suspend the laptop
did, however, cause a system crash (same as last time).

I don't have the insiders build installed, although I did install it a few
months ago to try out WSL 2.

Let me know what other tests I can perform on the SL2. I don't have any
plans to take Windows off it just yet.

qzed, 2019-07-30:
@dsi3 Thanks, I'll let you know when I have something that needs testing. I'll also try to figure out why the logging is not working, but that may take me a while.

I've also just noticed that IRPMon has a [new beta release](https://github.com/MartinDrab/IRPMon/releases/tag/v0.9-beta) which now officially supports displaying data, so if you want to you could have a look at that. I haven't tried that yet though, but it looks like you don't need to set the Kd_DEFAULT_Mask, kernel debugging mode, and DbgViewer any more:

> Log data associated with the requests
> - IRPs only
> - read, write, ioctl, pnp, query/set information...
> - turned off by default since it may be dangerous (turn it on by checking the Data item in the context menu)
> - displayed in the Request Details* form

dsi3, 2019-08-04:
@qzed I'm really sorry, I just realised I should've tested this before I installed Linux Mint today.

qzed, 2019-08-04:
@dsi3 No worries!

archseer, 2019-11-14:
This now works on SL3. If there's any SL1/2 users interested, we can provide instructions on how to build and enable the new VHF module to test if it works -- it also has general improvements so we'd like to retire the old module if possible. Just need someone to confirm it works first.

qzed, 2019-12-02:
As @archseer mentioned, there's something new to test: Can anyone with a SL1 or SL2 try and run this script: https://github.com/qzed/linux-surfacegen5-acpi/blob/tmp/laptop-keyboard/scripts/get_sl1_hiddesc.py and upload the output?

OanaIgnat, 2019-12-15:
:hand: SL1 here. The output I've got is the following:

```
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 0, 'pri': 2, 'bufid': 0}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 0, 'pri': 2, 'bufid': 1}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 0, 'pri': 2, 'bufid': 2}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 0, 'pri': 2, 'bufid': 3}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 1, 'pri': 2, 'bufid': 0}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 1, 'pri': 2, 'bufid': 1}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 1, 'pri': 2, 'bufid': 2}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 1, 'pri': 2, 'bufid': 3}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 2, 'pri': 2, 'bufid': 0}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 2, 'pri': 2, 'bufid': 1}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 2, 'pri': 2, 'bufid': 2}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 2, 'pri': 2, 'bufid': 3}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 3, 'pri': 2, 'bufid': 0}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 3, 'pri': 2, 'bufid': 1}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 3, 'pri': 2, 'bufid': 2}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 3, 'pri': 2, 'bufid': 3}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 4, 'pri': 2, 'bufid': 0}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 4, 'pri': 2, 'bufid': 1}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 4, 'pri': 2, 'bufid': 2}
Failed to execute request: {'tc': 8, 'cid': 4, 'iid': 4, 'pri': 2, 'bufid': 3}
[]
```

dsi3, 2019-12-15:
I no longer own a Surface so can't do any of these tests anymore. Good luck to everyone.

archseer, 2019-12-15:
Recently, I've found a way to extract the full HID descriptor from any device on Windows 10, but it's a bit involved. If the script doesn't work for these older devices, maybe we can just dump the whole HID descriptor, which should fix the caps lock, as well as maybe fix #4 .

You need to install the [Windows Driver Kit](https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk), then go to `C:\Program Files (x86)\Windows Kits\10\bin\x64` or something like that, and `windbg.exe` right click -> **run as administrator**.

Then:
- Run `.load hidkd.dll` to load the HID extensions.
- Run [`!hidtree`](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-hidkd-hidtree) to list all the devices.

We want to do the next steps for the (first few) entries, which are Mouse, Touchpad, Consumer Control, etc., 3~4 entries.

- Run [`!hidfdo <id here>`](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-hidkd-hidfdo) with the id specified in the `hidtree` (you can just click on the `hidfdo` links in the output).
- Run [`!hidrd <id> <number>`](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-hidkd-hidrd), as listed in the `Report Descriptor:` line. Copy and gist all the data returned (we want the raw data, but parsed data is also useful).

qzed, 2019-12-15:
@OanaIgnat Thanks! Just to be sure, the device `/sys/bus/serial/devices/serial0-0/rqst` exists? If that's the case then it looks like it's not using the same EC/communication interface as on the SL3.

In that case, this is probably going to take a bit of trial-and-error to figure out. If you want to help out/have the time, you could try running the script again after setting `"pri"` to `1´ [here](https://github.com/qzed/linux-surfacegen5-acpi/blob/tmp/laptop-keyboard/scripts/get_sl1_hiddesc.py#L61-L80) (although I doubt that's going to change much). I've also uploaded another script (https://github.com/qzed/linux-surfacegen5-acpi/blob/tmp/laptop-keyboard/scripts/sysfs_rqst_simple.py), which you can run via
```
./sysfs_rqst_simple.py <TC> <CID> <IID> <PRI> <SNC>
```
Specifically the output of the following combinations would be interesting:
```
./sysfs_rqst_simple.py 0x08 0x01 0x00 0x01 0x01
./sysfs_rqst_simple.py 0x08 0x02 0x00 0x01 0x01
./sysfs_rqst_simple.py 0x08 0x03 0x00 0x01 0x01

./sysfs_rqst_simple.py 0x15 0x04 0x00 0x01 0x01
```

OanaIgnat, 2019-12-15:
No, that file doesn't exist. What should I do then?

qzed, 2019-12-16:
@OanaIgnat You'll have to install a kernel with the patches in this repo, e.g. one of https://github.com/qzed/linux-surface/releases.

OanaIgnat, 2019-12-16:
I did that already.

qzed, 2019-12-16:
What version? I think the debug interface is not enabled in older versions. But if you're running a patched kernel you should also be able to build and load the module separately. Just run `make` inside the `module` directory, unload the modules that come with the kernel, e.g. via [`scripts/unload.sh`](https://github.com/qzed/linux-surfacegen5-acpi/blob/master/scripts/unload.sh) and `sudo insmod surface_sam.ko`, again in the `module` directory (you should probably also run `sudo modprobe -r battery` to prevent the device from suspending due to the battery interface disappearing, while there is no SAM/SAN driver loaded; you can load that again afterwards if you need the battery indicator while testing). The debug interface is enabled by default in the module.

lecram1989, 2020-01-24:
> @OanaIgnat Thanks! Just to be sure, the device `/sys/bus/serial/devices/serial0-0/rqst` exists? If that's the case then it looks like it's not using the same EC/communication interface as on the [...]

I ran sysfs_rqst_simple.py with the combinations above as su. It gave me each time the following output:
`Traceback (most recent call last):
  File "./scripts/sysfs_rqst_simple.py", line 41, in <module>
    main()
  File "./scripts/sysfs_rqst_simple.py", line 29, in main
    os.write(fd, data)
OSError: [Errno 5] Input/output error
`

lecram1989, 2020-01-24:
I followed archseer's description on how to get the full HID descriptors in Win10 using `winbg.exe`. I saved the output of `!hidtree` in hidtree.txt and the output of `!hidfdo` and `!hidrd` in the respective **x**_entry.txt 
[2_entry.txt](https://github.com/linux-surface/surface-aggregator-module/files/4109909/2_entry.txt)
[3_entry.txt](https://github.com/linux-surface/surface-aggregator-module/files/4109910/3_entry.txt)
[4_entry.txt](https://github.com/linux-surface/surface-aggregator-module/files/4109911/4_entry.txt)
[5_entry.txt](https://github.com/linux-surface/surface-aggregator-module/files/4109912/5_entry.txt)
[6_entry.txt](https://github.com/linux-surface/surface-aggregator-module/files/4109913/6_entry.txt)
[7_entry.txt](https://github.com/linux-surface/surface-aggregator-module/files/4109914/7_entry.txt)
[8_entry.txt](https://github.com/linux-surface/surface-aggregator-module/files/4109915/8_entry.txt)
[hidtree.txt](https://github.com/linux-surface/surface-aggregator-module/files/4109916/hidtree.txt)
[1_entry.txt](https://github.com/linux-surface/surface-aggregator-module/files/4109917/1_entry.txt)

archseer, 2020-02-29:
Hey @lecram1989 sorry for the delay but I finally had the time to port over output reports from the new driver to the SL/SL2 one. I think with this implementation it might work correctly.

Could you test out this commit? It's on the ` feature/sl1-output-report` branch.
https://github.com/linux-surface/surface-aggregator-module/commit/51576cbe390c2fe430c0ba8e98aa3de86bf1dbf8

I think @qzed might have given you instructions on how to [manually build and load the module](https://github.com/linux-surface/surface-aggregator-module#buildtest-the-module) (note that you have to unload the ones listed under the blacklist section first), but if you have any questions feel free to ping us on IRC.

archseer, 2020-02-29:
There's also some logging included. If you raise the kernel log level to debug you should be able to get debug output as well.

archseer, 2020-03-15:
\cc @sudosubin as well since you're now on the testers list too :)

SmartBoy84, 2020-05-06:
Has this issue been fixed yet?

qzed, 2020-05-16:
@SmartBoy84 We're still waiting for someone to test an implementation before we'll merge it, see https://github.com/linux-surface/surface-aggregator-module/issues/8#issuecomment-592884060 on how to help out.

archseer, 2020-06-03:
Ping @lecram1989 @sudosubin @SmartBoy84 again, it would be really great to get someone to test. If that commit works then SL1/SL2 could use the new VHF driver as well (just with a hard-coded descriptor).

sudosubin, 2020-06-11:
I am testing on below environment, but continues to fail.
- Device: Surface Laptop 2
- OS: `Kubuntu 20.04 LTS`
- Test kernel: `feature/sl1-output-report`, with latest `master` branch merged.

```sh
# Kernel version
$ uname -r
5.6.15-surface
```

```sh
# Error message
$ sudo insmod surface_sam.ko
insmod: ERROR: could not insert module surface_sam.ko: Invalid module format
```

Maybe `linux-surface-secureboot-mok` installed make it wrong? But I'm not sure how to make the test properly.

I will search more, and try with other ways. Thanks, and sorry for not being helpful :(

qzed, 2020-06-11:
This usually indicates that the module has been built with a different kernel version. You'll need to recompile (i.e. run `make clean all` inside the `module` directory) it for the current kernel.

If that doesn't work, `dmesg` should show a more specific error message.

qzed, 2020-10-07:
Fixed and implemented in master via #51

closed 2020-10-07 by qzed