linux-surface/surface-aggregator-module #9: Surface Laptop 2: Keyboard not working

dbranham20, 2018-12-21:
So I tried jakeday's repo for fixing the keyboard and nothing came out of that. I then found this repo in one of his issues tickets and tried installing it. Everything seemed to install and setup fine, but upon reboot, my keyboard still does not work. I'm running Linux Mint 19.1 on my Surface Laptop 2 and Secure Boot is off.

With instructions, I'm willing to test if needed!

qzed, 2018-12-21:
Do you have a kernel with the patches in this repository applied (e.g. one of [those](https://github.com/qzed/linux-surface/releases) two prebuilts)? If so, can you post a dmesg log?

dbranham20, 2018-12-21:
i did not originally, but i just installed the linux headers and linux image debian packages. Here is the 
[dmesg.txt](https://github.com/qzed/linux-surfacegen5-acpi/files/2704132/dmesg.txt)

qzed, 2018-12-21:
After you've installed the new kernel, have you reinstalled the module?

qzed, 2018-12-21:
Awesome! Luckily the SL2 seems to be pretty similar to SL1, I'll let you know if I need anything.

closed 2018-12-21 by qzed