linux-surface/surface-aggregator-module #12: [SP2017] Battery Indicator

brendon-anderson, 2018-12-27:
Hello!

I'm new to Linux and I'm having troubles getting this module to work on a Surface Pro 2017 with a fresh install of Ubuntu 18.04 LTS. I've installed [this](https://github.com/qzed/linux-surface/releases/tag/v4.18.16-pre1) precompiled kernel, which successfully gave me touch/pen/volume button support. Next, I tried to verify that the following configuration values are set:

```
CONFIG_SERIAL_DEV_BUS=y

CONFIG_SERIAL_DEV_CTRL_TTYPORT=y
```

I found that this was the case for the following files:

```
/boot/config-4.15.0-29-generic

/boot/config-4.15.0-43-generic
```

Are these the correct configuration files to be checking?

I then went on to `make` the module, and installed it using `make dkms-install`. I've also enabled the Battery Percentage option in Tweaks, but nothing has shown up in my Top Bar, even after reboot. It seems like the module is successfully running, since `lsmod | grep 'surface'` shows me `surfacegen5_acpi` is running. I've tried with and without blacklisting `surface_acpi` and `surface_platform`. I'm starting to run out of ideas, so any help would be greatly appreciated!

qzed, 2018-12-27:
The config files have to match the kernel version, so in this case the relevant config file should be something like `config-4.18.16-surface+`. Also if you've installed the kernel you linked to, you don't have to check the config (that's built into the kernel package).

At the moment, your easiest option, however, is to just install [this](https://github.com/qzed/linux-surface/releases/tag/v4.18.20-pre2) kernel. This kernel includes the module, so you don't have to install it separately, it should work out of the box once you've booted into it.

qzed, 2018-12-27:
Sorry, I have not changed anything in `setup.sh`, so all links are still to Jake's repo. If you've run `setup.sh` once you're good to go, the firmware is not removed when you install a new kernel (i.e. the three .deb files).

closed 2018-12-27 by qzed

brendon-anderson, 2018-12-27:
Great, thank you for your help!