linux-surface/surface-aggregator-module #14: nvidia 410 w/ bumblebee.

remingtonlang, 2019-02-24:
This looks ACPI related. I'm running Ubuntu 18.10. After upgrading to 4.19 from Jake, I found out the 390 drivers are unsupported in 4.19 so I upgraded the nvidia driver to 410.

410 no longer uses prime-select and prefers bumblebee/optirun to enable the graphics card.

```remi@sb2:~$ optirun firefox
[  549.113374] [ERROR]Cannot access secondary GPU - error: Could not load GPU driver

[  549.113409] [ERROR]Aborting because fallback start is disabled.
```

Looking at dmesg:
```[  265.746948] bbswitch: version 0.8
[  265.746960] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[  265.746978] bbswitch: Found discrete VGA device 0000:02:00.0: \_SB_.PCI0.RP05.PXSX
[  265.747354] bbswitch: No suitable _DSM call found.
```
The device ID properly maps to the nvidia card but that _DSM call looks like ACPI related. Any ideas?

qzed, 2019-02-26:
Hmm yeah, I haven't gotten bbswitch to work, not sure what we'd have to do to change that. Also I need to get up to date on the whole GPU switching thing, if I understand it correctly, getting vga-switcheroo to work would be the proper way forward (if that's even possible)? I'll definitely have a look into it.

qzed, 2019-04-11:
For anyone interested: I've been working on [this](https://github.com/qzed/linux-surfacebook2-mshw0153). While it's not a drop-in replacement for bbswitch and not supported by bumblebee, you can manually switch the dGPU on before calling optirun and switch it off afterwards.