linux-surface/surface-aggregator-module #18: Support for Performance-States Support (Performance/Power-Savings/Thermals)

qzed, 2019-04-11:
There is a command for the EC that can set the power-strategy, which is triggered by changing the power-slider on windows. I'm not sure if this applies to all devices.

On the Surface Book 2, this seems to heavily influence the automatic fan-control, at least for the dGPU. I can't check if this influences the CPU fan on the 15" model, as I only have the 13.5" model which comes without a CPU fan. Help here would be appreciated (more on this below).

I've so far tested this with the unigine heaven benchmark and at least from what I can tell it behaves differently on the two different settings that I've tested it so far (lowest- and highest). On the lowest performance setting (0x01) the fans will keep quiet, causing the dGPU to throttle, on the highest setting (0x04) fans will appropriately ramp up, keeping the dGPU cool.

__Helping Out/Testing:__
For this to work, you need to use this module, i.e. go into the `module` directory, build it (`make clean all`), unload the `surface-acpi` module (`rmmod surface_acpi`) and insert the `surfacegen5_acpi` module (`insmod surfacegen5_acpi.ko`).

Then you can run the provided script (`scripts/sysfs_rqst.py`) to set the performance state to highest-performance. If you want to experiment, you can also change the state in the script by changing the following line, replacing `0x04` by the performance state of your choice (`0x01` to `0x04`):
```
21:    data = bytes(performance_state_request(state=0x04))
```

__Requests from Logs:__ (Implementation Stuff)
```
low perf:  aa 55 80 0c 00 a5 d6 c9  80 03 01 00 00 ba 01 03  01 00 00 00  1b e5
high perf: aa 55 80 0c 00 93 43 9f  80 03 01 00 00 a8 01 03  04 00 00 00  c6 0e
```
I've only looked at the lowest (`0x01`) and highest performance (`0x04`) state so far, but I think it's quite likely that there are also states in-between (i.e. `0x02` and `0x03`).

dkabot, 2019-04-11:
I'm gonna put myself out there as clueless and say that the script requires a path which I can't grok what that would be.
Would it be a sysfs path for some MSHW device this driver controls?

qzed, 2019-04-11:
@qzed Sorry, you're right! Totally forgot to mention that. I'll update the instructions right away.

qzed, 2019-04-11:
@dkabot Updated, thanks!

dkabot, 2019-04-11:
Quick testing is promising; CPU and GPU fans are on and much faster.
An `nvidia-smi dmon` shows much better temps only seeming to ever reach 76C, and wattage only majorly dropped when I accidentally lost the power adapter out of its socket (oops).

This also reminds me, 1-4 are probably matching tfhe Windows power profiles exactly, as this is what happened on "Best Performance" (I was scared the CPU fan whirring so much would break something, with how small it is).
Ofc, given this is Linux, the system tends to fare better in basic operation even without maxing out the power profile ;)

Makes me wonder what the default value was.

qzed, 2019-04-11:
Awesome!

> 1-4 are probably matching tfhe Windows power profiles exactly

I think so too, but I'll verify this tomorrow with a bit more testing and logging on windows.

> Makes me wonder what the default value was.

Yeah, me too. I'd say probably 2, if that corresponds to the default setting in Windows.

qzed, 2019-04-12:
Okay, I've done a bit more research and mapped the values to the respective names in Windows (see updated first comment). It looks like state 1 is the default state and the states are not strictly sequential in terms of their performance level, as state 2 seems to be the battery-saver state.

dkabot, 2019-04-12:
Interesting; good to know we aren't defaulted into battery saver and can invoke it.
I wonder how usable the system would be in day to day tasks using it for higher battery life.

qzed, 2019-04-12:
I guess we have to test that. I'm planning on adding a module-parameter so you can set an initial state that is being applied when the module gets loaded (or choose to keep it as-is).

lynscott, 2019-04-12:
@qzed nice! I'll try it out

qzed, 2019-04-12:
I've implemented a basic driver for this, so you can now set the performance-mode via sysfs, specifically by writing the mode value (1-4) to/reading it from `/sys/devices/platform/MSHW0107:00/perf_mode`. I'll update first comment and readme with more details.

dkabot, 2019-04-12:
Nice; I was just trying to do a more straightforward script where I just say a number and nothing more, and now I can very readily do that.
Now operating on power state 2, hopefully I won't notice anything different performance-wise.

qzed, 2019-04-12:
Readme is updated: https://github.com/qzed/linux-surfacegen5-acpi#setting-the-performance-mode.

@dkabot You can also use the module-parameters to set the states on load/boot as described there. I've thought about writing a small command-line utility to make stuff like this more accessible, including the dGPU power state. Might be a project for the future.

kitten, 2019-04-13:
I've just started testing this and it's nothing fancy, but here's a small script to set up systemd targets & services that set `0x02` (battery saver) when AC is diconnected, and `0x04` (best) when AC is connected. https://gist.github.com/kitten/cd78d8531873d5dbea48b1d7bdafd271

qzed, 2019-04-13:
@kitten Awesome, thanks!

qzed, 2019-04-13:
I've started a small command-line application to provide a simpler interface to all the sysfs attributes/specia-devices here: https://github.com/qzed/linux-surface-control. Everything should be supported as of now, but it still needs a bit of polishing.

remingtonlang, 2019-04-14:
@qzed Looks like somewhere dgpu_power isn't setting correctly on SB2. Running the latest kernel module (synced and rebuilt a few minutes before this post)

```
sudo ./surface status
Error: Cannot access device.
       No such file or directory (os error 2).
```

```
--SNIP--
stat("/sys/devices/platform/MSHW0107:00", {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
openat(AT_FDCWD, "/sys/devices/platform/MSHW0107:00/perf_mode", O_RDONLY|O_CLOEXEC) = 3
read(3, "2\n", 4)                       = 2
close(3)                                = 0
stat("/sys/devices/platform/MSHW0153:00", {st_mode=S_IFDIR|0755, st_size=0, ...}) = 0
openat(AT_FDCWD, "/sys/devices/platform/MSHW0153:00/dgpu_power", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
write(2, "Error: ", 7Error: )                  = 7
write(2, "Cannot access device", 20Cannot access device)    = 20
write(2, ".\n", 2.
)                      = 2
write(2, "       ", 7       )                  = 7
write(2, "No such file or directory", 25No such file or directory) = 25
write(2, " (os error ", 11 (os error )             = 11
write(2, "2", 12)                        = 1
write(2, ")", 1))                        = 1
write(2, ".\n", 2.
)                      = 2
sigaltstack({ss_sp=NULL, ss_flags=SS_DISABLE, ss_size=8192}, NULL) = 0
munmap(0x7fdd2d384000, 8192)            = 0
exit_group(0)                           = ?
+++ exited with 0 +++
```

```
ls /sys/devices/platform/MSHW0153:00/
driver_override  firmware_node  modalias  power  subsystem  uevent
```

qzed, 2019-04-15:
@remingtonlang the dGPU stuff is actually implemented in https://github.com/qzed/linux-surfacebook2-mshw0153, have you installed that?

dkabot, 2019-04-23:
The latest Jakeday releases (`4.19.36-1` and `5.0.9-1`) appear to have this merged in and existing by default now.
One look at the commit history probably makes this obvious, but it's still a good milestone to have it regardless.

qzed, 2019-04-23:
Right, thanks for noting that! Support is still missing for non-SB2 devices as I want to make sure that this behaves similarly on the other devices. If anyone wants to help out, we still need some logs (of moving the performance-slider on Windows) from the other devices (Laptop 1, Laptop 2, Pro 5 and Pro 6, see readme on how to get them).

lynscott, 2019-04-24:
Is the surface control tool set in a default state in the latest release?
It doesn't work for me but using the perf_mode sysfs attribute works.

lynscott, 2019-04-24:
@qzed surface: command not found

qzed, 2019-04-24:
@lynscott I assume you're using the .deb? Seems like a packaging mistake, I'll have a look at it right away. Since I'm on a different distro I tend to not notice these kinds of things (although I try to make sure that the package is working), so reporting such issues are much appreciated! Thank you!

lynscott, 2019-04-24:
@qzed No problem! Yes I used the .deb file paired with the latest kernel release.

qzed, 2019-04-24:
@lynscott Okay, the package seems to be valid and contain the executable. Can you try re-installing it and check if there are any warnings or errors during installation, and can you also check if `/usr/bin/surface` (that file should be the executable) exists?

matjohn2, 2019-05-08:
Truly excellent work! 

Wanted to leave feedback in case it's useful. I'm on a 15" i7 SB2.
Kernel: `5.0.7-surface-linux-surface`. 
Nvidia Driver: 418.56 
Apt source: `deb http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu cosmic main`
Package Name: `xserver-xorg-video-nvidia-418/cosmic,now 418.56-0ubuntu0~gpu18.10.1`

## Side Note with Nvidia card disabled at boot using just the nvidia drivers.
Originally had some issues with the nouveau kernel module stealing the card at bootup, even after installing the nvidia drivers deleting the nouveau.ko module from `/lib/modules` it kept appearing (still no idea where it was coming from, wasnt even in my initrd).

Eventually removing `apt remove xserver-xorg-video-nouveau` allowed me to have the nvidia driver loaded correctly. With the ability to select intel vs GTX1060 in the `nvidia-settings` app under "PRIME Profiles"

If i select Intel and log out/back in, the nvidia drivers aren't even loaded, verified with `lsmod`.
(This is without having to use the "disable GPU" driver discussed above.) This is IDEAL for me, as i want to use the intel GFX to power the display 99% of the time, often passing the Nvidia GPU through to KVM virtual machines.

## Testing
With the new ACPI module loaded and `surface performance set  4`, i started testing with `glmark2`, AC was unplugged, the GPU maxed out at about 7W power draw according to nvidia-smi, with power and software throttles activated. However, the fan did start to make noise, which didnt happen at `surface performance set 1`

```
~$ nvidia-smi -q -d PERFORMANCE

==============NVSMI LOG==============

Timestamp                           : Wed May  8 04:11:47 2019
Driver Version                      : 418.56
CUDA Version                        : 10.1

Attached GPUs                       : 1
GPU 00000000:02:00.0
    Performance State               : P8
    Clocks Throttle Reasons
        Idle                        : Not Active
        Applications Clocks Setting : Not Active
        SW Power Cap                : Active
        HW Slowdown                 : Not Active
            HW Thermal Slowdown     : Not Active
            HW Power Brake Slowdown : Not Active
        Sync Boost                  : Not Active
        SW Thermal Slowdown         : Active
        Display Clock Setting       : Not Active
```

With AC then plugged in (i'm using the USB-C port in the SB2 and a Macbook Pro power supply for charging instead of carrying round the separate SB2 charger) the SW power cap disabled and I saw usage of ~47W power draw, the performance mode of the card immediately moved to P0 and my glmark2 results got a lot better, my fans ramped up through the tests to louder than I've ever heard them, in the short test the card stayed at P0 performance with temps around ~58C.

```
~$ nvidia-smi -q -d PERFORMANCE

==============NVSMI LOG==============

Timestamp                           : Wed May  8 04:17:07 2019
Driver Version                      : 418.56
CUDA Version                        : 10.1

Attached GPUs                       : 1
GPU 00000000:02:00.0
    Performance State               : P0
    Clocks Throttle Reasons
        Idle                        : Not Active
        Applications Clocks Setting : Not Active
        SW Power Cap                : Not Active
        HW Slowdown                 : Not Active
            HW Thermal Slowdown     : Not Active
            HW Power Brake Slowdown : Not Active
        Sync Boost                  : Not Active
        SW Thermal Slowdown         : Not Active
        Display Clock Setting       : Not Active
```

Thanks for all your hard work here @qzed if you need any more tests please just shout!

qzed, 2019-05-08:
@matjohn2 Thanks for the extensive write-up! I always assumed that there is some power limiting going on when the device is disconnected from a power source, but having confirmation for that is pretty nice.

As for the nouveau driver loading: I've had some trouble with that in the past. It looks like Gnome loads the driver eventually if it can find the device. [This](https://github.com/jakeday/linux-surface/issues/264#issuecomment-427452156) fixed it for me, specifically the `alias nouveau off` line which has the drawback that nouveau gets completely unloadable.

MadcowD, 2019-06-05:
@qzed I've loaded this new ACPI module and set performance mode to 4 with the latest Jakeday 5.0.x release on his release page and NVIDIA-430 driver on Ubuntu 1804.

When I unplug the charger it still throttles and slows down to a maximum of 5W on Nvidia-smi (guessing the clock of the GPU is being throttled). Why should this still be happening with your module?

Also maybe this is a clue: the dgpu command ield the following error:
```
sudo surface dgpu get
Error: Cannot access device.
       No such file or directory (os error 2).
```

MadcowD, 2019-06-06:
@qzed So to confirm, this makes the surface book unusable when the AC is disconnected.

qzed, 2019-06-06:
@MadcowD Unusable as in desktop stuttering? In that case I'd recommend installing the module I linked above and only using the dGPU for the programs you really need the additional performance (e.g. via optimus) and use the Intel GPU for desktop rendering. This should also give you a better battery life. Also, you might need to set the performance mode after you've disconnected the AC. If I remember correctly, it resets to normal when unplugging.

Resonious, 2019-07-08:
I've done a few runs of TessMark X8 on Windows with AC connected and disconnected. With the slider set to "Better Performance," the scores seem land between 42000 and 60000 regardless of whether or not the charger is plugged in.

Looking at `nvidia-smi` during the benchmark, however, it seems the power consumption was around 67W with AC connected, but capped at 56W with AC disconnected. So, maybe Windows still exhibits AC-based throttling, but to a much lesser extent? `.\nvidia-smi.exe -q -d performance` was showing the same info both when plugged in and not. The slider was on "Better Performance" the whole time.

qzed, 2019-07-09:
@Resonious Thanks for your work!

> it seems the power consumption was around 67W with AC connected, but capped at 56W with AC disconnected.

Makes sense that there is some limitation. At some point we're also limited by the amount of wattage the battery itself can deliver (not sure if that's what you hit on Windows), but this definitely seems to be a lot more than the reported 7W under Linux.

As mentioned before, I suspect that there's something going on with the driver on Linux. There is a `PowerMizer` setting in `nvidia-settings` that you could try and play around with (https://www.nvidia.com/object/feature_powermizer.html, you can also change that via X.org settings somehow). Unfortunately though I haven't yet figured out how to get `nvidia-settings` to run or found any other way to change that. It seems that you can only run it on an X-Server instance that itself is running on the nvidia GPU, so it's not going to work via optimus.

To get `nvidia-settings` to run (or the X.org settings for the nvidia GPU to be applied), you'd probably have to enable the dGPU at boot via the module parameters (see https://github.com/qzed/linux-surfacebook2-mshw0153#via-module-parameters, note that you have to specify them for `surface_acpi` if you're using the patches from my latest PR) to then be able to start an X-server on it (which may require some additional configuration).

Why this PowerMizer setting is only available via `nvidia-settings`/the nvidia X-Server driver (and not via `nvidia-smi`) is beyond me... That being said, there's also no way to specify/change the power limit or power-state on the dGPU directly via `nvidia-smi`, because of course that's been disabled.

So what I'm proposing is that we should probably test this on an X-Server instance running on the nvidia GPU and play around with the settings there.

Resonious, 2019-07-09:
I've been using [nvidia-xrun](https://github.com/Witko/nvidia-xrun) to open an X server with the Nvidia proprietary driver. Running `nvidia-settings` seems to spit out `ERROR: Unable to find display on any available system.`, but I was able to open the GUI successfully with `nvidia-settings -c $DISPLAY` for some reason.

[image]

It looks like there's a "Prefer Maximum Performance" option, but it doesn't seem to change anything when I select it on battery power. Nothing seems to make the performance levels budge on battery. When on AC power, I can see the performance levels go up and down when I move the mouse and do nothing respectively.

I'll try playing with the `nvidia` module parameters and other X settings I can dig up.

qzed, 2019-07-09:
Hmm, I had hoped that would help a bit. I'll have a look at the SMC/EC communication again when I find the time to do that and check if I've missed something.

archseer, 2020-02-29:
@danielzgtg @fxgsell @lecram1989 @Knard

Hi! You're on the [testers list on the wiki](https://github.com/linux-surface/linux-surface/wiki/Testers-and-Contributing) and I could use your help. We would like to enable performance states for more devices but we're not sure if the setting works on all devices.

Performance states are basically the performance slider available on windows where you can set the performance to Recommended / Battery Saver / Better Performance / Best.

To test whether this works on your device:

Save this script to a file named `rqst.py`:
```python
#!/usr/bin/env python3
import sys
import os

PATH_DEV_RQST = '/sys/bus/serial/devices/serial0-0/rqst'

def performance_state_request(state):
    return bytes([0x03, 0x03, 0x00, 0x01, 0x00, 0x04, state, 0x00, 0x00, 0x00])

def main():
    fd = os.open(PATH_DEV_RQST, os.O_RDWR | os.O_SYNC)

    #            [  TC,  CID,  IID,  PRI,  SNC,  CDL, payload...]
    data = bytes(map(lambda x:int(x, 16), sys.argv[1:]))

    os.write(fd, data)

    os.lseek(fd, 0, os.SEEK_SET)
    length = os.read(fd, 1)[0]
    data = os.read(fd, length)

    os.close(fd)

    print(' '.join(['{:02x}'.format(x) for x in data]))

if __name__ == '__main__':
    main()
```

Then run the lines with `$` in your shell:
```
$ chmod +x rqst.py
```
Check your current performance setting:
```
$ sudo rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
01 00 00 00 03 00 01 00
```
The leading 01 means you're in the [default mode](https://github.com/linux-surface/surface-aggregator-module/issues/18#issue-432247668).

Next, change the mode to performance (0x04):
```
$ sudo rqst.py 0x03 0x03 0x00 0x01 0x00 0x01 0x04

```
Check the current value and it should return a value starting with `04`:
```
$ sudo rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
04 00 00 00 03 00 04 00
```
Now set it back to `01`:
```
$  sudo rqst.py 0x03 0x03 0x00 0x01 0x00 0x01 0x01

```
And confirm:
```
$ sudo rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
01 00 00 00 03 00 01 00
```

Send me a terminal log of these commands. If everything worked fine I'll enable performance states for your device.

danielzgtg, 2020-03-01:
```
home@daniel-tablet1:~$ cat > rqst.py << "END"
> #!/usr/bin/env python3
> import sys
> import os
> 
> PATH_DEV_RQST = '/sys/bus/serial/devices/serial0-0/rqst'
> 
> def performance_state_request(state):
>     return bytes([0x03, 0x03, 0x00, 0x01, 0x00, 0x04, state, 0x00, 0x00, 0x00])
> 
> 
> def main():
>     fd = os.open(PATH_DEV_RQST, os.O_RDWR | os.O_SYNC)
> 
>     #            [  TC,  CID,  IID,  PRI,  SNC,  CDL, payload...]
>     data = bytes(map(lambda x:int(x, 16), sys.argv[1:]))
> 
>     os.write(fd, data)
> 
>     os.lseek(fd, 0, os.SEEK_SET)
>     length = os.read(fd, 1)[0]
>     data = os.read(fd, length)
> 
>     os.close(fd)
> 
>     print(' '.join(['{:02x}'.format(x) for x in data]))
> 
> 
> if __name__ == '__main__':
>     main()
> END
home@daniel-tablet1:~$ chmod +x rqst.py 
home@daniel-tablet1:~$ sudo ./rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
[sudo] password for home: 
04 00 00 00 03 00 04 00
home@daniel-tablet1:~$ sudo ./rqst.py 0x03 0x03 0x00 0x01 0x00 0x01 0x04
[sudo] password for home: 

home@daniel-tablet1:~$ sudo ./rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
[sudo] password for home: 
04 00 00 00 03 00 04 00
home@daniel-tablet1:~$ sudo ./rqst.py 0x03 0x03 0x00 0x01 0x00 0x01 0x01
[sudo] password for home: 

home@daniel-tablet1:~$ sudo ./rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
[sudo] password for home: 
01 00 00 00 03 00 01 00
home@daniel-tablet1:~$ 

```

danielzgtg, 2020-03-01:
I don't think you expected my tablet to be already in performance mode, so the second `rqst` did nothing. I reran the first 3 `rqst`s:

```
home@daniel-tablet1:~$ sudo ./rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
[sudo] password for home: 
01 00 00 00 03 00 01 00
home@daniel-tablet1:~$ sudo ./rqst.py 0x03 0x03 0x00 0x01 0x00 0x01 0x04
[sudo] password for home: 

home@daniel-tablet1:~$ sudo ./rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
[sudo] password for home: 
04 00 00 00 03 00 04 00
home@daniel-tablet1:~$ 

```

archseer, 2020-03-01:
Alright cool, that means it works on SP7 (I expected that since it's also working on SL3). Thanks for testing!

knard, 2020-03-01:
I got the following results :
```
knard@knard-surface:~$ sudo ./rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
01 00 00 00 05 00 01 00
knard@knard-surface:~$ sudo ./rqst.py 0x03 0x03 0x00 0x01 0x00 0x01 0x04

knard@knard-surface:~$ sudo ./rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
04 00 00 00 05 00 04 00
knard@knard-surface:~$ sudo ./rqst.py 0x03 0x03 0x00 0x01 0x00 0x01 0x01

knard@knard-surface:~$ sudo ./rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
01 00 00 00 05 00 01 00
```

archseer, 2020-03-03:
Cool, I enabled performance states for SP5/6/7: https://github.com/linux-surface/surface-aggregator-module/commit/344033be0d828e2751a1ac47b70e4112120e832a

(SP5 uses the same hardware ID as 5)

It should show up in the kernel builds once the patches are updated.

fxgsell, 2020-03-03:
Looks like it also works on the AMD version.

`
(base) fx@FX-Surface-Laptop-3:~$ sudo ./rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
01 00 00 00 01 00 01 00
(base) fx@FX-Surface-Laptop-3:~$ sudo ./rqst.py 0x03 0x03 0x00 0x01 0x00 0x01 0x04

(base) fx@FX-Surface-Laptop-3:~$ sudo ./rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
04 00 00 00 01 00 04 00
(base) fx@FX-Surface-Laptop-3:~$ sudo ./rqst.py 0x03 0x03 0x00 0x01 0x00 0x01 0x01

(base) fx@FX-Surface-Laptop-3:~$ sudo ./rqst.py 0x03 0x02 0x00 0x01 0x01 0x00
01 00 00 00 01 00 01 00
(base) fx@FX-Surface-Laptop-3:~$ 
`

archseer, 2020-03-19:
Pushed https://github.com/linux-surface/surface-aggregator-module/commit/6a69a85c8649ac5df4748a09e25d8b7ae40b7d0a

That leaves us with Surface Laptop 1 and 2 being undetermined.

qzed, 2020-07-24:
Since we know that performance modes are available on every other device with SAM-over-SSH, I'm going to assume they are also available on the SL1 and SL2. Just added the required SID devices, so there should now be support for those.

closed 2020-07-24 by qzed

aarow12, 2020-09-25:
Has anyone found a solution or workaround to the dgpu power/clock limit(throttle) when not plugged into an A/C power charge chord? I'm running into the same problem mentioned throughout this issue. When plugged in, the dgpu runs great, however, when not plugged in, performance is significantly limited. I'm running the dgpu on the Linux-Surface kernel on my Surface Book 3 with the surface control package and Bumblebee using `optirun`. So far, I have not been able to find a way to change the power/clock limit:
```
optirun -b none nvidia-settings -c :8 -q GPUCurrentClockFreqsString

Attribute 'GPUCurrentClockFreqsString' (aarowSB3:8.0): nvclock=300,
nvclockmin=300, nvclockmax=645, nvclockeditable=1, memclock=405,
memclockmin=405, memclockmax=405, memclockeditable=1, memTransferRate=810,
memTransferRatemin=810, memTransferRatemax=810, memTransferRateeditable=1
Attribute 'GPUCurrentClockFreqsString' (aarowSB3:8[gpu:0]): nvclock=300,
nvclockmin=300, nvclockmax=645, nvclockeditable=1, memclock=405,
memclockmin=405, memclockmax=405, memclockeditable=1, memTransferRate=810,
memTransferRatemin=810, memTransferRatemax=810, memTransferRateeditable=1
```
and
```
nvidia-smi -q -d CLOCK                                     

==============NVSMI LOG==============

Timestamp                                 : Fri Sep 25 15:04:18 2020
Driver Version                            : 450.66
CUDA Version                              : 11.0

Attached GPUs                             : 1
GPU 00000000:02:00.0
    Clocks
        Graphics                          : 300 MHz
        SM                                : 300 MHz
        Memory                            : 405 MHz
        Video                             : 540 MHz
...
```
Whereas when the computer is plugged into the A/C charger, clock frequencies are 1020 MHz. (don't have access to a charger currently) 

Any help is greatly appreciated!

qzed, 2020-09-26:
Unfortunately I'm not aware of any progress on this. There is a [`function`](https://github.com/linux-surface/acpidumps/blob/9e5706ab2409139b16eda97b49f03af0c3a1da00/surface_studio_2/ssdt6.dsl#L4009-L4015) in ACPI to set the dGPU power limit, but this eventually makes a RQSG call to the Surface ACPI Notify (SAN) driver. So without us knowing what that RQSG call specifically does, we can't implement it and I assume that's what's missing here.

So to figure this out, you'll probably have to reverse-engineer the Windows SAN driver and try to figure out what it does in that case. Also there are no guarantees that that's actually going to change anything, that's just a guess based on the message in ACPI.

aarow12, 2020-09-27:
Thanks so much for your reply! I'm new to reading ACPI source language, but after researching a bit and looking at the function and other parts of the file you referred, I believe I can see basically what you are saying. To make sure I understand, the `\DGFX.SPL0` function calls `\_SB._SAN.SDXP` which then calls `\_SB._SAN.RQSG`, which is a `MethodObj`. Does that sound right? If so, is  `\_SB._SAN.RQSG` defined anywhere else in the ACPI sources, or is it a pretty generic function for drivers in general, and which we don't have the source code for? Thanks again! @qzed

qzed, 2020-09-27:
Yes, sounds about right. `\_SB._SAN.RQSG` is defined in [`dsdt.dsl`](https://github.com/linux-surface/acpidumps/blob/05cba925f3a515f222acb5b3551a032ddde958fe/surface_book_2/dsdt.dsl#L15583) and essentially calls `\_SB._SAN.RQSX`. That performs a [write-read operation](https://github.com/linux-surface/acpidumps/blob/05cba925f3a515f222acb5b3551a032ddde958fe/surface_book_2/dsdt.dsl#L15627) to a GenericSerialBus operation region (`SAN0`). This operation region access must be implemented by a driver and is not handled by ACPI code itself. On Linux, that's implemented [here](https://github.com/linux-surface/surface-aggregator-module/blob/3777b0c251708bbfe8a24746e79b22da7b279fb9/module/src/clients/surface_acpi_notify.c#L630).

Note that this operation region handles both calls from `RQSG` and `RQST`. `RQST` is translated to EC requests to the Surface Aggregator Module EC, but for `RQSG` we don't know how to handle those. So `RQSG` is currently just forwarded to notifiers via [`san_dgpu_notifier_call()`](https://github.com/linux-surface/surface-aggregator-module/blob/3777b0c251708bbfe8a24746e79b22da7b279fb9/module/src/clients/surface_acpi_notify.c#L619), which is basically just a way to give any other (future) driver access to this by allowing them to register a callback.

So, in essence, the call is forwarded to the Surface ACPI Notify driver, but the Linux driver doesn't know what to do with it. So we'd need to figure out what to do with it and implement it. Basically, that would require us to reverse-engineer the Windows driver and figure out what it is doing with that after the operation region access.