linux-surface/surface-aggregator-module #21: Surface Pro 7

aleksfadini, 2019-10-24:
Any chance that this could work on the new Surface Pro 7 as well?
Jakeday kernel battery/acpi does not work.

I tried to run "make" in /modules and got this:

`
make -C /lib/modules/"5.3.6-1-surface"/build M= modules
make[1]: *** /lib/modules/5.3.6-1-surface/build: No such file or directory.  Stop.
make: *** [Makefile:29: all] Error 2

`

qzed, 2019-10-25:
We've discussed the Laptop 3 on IRC. I think the Pro 7 should be fairly similar (meaning that we would need to write some sort of stub-driver which provides a battery interface and talks to the EC, luckily the last part is the same from the previous generations). Can you upload an acpidump (`sudo acpidump > acpidump.out`)?

Regarding the make error: Have you installed a new kernel and did not reboot? What distro are you running? In any case, if it doesn't work with the surface-linux kernel, it will, at the moment, not work with the module either.

aleksfadini, 2019-10-25:
Hey man! I will post that as soon as I get home. It would mean a lot if you could implement the battery in it. 
I did use the surface-linux arch repo and wanted to compile from source as a separate move :)

I would be happy to donate via PayPal if you can work on it, could you post your PayPal account somewhere?

aleksfadini, 2019-10-26:
Here is my acpi-dump (had to add .txt to upload it to GH)
[acpidump.out.txt](https://github.com/qzed/linux-surfacegen5-acpi/files/3774222/acpidump.out.txt)

aleksfadini, 2019-10-26:
On a side note, if you could include this patch for the surface linux kernel, you could easily fix the touchscreen as well (and maybe it is a similar situation for the surface book 3):

> 
> -touchscreen (related to intel_lpss blacklist otherwise wont boot)
> 
> Here is more info on touchscreen issue:
> 
>     First Boot
>     To boot fully without the boot process hanging one must blacklist the 'intel_lpss_pci' module, which can be done on a live media installation by selecting the boot option you want and pressing 'e', navigating to the end of the boot options string, and adding 'modprobe.blacklist=intel_lpss_pci'. This should allow booting into the full OS and further installation to the internal SSD.
> 
>     Intel LPSS
>     To fix the intel_lpss_module the patch from this paste must be applied to the kernel.
> 
>     https://pastebin.com/sqPv8ShP
> 
>     Remember to be cautious about running code or installing patches from third party sources if you do not know what it does.
> 
>     This patch should enable the system to boot without blacklisting the module, further enabling access to the touchscreen.
> 

from https://wiki.archlinux.org/index.php/Dell_XPS_13_2-in-1_(7390)

qzed, 2019-10-27:
> Big fan of your work, just sent you a donation brother

Thank you, this is much appreciated!

I can't give you a concrete time-frame when the battery stuff will work, but I hope that it doesn't take too long. At the moment, my first priority is getting the keyboard to work on the Laptop 3, but I will work on battery status on both devices after that. In essence, all those things are fairly similar to the previous generations, it's just that some ACPI stuff has changed.

If you know anything about ACPI: the battery definitions, including _BIX and _BST have been removed and it seems to rely solely on the EC requests. Those are the same as on previous generations, though, so I think the difficult part is done. Unfortunately I don't know much about the Linux battery subsystem and thus I don't have any idea how long this will take to implement. The keyboard should be simpler to implement as it seems that this is basically just a matter of different IDs.

> On a side note, if you could include this patch for the surface linux kernel, you could easily fix the touchscreen as well (and maybe it is a similar situation for the surface book 3):

Thank you for the link. Could you try the patch (I could compile a kernel for you if you want me to)? Also do you know what type of touchscreen the device uses (e.g. I2C)?

aleksfadini, 2019-10-27:
I would definitely try a kernel if you were so kind to compile one. Maybe it could be surface-linux-pro7 or something like that, in the arch repo?

I didn't know the keyboard of he sf laptop 3 doesn't work.

qzed, 2019-10-27:
@aleksfadini I've compiled and uploaded a kernel with the patch: https://github.com/qzed/linux-surface/releases/tag/v5.3-testing.1

aleksfadini, 2019-10-27:
That is awesome! Will install them with sudo pacman -U and report back asap. Infinite thanks.

aleksfadini, 2019-10-27:
Can confirm, with the v5.3-testing.1 patched kernel the boot does not hang, so I could remove the kernel parameter:
`'modprobe.blacklist=intel_lpss_pci`
And still boot normally!
Now I should be able to configure the touch screen. Looking into what commands to use for diagnostics and to figure out what type it is (I2C?)

aleksfadini, 2019-10-27:
By the way, installinf the v5.3-testing.1 on arch with sudo pacman -U gives this error:

`
 lding image from preset: /etc/mkinitcpio.d/linux-surface-dev.preset: 'default'
 15   -> -k /boot/vmlinuz-linux-surface-dev -c /etc/mkinitcpio.conf -g
 14 /boot/initramfs-linux-surface-dev.img
 13 ==> Starting build: 5.3.0-1-surface-dev
 12   -> Running build hook: [base]
 11   -> Running build hook: [udev]
 10   -> Running build hook: [autodetect]
  9   -> Running build hook: [modconf]
  8   -> Running build hook: [block]
  7   -> Running build hook: [filesystems]
  6   -> Running build hook: [keyboard]
  5   -> Running build hook: [fsck]
  4 ==> ERROR: module not found: `surface_acpi'
  3 ==> Generating module dependencies
  2 ==> Creating gzip-compressed initcpio image:
  1 /boot/initramfs-linux-surface-dev.img
17  ==> WARNING: errors were encountered during the build. The image may not be
  1 complete.
  2 ==> Building image from preset: /etc/mkinitcpio.d/linux-surface-dev.preset:
  3 'fallback'
  4   -> -k /boot/vmlinuz-linux-surface-dev -c /etc/mkinitcpio.conf -g
  5 /boot/initramfs-linux-surface-dev-fallback.img
  6 -S autodetect
  7 ==> Starting build: 5.3.0-1-surface-dev
  8   -> Running build hook: [base]
  9   -> Running build hook: [udev]
 10   -> Running build hook: [modconf]
 11   -> Running build hook: [block]
 12 ==> WARNING: Possibly missing firmware for module: aic94xx
 13 ==> WARNING: Possibly missing firmware for module: wd719x
 14   -> Running build hook: [filesystems]
 15   -> Running build hook: [keyboard]
 16   -> Running build hook: [fsck]
 17 ==> ERROR: module not found: `surface_acpi'
 18 ==> Generating module dependencies
 19 ==> Creating gzip-compressed initcpio image:
 20 /boot/initramfs-linux-surface-dev-fallback.img
 21 ==> WARNING: errors were encountered during the build. The image may not be
 22 complete.
 23 error: command failed to execute correctly

`

However the kernel works fine!

qzed, 2019-10-28:
With regards to the error: It looks like you've installed surface_acpi via dkms in the past, but something may have gone wrong. You could try and play around with `dkms` to remove it.

qzed, 2019-10-28:
Thanks for testing, I'll include the patch then.

aleksfadini, 2019-10-28:
> With regards to the error: It looks like you've installed surface_acpi via dkms in the past, but something may have gone wrong. You could try and play around with `dkms` to remove it.

Yes! that's correct. I tried things I didn't quite understand in the hope to get the battery going. Will work on it.

Output of less /proc/bus/input/devices:
`
 I: Bus=0019 Vendor=0000 Product=0005 Version=0000
  1 N: Name="Lid Switch"
  2 P: Phys=PNP0C0D/button/input0
  3 S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0D:00/input/input0
  4 U: Uniq=
  5 H: Handlers=event0
  6 B: PROP=0
  7 B: EV=21
  8 B: SW=1
  9
 10 I: Bus=0019 Vendor=0000 Product=0006 Version=0000
 11 N: Name="Video Bus"
 12 P: Phys=LNXVIDEO/video/input0
 13 S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input1
 14 U: Uniq=
 15 H: Handlers=kbd event1
 16 B: PROP=0
 17 B: EV=3
 18 B: KEY=3e000b00000000 0 0 0
 19
 20 I: Bus=0003 Vendor=045e Product=09c0 Version=0111
 21 N: Name="Microsoft Surface Type Cover Keyboard"
 22 P: Phys=usb-0000:00:14.0-8/input0
 23 S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:045E:09C0.0001/input/input2
 24 U: Uniq=
 25 H: Handlers=sysrq kbd event2 leds
 26 B: PROP=0
 27 B: EV=120013
 28 B: KEY=1000000000007 ff800000000007ff febeffdff3cfffff fffffffffffffffe
 29 B: MSC=10
 30 B: LED=7
 31
 32 I: Bus=0003 Vendor=045e Product=09c0 Version=0111
 33 N: Name="Microsoft Surface Type Cover Consumer Control"
 34 P: Phys=usb-0000:00:14.0-8/input0
 35 S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:045E:09C0.0001/input/input4
 36 U: Uniq=
 37 H: Handlers=kbd event3
 38 B: PROP=0
 39 B: EV=1f
 40 B: KEY=3f000301ff 0 0 483ffff17aff32d bfd4444600000000 1 130ff38b17c000 677bfad9415fed 19ed68000004400 10000002
 41 B: REL=1040
 42 B: ABS=100000000
 43 B: MSC=10
 44
 45 I: Bus=0003 Vendor=045e Product=09c0 Version=0111
 46 N: Name="Microsoft Surface Type Cover Touchpad"
 47 P: Phys=usb-0000:00:14.0-8/input0
 48 S: Sysfs=/devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:045E:09C0.0001/input/input6
 49 U: Uniq=
 50 H: Handlers=event4 mouse0
 51 B: PROP=5
 52 B: EV=1b
 53 B: KEY=e520 10000 0 0 0 0
 54 B: ABS=260800000000003
 55 B: MSC=20
 56
 57 I: Bus=0010 Vendor=001f Product=0001 Version=0100
 58 N: Name="PC Speaker"
 59 P: Phys=isa0061/input0
 60 S: Sysfs=/devices/platform/pcspkr/input/input21
 61 U: Uniq=
 62 H: Handlers=kbd event5
 63 B: PROP=0
 64 B: EV=40001
 65 B: SND=6
 66
 67 I: Bus=0000 Vendor=0000 Product=0000 Version=0000
 68 N: Name="HDA Intel PCH Mic"
 69 P: Phys=ALSA
 70 S: Sysfs=/devices/pci0000:00/0000:00:1f.3/sound/card0/input22
 71 U: Uniq=
 72 H: Handlers=event6
 73 B: PROP=0
 74 B: EV=21
 75 B: SW=10
 76
 77 I: Bus=0000 Vendor=0000 Product=0000 Version=0000
 78 N: Name="HDA Intel PCH Headphone"
 79 P: Phys=ALSA
 80 S: Sysfs=/devices/pci0000:00/0000:00:1f.3/sound/card0/input23
 81 U: Uniq=
 82 H: Handlers=event7
 83 B: PROP=0
 84 B: EV=21
 85 B: SW=4

`

Do you think the touchpad is still not being seen by the kernel? How do I find it if it is I2C?

aleksfadini, 2019-10-28:
I got this:

`dmesg | grep i2c
[    1.631795] Modules linked in: i915(+) intel_gtt i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt
fb_sys_fops drm agpgart mei hid_multitouch usbhid hid_generic i2c_hid hid_sensor_hub hid
[    1.632107] Modules linked in: i915(+) intel_gtt i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt
fb_sys_fops drm agpgart mei hid_multitouch usbhid hid_generic i2c_hid hid_sensor_hub hid
[    1.633518] Modules linked in: i915(+) intel_gtt i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt
fb_sys_fops drm agpgart mei hid_multitouch usbhid hid_generic i2c_hid hid_sensor_hub hid
[   10.606125]  nls_cp437 intel_uncore intel_spi_pci intel_spi spi_nor vfat snd_timer fat intel_rapl_perf pcspkr cfg80211 snd idma64 mtd soundcore mei_me intel_lpss_pci intel_lpss intel_ish_ipc int3403_thermal int340x_thermal_zone rfkill intel_ishtp wmi surface_sam_san surfacepro3_button surface_sam_ssh evdev soc_button_array mac_hid int3400_thermal acpi_tad acpi_thermal_rel dptf_power ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 xhci_pci
xhci_hcd crc32c_intel intel_ipts i915 intel_gtt i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm agpgart mei hid_multitouch usbhid hid_generic i2c_hid hid_sensor_hub hid
`

qzed, 2019-10-28:
> Yeah, looks like the touchscre

You mean touchscreen? Looks like it's not present there.

> How do I find it if it is I2C?

Unfortunately, I2C devices are difficult to identify. There's no mechanism built into the specification as for example on USB or PCIe. You'd likely have to look around on Windows if you can find anything there. A good place to start would be the device manager. You could try to look for the device by its connection (View->Devices by connection) and figure out its "BIOS device name" property, if it has one (right click -> properties -> details -> search for the entry in the drop-down menu). Also other stuff like which driver it uses.

aleksfadini, 2019-10-28:
I entirely eradicated Windows. I realize this could be particularly bad for updating firmware reasons too. Maybe I can repartition and dual boot... Touchscreen is priority number 1, battery number 2. Where can I find you on IRC? If we could find some way to fix the touchscreen that would really help.

qzed, 2019-10-28:
> I realize this could be particularly bad for updating firmware reasons too.

Thanks to @StollD updating the firmware can also be done from Linux and is arguably better (you can also down-grade, unlike on Windows). The problem is only in getting the latest firmware files. It usually takes some time until MS uploads them to their [page](https://support.microsoft.com/en-us/help/4023482/surface-download-drivers-and-firmware-for-surface).

Apart form that I'd recommend you still keep Windows around for a while, it's a new device after all.

> Where can I find you on IRC?

freenode/##linux-surface, when I'm at home

Hugal31, 2019-10-30:
I was told to dump my Windows devices information here about the Battery and the Touchscreen.

The battery:
```
Device {BFD21D0B-D701-435B-AF19-4A20677898D3}\SurfaceBattery\3&c7da34a&0&SID_DEVICE_03

Driver Name: oem16.inf
Class Guid: {72631e54-78a4-11d0-bcf7-00aa00b7b32a}
Driver Date: 08/22/2019
Driver Version: 1.60.139.0
Driver Provider: Surface
Driver Section: SurfaceBattery_Device.NT
Driver Rank: 0xFF0000
Matching Device Id: {BFD21D0B-D701-435B-AF19-4A20677898D3}\SurfaceBattery
Outranked Drivers: 
Device Updated: false
Parent Device: ACPI\MSHW0116\2&daba3ff&1
```

Intel(R) Precise Touch Device:
```
Device PCI\VEN_8086&DEV_34E4&SUBSYS_00391414&REV_30\3&11583659&0&B4

Driver Name: oem68.inf
Class Guid: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Driver Date: 07/30/2019
Driver Version: 2.1.0.59
Driver Provider: Intel Corporation
Driver Section: precisetouch.Inst.NT
Driver Rank: 0xFF2001
Matching Device Id: PCI\VEN_8086&DEV_34E4
Outranked Drivers: 
Device Updated: false
Parent Device: ACPI\PNP0A08\0
```

<details>
<summary>Its children</summary>

Bluetooth OOB Coupling Channel

```
Device HID\VEN_8086&DEV_34E4&SUBSYS_00391414&REV_30&Col09\4&1f1b4878&0&0008

Driver Name: BthOob.inf
Class Guid: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Driver Date: 06/21/2006
Driver Version: 10.0.18362.145
Driver Provider: Microsoft
Driver Section: BthOob.Install.NT
Driver Rank: 0xFF0004
Matching Device Id: HID\VID_045E&UP:FFF4_U:0001
Outranked Drivers: input.inf:HID_DEVICE_UPR:FF00-FFFF:00FF1006 input.inf:HID_DEVICE:00FF1007
Device Updated: false
Parent Device: PCI\VEN_8086&DEV_34E4&SUBSYS_00391414&REV_30\3&11583659&0&B4
```

HID-compliant pen
```
Device HID\VEN_8086&DEV_34E4&SUBSYS_00391414&REV_30&Col08\4&1f1b4878&0&0007

Driver Name: input.inf
Class Guid: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Driver Date: 06/21/2006
Driver Version: 10.0.18362.175
Driver Provider: Microsoft
Driver Section: HID_Raw_Inst.NT
Driver Rank: 0xFF1005
Matching Device Id: HID_DEVICE_UP:000D_U:0002
Outranked Drivers: input.inf:HID_DEVICE:00FF1006
[... 119 lines cut ...]
Parent Device: PCI\VEN_8086&DEV_34E4&SUBSYS_00391414&REV_30\3&11583659&0&B4
```

Surface Virtual Function Enum Device
```
Device HID\VEN_8086&DEV_34E4&SUBSYS_00391414&REV_30&Col04\4&1f1b4878&0&0003

Driver Name: heat.inf
Class Guid: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Driver Date: 06/21/2006
Driver Version: 10.0.18362.1
Driver Provider: Microsoft
Driver Section: HeatHid.Inst.NT
Driver Rank: 0xFF0005
Matching Device Id: HID_DEVICE_UP:000D_U:000F
Outranked Drivers: input.inf:HID_DEVICE:00FF1006
Device Updated: false
Parent Device: PCI\VEN_8086&DEV_34E4&SUBSYS_00391414&REV_30\3&11583659&0&B4
```
</details>

And the driver files:
[firmware.zip](https://github.com/qzed/linux-surfacegen5-acpi/files/3787762/firmware.zip)

I hope it helps. Thanks for all your work!

StollD, 2019-10-30:
> Intel(R) Precise Touch Device:

This makes me wonder: Could the intel variants still be using IPTS, but maybe it got modified so much that the old driver from intel we are using cannot detect it anymore?

If you look at the firmware packages from Microsoft, the SP7 and the SL3-Intel still contain the `itouch` driver, that is also in the downloads for older surfaces with IPTS. However it has seen some updates since it's last release in a surface product as it seems.

The driver in the firmware package for SB2 seems reference the following PCI devices:
* `8086:9d3e` (thats what I see on my SB2)
* `8086:a13e` (this might be the for SB1 / SP4 who seem to use different IPTS hardware)

The new driver has all of these:
* `8086:9d3e`
* `8086:a13e`
* `8086:9de4`
* `8086:a364`
* `8086:34e4`
* `8086:3de4`
* `8086:02e4`
* `8086:06e4`

All these new IDs are commented with stuff like `CNP` (cannonpoint?), `ICL` (icelake?) etc, so I guess they refactored IPTS somewhere after Kabylake.

```
%precisetouch.DeviceDesc%    = precisetouch.Inst, PCI\VEN_8086&DEV_9D3E
%precisetouch.DeviceDesc%    = precisetouch.Inst, PCI\VEN_8086&DEV_A13E

;CNP_PCH-LP HECI3 HWID
%precisetouch.DeviceDesc%    = precisetouch.Inst, PCI\VEN_8086&DEV_9DE4
;CNP_PCH-H HECI3 HWID
%precisetouch.DeviceDesc%    = precisetouch.Inst, PCI\VEN_8086&DEV_A364

;ICL_PCH-LP HECI3 HWID
%precisetouch.DeviceDesc%    = precisetouch.Inst, PCI\VEN_8086&DEV_34E4
;ICL_PCH-H HECI3 HWID
%precisetouch.DeviceDesc%    = precisetouch.Inst, PCI\VEN_8086&DEV_3DE4

;CML_PCH-LP HECI3 HWID
%precisetouch.DeviceDesc%    = precisetouch.Inst, PCI\VEN_8086&DEV_02E4
;CMP_PCH-H HECI3 HWID
%precisetouch.DeviceDesc%    = precisetouch.Inst, PCI\VEN_8086&DEV_06E4
```

What is missing from the firmware bundle is the driver that used to provide the driver level firmware, which makes me guess that they might have moved it into the touch firmware of the BIOS somehow. We will have to figure that out somehow. 

@Hugal31 could you boot into linux and post the output of these two commands?
```bash
# lspci will not show proper device IDs by default...
$ lspci -nn
$ ls -l /sys/bus/mei/devices/
```

Maybe we can identify the new MEI ID for IPTS that way, and try getting the driver to probe.

Also, the IPTS driver on Windows seems to store some settings in the registry, maybe that can help us with where it gets the driver firmware from: Could you run `regedit`, and search for `itouch` or `IntTouch` under `HKEY_LOCAL_MACHINE`? And then just screenshot everything you can find under it (or export the key, IIRC that was possible but I am not sure).

The older drivers seem to use `HKLM\SYSTEM\CurrentControlSet\Services\IntTouch` so that might be worth checking out first (the new driver doesn't have the full path specified).

aleksfadini, 2019-10-30:
@StollD  thank you for looking into this. Using SP7 here as well, but no Windows installed, so here are the outputs I can give:

$ lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Device [8086:8a12] (rev 03)
00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:8a52] (rev 07)
00:04.0 Signal processing controller [1180]: Intel Corporation Device [8086:8a03] (rev 03)
00:0d.0 USB controller [0c03]: Intel Corporation Ice Lake Thunderbolt 3 USB Controller [8086:8a13] (rev 03)
00:12.0 Serial controller [0700]: Intel Corporation Device [8086:34fc] (rev 30)
00:14.0 USB controller [0c03]: Intel Corporation Ice Lake-LP USB 3.1 xHCI Host Controller [8086:34ed] (rev 30)
00:14.2 RAM memory [0500]: Intel Corporation Device [8086:34ef] (rev 30)
00:14.3 Network controller [0280]: Intel Corporation Device [8086:34f0] (rev 30)
00:15.0 Serial bus controller [0c80]: Intel Corporation Ice Lake-LP Serial IO I2C Controller #0 [8086:34e8] (rev 30)
00:16.0 Communication controller [0780]: Intel Corporation Device [8086:34e0] (rev 30)
00:16.4 Communication controller [0780]: Intel Corporation Device [8086:34e4] (rev 30)
00:19.0 Serial bus controller [0c80]: Intel Corporation Ice Lake-LP Serial IO I2c Controller #4 [8086:34c5] (rev 30)
00:1d.0 PCI bridge [0604]: Intel Corporation Ice Lake-LP PCI Express Root Port #9 [8086:34b0] (rev 30)
00:1e.0 Communication controller [0780]: Intel Corporation Ice Lake-LP Serial IO UART Controller #0 [8086:34a8] (rev 30)
00:1f.0 ISA bridge [0601]: Intel Corporation Ice Lake-LP LPC Controller [8086:3482] (rev 30)
00:1f.3 Audio device [0403]: Intel Corporation Device [8086:34c8] (rev 30)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Ice Lake-LP SPI Controller [8086:34a4] (rev 30)
01:00.0 Non-Volatile memory controller [0108]: SK hynix Device [1c5c:1327]

$ ls -l /sys/bus/mei/devices/
total 0
lrwxrwxrwx 1 root root 0 Oct 30 14:30 mei::082ee5a7-7c25-470a-9643-0c06f0466ea1:00 -> ../../../devices/pci0000:00/0000:00:16.0/mei::082ee5a7-7c25-470a-9643-0c06f0466ea1:00
lrwxrwxrwx 1 root root 0 Oct 30 14:30 mei::309dcde8-ccb1-4062-8f78-600115a34327:01 -> ../../../devices/pci0000:00/0000:00:16.0/mei::309dcde8-ccb1-4062-8f78-600115a34327:01
lrwxrwxrwx 1 root root 0 Oct 30 14:30 mei::42b3ce2f-bd9f-485a-96ae-26406230b1ff:01 -> ../../../devices/pci0000:00/0000:00:16.0/mei::42b3ce2f-bd9f-485a-96ae-26406230b1ff:01
lrwxrwxrwx 1 root root 0 Oct 30 14:30 mei::4fcc395c-a9e5-4647-bc68-47bad7cc6bd3:01 -> ../../../devices/pci0000:00/0000:00:16.0/mei::4fcc395c-a9e5-4647-bc68-47bad7cc6bd3:01
lrwxrwxrwx 1 root root 0 Oct 30 14:30 mei::55213584-9a29-4916-badf-0fb7ed682aeb:02 -> ../../../devices/pci0000:00/0000:00:16.0/mei::55213584-9a29-4916-badf-0fb7ed682aeb:02
lrwxrwxrwx 1 root root 0 Oct 30 14:30 mei::6861ec7b-d07a-4673-856c-7f22b4d55769:01 -> ../../../devices/pci0000:00/0000:00:16.0/mei::6861ec7b-d07a-4673-856c-7f22b4d55769:01
lrwxrwxrwx 1 root root 0 Oct 30 14:30 mei::8c2f4425-77d6-4755-aca3-891fdbc66a58:01 -> ../../../devices/pci0000:00/0000:00:16.0/mei::8c2f4425-77d6-4755-aca3-891fdbc66a58:01
lrwxrwxrwx 1 root root 0 Oct 30 14:30 mei::8e6a6715-9abc-4043-88ef-9e39c6f63e0f:02 -> ../../../devices/pci0000:00/0000:00:16.0/mei::8e6a6715-9abc-4043-88ef-9e39c6f63e0f:02
lrwxrwxrwx 1 root root 0 Oct 30 14:14 mei::b638ab7e-94e2-4ea2-a552-d1c54b627f04:01 -> ../../../devices/pci0000:00/0000:00:16.0/mei::b638ab7e-94e2-4ea2-a552-d1c54b627f04:01
lrwxrwxrwx 1 root root 0 Oct 30 14:30 mei::dd17041c-09ea-4b17-a271-5b989867ec65:01 -> ../../../devices/pci0000:00/0000:00:16.0/mei::dd17041c-09ea-4b17-a271-5b989867ec65:01
lrwxrwxrwx 1 root root 0 Oct 30 14:30 mei::f908627d-13bf-4a04-b91f-a64e9245323d:01 -> ../../../devices/pci0000:00/0000:00:16.0/mei::f908627d-13bf-4a04-b91f-a64e9245323d:01

StollD, 2019-10-30:
Ok, I forgot that the kernel mei module needs to know about the PCI device first (in fact IPTS still patches in the device for surface because upstream doesn't have it).

You can either recompile the kernel to add the IDs (I will prepare a patch for that later), but for the moment you can bind the driver to the device manually:

```bash
$ echo '0000:00:16.4' | sudo tee /sys/module/mei_me/driver/pci:mei_me/bind
```
After that please rerun `ls -l /sys/bus/mei/devices/` and post the output. If we are lucky, the MEI ID for IPTS hasn't changed (just the PCI ID), and the IPTS driver would probe without modifications.

aleksfadini, 2019-10-30:
]$ echo '0000:00:16.4' | sudo tee /sys/module/mei_me/driver/pci:mei_me/bind
[sudo] password for aleks:
tee: '/sys/module/mei_me/driver/pci:mei_me/bind': No such file or directory
0000:00:16.4

Does that mean I'm not lucky?

StollD, 2019-10-30:
No, it means I am too stupid to type correctly. Please try again with `drivers` instead of `driver`. Oops.

aleksfadini, 2019-10-30:
No, it means I don't know much about Linux and thankfully we have people like you.

That folder didn't seem to work either, I tried this:

$ sudo echo '0000:00:16.4' | sudo tee /sys/module/mei_me/drivers/pci\:mei_me/module/drivers/bind
tee: '/sys/module/mei_me/drivers/pci:mei_me/module/drivers/bind': Permission denied
0000:00:16.4

StollD, 2019-10-30:
`$ sudo echo '0000:00:16.4' | sudo tee /sys/module/mei_me/drivers/pci:mei_me/module/drivers/bind` you have a duplicated `/module/drivers` in there. 

The full command should be `echo '0000:00:16.4' | sudo tee /sys/module/mei_me/drivers/pci:mei_me/bind`. I should have just posted that again, earlier, sorry.

aleksfadini, 2019-10-30:
Well, I tried that after your second message about replacing driver with drivers and I got this:
$ echo '0000:00:16.4' | sudo tee /sys/module/mei_me/drivers/pci:mei_me/bind
0000:00:16.4
tee: '/sys/module/mei_me/drivers/pci:mei_me/bind': No such device

aleksfadini, 2019-10-30:
I guess THAT means that I'm not lucky!

StollD, 2019-10-30:
Ok, that is a bit more clear. Can you post the result of `ls -l /sys/bus/pci/devices`? I copied that device ID from my surface, maybe there are subtile differences.

Also, are you on IRC? If yes, we should continue this there (`freenode/##linux-surface`). I don't really want to spam qzeds repo with this.

Hugal31, 2019-10-31:
There are a lot of results in the registers for itouch, mainly in HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Classes\Interface\{Some IDs}.
I couldn't find anything interesting, nor the ID 745a17a0-74d3-11d0-b6fe-00a0c90f57da (the touch screen ID)

qzed, 2019-12-06:
The battery should now work. The patches are included in the latest linux-surface releases.

As the touchscreen is not related to the surface-acpi and there's some valuable info in this issue, I'm going to move it to the linux-surface-kernel repo.