linux-surface/surface-aggregator-module #25: Strange trackpad behaviour on Surface Laptop 3 13"

qw-in, 2019-12-27:
First off, thanks very much for all your work on supporting these devices, much appreciated! :pray: 

### Summary

I am having some very strange trackpad behaviour, guessing it is related to performance modes

1. Trackpad is extrememly jumpy under certain circumstances
    1. Fine when AC power is connected
    1. Fine when a logitech bluetooth mouse usb dongle is connected
        - Even when the mouse is off
    1. Fine when cpu is under load (all cores seems to be required)
1. Whenever headphones are plugged in and playing audio, fan ramps up and stays up until audio stops playing (potentially unrelated)

### Stopgap solution

Leave a wireless mouse dongle plugged in at all times lol

### Details
- Surface laptop 3 13"
- [Manjaro 5.4.2-1](https://gitlab.manjaro.org/packages/core/linux54) with [archseers patch subset](https://github.com/archseer/arch-linux-surface)

``

### Tried
1. All different combinations of Xorg / Wayland and libinput / synaptics
    - No effect, still laggy. Synaptics is the least bad, but still the same symptoms
1. Setting the power mode to `4`, as outlined in #18 
    - (python script does not print anything when executed)
1. Different kernels (5.4.1, 5.5rc)
    - Same behaviour on all others
    - 5.5 was not usable for other reasons, may try again in future and will update

qw-in, 2019-12-27:
It is due to [tlp](https://wiki.archlinux.org/index.php/TLP) :slightly_frowning_face:

closed 2019-12-27 by qw-in

qw-in, 2019-12-27:
If anyone else runs into this, here is the fix:

edit `/etc/default/tlp` an update the following lines

```
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=on

RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau nvidia pcieport radeon usbhid hid_multitouch hid_generic surface_sam"
```

qzed, 2019-12-27:
Awesome that you've figured that out! Another thing that might cause some lag is the debug output produced by the module (currently that is enabled by default, and the touchpad provides quite a lot of that). I guess I should change that, as that was originally only intended for development and debugging.

archseer, 2020-01-01:
I noted this on my arch install guide, you need to blacklist `"PCI Device Intel Corporation Ice Lake-LP USB 3.1 xHCI Host"` or it will start to drop SSH events and get jumpy. Will add it to the device quirks page.

gitmirko, 2020-10-19:
> I noted this on my arch install guide, you need to blacklist `"PCI Device Intel Corporation Ice Lake-LP USB 3.1 xHCI Host"` or it will start to drop SSH events and get jumpy. Will add it to the device quirks page.

@archseer  could you please advice how to do this? I will update the wiki to make this part clearer, but right now I didn't understand how to blacklist what you stated above.

Mirko