linux-surface/surface-aggregator-module #29: Power event 0x53

danielzgtg, 2019-12-29:
On my Surface Pro 7, dmesg is flooded with the following lines:

```
surface_sam_sid_power: power event (cid = 0x53)
surface_sam_sid_power: unhandled power event (cid = 0x53)
```

fxgsell, 2019-12-30:
I also see a handful of those on the Laptop3.

archseer, 2019-12-30:
We think it might be equivalent to https://github.com/linux-surface/surface-aggregator-module/blob/3a34bb8c7b4c14ff84283b8b2e0188268b199fcc/doc/requests.txt#L85 on gen7 (Surface Pro 7, Surface Laptop 3)

I have a raw dump of the event, but we weren't quite able to figure out the payload:

```
surface_sam_ssh serial0-0: recv: received buffer (size: 39)
recv: 00000000: aa 55 80 1d 00 ab 4b 5c 80 02 00 01 01 02 00 53
recv: 00000010: 01 ba 85 01 00 86 23 01 00 65 00 21 1d 00 00 00
recv: 00000020: 00 c8 32 00 00 55 e1
surface_sam: power event (cid = 0x53)
surface_sam: unhandled power event (cid = 0x53)
```

qzed, 2020-04-22:
Based on some reverse-engineering, the payload has the form `8bit (unused), 32bit, 32bit, 16bit, 16bit, 32bit, 32bit`. Converted to integers, this yields `1, 99770, 74630, 101, 7457, 0, 13000`.

@archseer: Do you see any connection to other battery values?

archseer, 2020-04-22:
@qzed I went and recorded some more event values. This time I logged only the `pld` buffer value:

https://gist.github.com/archseer/dc3b41a378083a6d2a6205bf1d4f8d6c

Sidenote: I seem to have gotten a few warnings on `surface_sam_ssh`, but I think that's during me live reloading the module:

```
event: unhandled event (rqid: 0015)
event: unhandled event (rqid: 0015)
event: unhandled event (rqid: 0015)
```

qzed, 2020-04-22:
Here's that decoded into numbers:
```
[(1, 108360, 76500, 99, 7479, 0, 13000),
 (1, 107340, 76260, 99, 7438, 0, 13000),
 (1, 108460, 76520, 99, 7414, 0, 13000),
 (1, 109480, 76760, 99, 7671, 60000, 13000),
 (1, 110500, 77000, 99, 7686, 60000, 13000),
 (1, 111730, 77290, 99, 7706, 60000, 13000),
 (1, 112750, 77530, 99, 7735, 60000, 13000),
 (1, 113760, 77770, 99, 7766, 60000, 13000)]
```

So looks like the 6th value has to do something with charging... Could that be a 60W charger? I think the 5th could maybe be voltage (in mV). The first value could be a status thing. As far as I can see this gets ignored in the Windows driver. I suppose you don't have any clue what `TpUpdate` means either?

qzed, 2020-04-24:
Oh, I forgot to answer to the 0x0015 events during reloading: Those are input events (TC=0x15). They should _normally_ get disabled when the sid_vhf module is unloaded. I suspect that those are handled a bit differently and cannot actually be disabled, which would kind-of make sense given their nature.

larsschellhas, 2026-11-15:
I ran into `0x53` while working on Surface power management on Linux (linux-surface/linux-surface#2383), and I think the payload can now be named. The decode above (u8 + u32 + u32 + u16 + u16 + u32 + u32, little endian) matches what I see on a Surface Book 3 15".

| Offset | Meaning | Unit | SB3, display part (SAM) | SB3, base (KIP) |
|---|---|---|---|---|
| @0 | version | – | 1 | 1 |
| @1 | battery peak power | mW | 47 780 | 157 904 |
| @5 | battery sustained power | mW | 35 510 | 84 880 |
| @9 | battery internal resistance | mΩ | 168 | 123 |
| @11 | battery no-load voltage | mV | 8 111 | 12 488 |
| @13 | available input power | mW | 45 720 on battery, 73 494 with the Surface PSU | 65 000 USB-C only, 119 250 with both |
| @17 | rest-of-platform power | mW | 9 700 | 13 000 / 28 000 |

How I got there:
- **No-load voltage (@11):** confirmed against `voltage_now` of the matching battery.
- **Available input power (@13):** the only field that follows the power source. It's the 0 → 60000 step in the log above.
- **The other names:** taken from how Windows uses the values. The Surface SMF power management computes psys from them. For example, psys PL1 = available input power + 23 W (capped at 45 W) on the SB3. That's why psys never binds on Windows, while Linux stays at the 25/40 W boot value (linux-surface/kernel#117). I'm least sure about @17.

On the SB3, the event only comes on power-source changes (after BAT `0x17`), once from SAM and once or twice from KIP. The same data can be requested at any time: TC `0x02`, CID `0x53`, IID `0x01`, with response, sent to SAM (TID 1) and KIP (TID 2).

For now my user-space prototype requests this through the debug cdev, which needs root. A small client driver that handles the event, and requests the data once at probe, would make it usable without the cdev. It could expose the values as read-only sysfs or hwmon power attributes.

Would that be welcome, and is there a preferred place for it (battery driver, a separate driver)? I'd be glad to help with it and test on the SB3.

qzed, 2026-11-15:
> Would that be welcome, and is there a preferred place for it (battery driver, a separate driver)? I'd be glad to help with it and test on the SB3.

Welcome definitely. How to structure the driver and everything around it is a good question. I think mostly what you opened as PR and proposed in https://github.com/linux-surface/linux-surface/issues/2383 makes sense. So exposing them via the battery driver and then a separate user-space daemon. I'm not sure yet how to handle and expose all of this (e.g. whether to just always show the attributes or whether to only show them if they are actually supported) and how the detection should work. Normally, if you just query stuff that is not supported on the EC and expect an answer, you'll run into a timeout. So

https://github.com/linux-surface/kernel/pull/195/changes#diff-cad12d0907d58e06f82f7c1bfd0a42018d10ffcc4349f08b37885b124736effaR265-R266

is probably not something that would work well across all devices. We could use device properties, like we have for the fan profile. That requires more manual specification, but if there's not a clean way to query functionality from the EC I think that's all we have.

https://github.com/linux-surface/kernel/blob/521f20ef604adbb6d3e3dd025c76c85bb9088a8c/drivers/platform/surface/surface_aggregator_registry.c#L78-L90

larsschellhas, 2026-11-16:
Thanks, agreed. Probing by request is gone; #195 now uses a registry property instead, like `has_fan`.

- `has_dptf` on the battery and AC nodes. Without it, no 0x53 or adapter-rating request is sent and the attributes are hidden.
- Set for: SB3, SL3 15" AMD, SL4, SL5, SL6, SLG1–3, SLS1/2, SP7+, SP8, SP9, SP9 5G, SP10, SP11 (Intel and ARM). Only the SB3 is measured; for the others the Windows power config of the device uses this data. Not set (no evidence): SL3 13" Intel, SP7, SP12 Intel. That needed new groups `sl3_dptf`, `sp7_dptf`, `sp9_dptf`.
- Tested on the SB3 (7.3.4): values as before, PSU unplug/replug tracked, and 60 s idle with tracepoints: no 0x53 request, no timeout.
- Side note: `adapter_rating_mw` queries the EC on every read. I'll read it less often in the daemon; the driver could cache it if you prefer.

Questions: is the name and placement OK? Splitting the groups, or rather set it for the whole group? Hide the attributes without the property (as now)?

qzed, 2026-11-16:
> Side note: adapter_rating_mw queries the EC on every read. I'll read it less often in the daemon; the driver could cache it if you prefer.

I think querying it every read should be fine, at least I don't think it should overwhelm the EC. Hiding unsupported attributes as done now makes sense to me. I'll have a look at the rest now.