linux-surface/surface-aggregator-module #33: [ACPI][Surface Pro 5/2017] Battery POWER_NOW reports wrong value

tarmack, 2020-01-14:
The value reported by the acpi driver for the power draw is incorrect (or at least rather high, tens of watts regularly).

Looking at the surface_sam_ssd code and the acpi code I can't find where these values even get evaluated. Am I correct in assuming that these values go straight from the hardware (ssd message) into the kernel ACPI structures? Because in that case I assume that the hardware is not giving useful information and it may be useful to try and correct the value. Is there any knowledge available on how windows handles this data? (I guess not).

qzed, 2020-01-14:
Moved it to the SAM repo as that's an issue related to this module.

tarmack, 2020-01-16:
I did some collecting of data from _BST and can confirm that the numbers returned are completely random. Ten different readings in one second give ten completely different results even though the _BIF reports a 1000ms averaging period.

I think it is safe to conclude that the problem is with the firmware and there is nothing that the kernel module can do to fix this.

qzed, 2020-01-16:
Can you post some of the _BST and one _BIX output? It's also just the power_now reading that's bad? I'd be interested to look at it and check if there's something that could hint to those values being invalid or something.

tarmack, 2020-01-16:
It is indeed just the `power_now` that is bad, all other values seem valid and correct.

_BIX:
```
00 00 00 00 00 c8 af 00 00 2e bd 00 00 01 00 00 00 92 1d 00 00 5e 1a 00 00 46 05 00 00 06 00 00 00 e8 03 00 00 ff ff ff ff ff ff ff ff e8 03 00 00 e8 03 00 00 0a 00 00 00 0a 00 00 00 4d 31 30 30 39 31 36 39 00 00 00 00 00 00 00 00 00 00 00 00 00 35 32 35 36 34 37 36 39 31 39 00 4c 49 4f 4e 00 53 4d 50 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

_BST:
#> while sleep 0.1; do sudo python sysfs_rqst.py; done
```
01 00 00 00 61 09 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 93 ef 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 47 68 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 cd 52 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 f5 06 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 a1 70 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 03 d2 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 82 eb 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 72 4b 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 ad ac 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 b3 cc 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 66 1c 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 c1 4a 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 b2 93 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 21 86 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 b5 c7 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 2d 2d 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 c7 9d 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 89 99 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 7c bc 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 56 ef 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 d6 0c 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 0f c3 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 01 a4 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 29 d2 00 00 e4 75 00 00 b4 1e 00 00
01 00 00 00 43 2c 00 00 e4 75 00 00 b4 1e 00 00
```

qzed, 2020-01-17:
Neat, thanks!

qzed, 2020-01-20:
There is indeed no indication anywhere else that those values are not trustworthy.

We could implement a quirk for the SP2017 setting the power_now for the device to `0xFFFFFFFF` (meaning unsupported) which should be handled by the ACPI battery driver accordingly, but that would require us intervening in the SAN driver, which I don't particularly like. That would also block out the readings in case MS decides to fix that in firmware until we revert that change. I'm also not sure if such a quirk shouldn't rather be implemented in the ACPI battery in a more generic way, as there might be other devices with faulty battery readings.

Given that it doesn't break anything in user-space I'd like to keep it as is for now, unless there is some major issue caused by this.

closed 2020-01-20 by qzed