linux-surface/surface-aggregator-module #34: DGPU auto turns on itself during sleep.

StealthBadger747, 2020-02-05:
I also have a power drain issue while sleeping. Although this could be unrelated.

qzed, 2020-02-05:
This likely happens because something wakes the device and suspends it again instantly. During resume, the PCIe core wakes up the device. On a full wake-up/resume process, the SAM dGPU driver will then suspend the device again. However, when the wake-up process is aborted after the dGPU has been turned on by PCIe-core and the device is suspended again (as may happen during spurious wake-up interrupts), the wakeup-handler for the SAM dGPU driver will not be executed. Thus the device stays on.

I don't have any control over this as the Linux driver infrastructure takes care of these calls, so there's not really anything that I can do here (apart from heavily modifying PCIe core/hot-plug drivers, which I'm not too keen on and will probably result in more of a hack than a driver that could be pursued upstream).

To avoid this you could have a look at your dmesg logs and try to figure out what causes those spurious wakes.

qzed, 2020-07-24:
Closing this as I don't see a way to reliably fix that right now.

closed 2020-07-24 by qzed