linux-surface/surface-aggregator-module #36: the command surface dgpu set on doesn't actually turn on the gpu.

BalenCPP, 2020-02-25:
So I have the linux-surface 5.5.4-1 installed, along with nvidia-dkms. I use surface-control to turn on the dGPU but when I try to switch to the dgpu using optimus-manager, it doesn't work. I thought that maybe the dGPU isn't on and so when I do lspci | grep 'VGA', it only prints `
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
00:13.0 Non-VGA unclassified device: Intel Corporation Sunrise Point-LP Integrated Sensor Hub (rev 21)
`

BalenCPP, 2020-02-25:
> Is there anything in the dmesg log when you turn it on/off? Can you verify that this is happening without invoking any other programs that may change the dGPU power?

Setting it off and then on again produces this:
 

> [ 1433.029487] surface_dgpu_hps MSHW0153:00: setting dGPU power state to 'off'
> [ 1433.029526] surface_dgpu_hps MSHW0153:00: setting dGPU direct power to 'off'
> [ 1462.824139] surface_dgpu_hps MSHW0153:00: setting dGPU power state to 'on'

which I am guessing is normal behavior.

I also have optimus-manager removed and uninstalled.

qzed, 2020-02-26:
Given that `nvidia-smi` works and shows you the device, the dGPU seems to work properly. `nvidia-smi` working also means that the driver loads and works. Did you filter the `lspci` output in any way?.

BalenCPP, 2020-02-26:
02:00.0 3D controller: NVIDIA Corporation GP106M [GeForce GTX 1060 Mobile] (rev a1)
is found when I do lspci. 
However there's also "OpenGL renderer string: Mesa DRI Intel(R) UHD Graphics 620 (Kabylake GT2)" so I am guessing there's something wrong with the arch linux configuration and not the surface aggregator module.

qzed, 2020-02-26:
Yeah, that looks like a configuration issue then. You can try and run `glxgears` via `optirun` or `pvkrun` (or the likes). That should work and show the correct renderer.

>  I still don't know why the dGPU is generating heat when it's not being used.

That seems to be normal. It's the reason why I turned it off by default.

BalenCPP, 2020-02-26:
> Yeah, that looks like a configuration issue then. You can try and run `glxgears` via `optirun` or `pvkrun` (or the likes). That should work and show the correct renderer.
>
>
> That seems to be normal. [...]

What about running everything through the dGPU? I followed this section of the [arch linux wiki](https://wiki.archlinux.org/index.php/NVIDIA_Optimus#Use_NVIDIA_graphics_only) but it's of no use.

qzed, 2020-02-26:
If you want to run everything via the dGPU you'll either have to enable it at start-up by setting the module parameters accordingly (https://github.com/linux-surface/surface-aggregator-module#via-module-parameters) or manually enable it and then start a new xorg session on the dGPU (key is you have to restart xorg after turning the dGPU on, otherwise it won't be recognized).

I'm not sure how to do that as I haven't tried it yet, but it should be possible. In any case, getting xorg to do the right stuff will require some configuration.

BalenCPP, 2020-02-26:
I already have a file under /etc/modprobes.d which contains "options surface_sam_hps dgpu_power_init=1". That ought to do it, right?

BalenCPP, 2020-02-26:
outouIt's all working quite good now, with the following file content /etc/X11/xorg.conf.d/10-nvidia-drm-utputclass.xonf:

>Section "ServerLayout"
  >  Identifier "layout"
    Option "AllowNVIDIAGPUScreens"
EndSection
Section "OutputClass"
    Identifier "intel"
    MatchDriver "i915"
    Driver "modesetting"
EndSection
Section "OutputClass"
    Identifier "nvidia"
    MatchDriver "nvidia-drm"
    Driver "nvidia"
    Option "AllowEmptyInitialConfiguration"
    Option "PrimaryGPU" "yes"
    ModulePath "/usr/lib/nvidia/xorg"
    ModulePath "/usr/lib/xorg/modules"
EndSection

Thanks a lot @qzed

closed 2020-02-26 by BalenCPP

qzed, 2020-02-26:
> I already have a file under /etc/modprobe.d which contains "options surface_sam_hps dgpu_power_init=1". That ought to do it, right?

Yes.

> It's all working quite good now, with the following file content /etc/X11/xorg.conf.d/10-nvidia-drm-utputclass.xonf

Nice!