linux-surface/surface-aggregator-module #37: Issue with dgpu control on latest kernel

Doridian, 2020-03-13:
I recently installed Fedora 31 on my Surface Book 2 15".
After booting into the linux-surface kernel, most of the surface commands (as expected) work. However the dgpu control just gives me the following error:

```
[root@capefox 5.5.8-1.surface.fc31.x86_64]# surface dgpu get
Error: Cannot access device.
       No such file or directory (os error 2).
```

```
[root@capefox 5.5.8-1.surface.fc31.x86_64]# surface performance get
Performance-mode is 'Best Performance' (4)
[root@capefox 5.5.8-1.surface.fc31.x86_64]# surface status
       Device-Mode: Laptop
  Performance-Mode: Best Performance (4)
```

Looking at the repo, this seems to used to have been a seperate module, but that GitHub page now states it is completely integrated with the linux-surface project? Do I need to `modprobe` a certain module?

qzed, 2020-03-13:
Okay, so all in all this looks like an issue with the module, not the command line utility, so I'm moving the issue. All in all it seems like the hps/dgpu module isn't loading (error -38/ENOSYS).

Apart from that, I haven't been able to track down what exactly causes this. It looks like this could come either from the PCI subsystem or from the IRQ subsystem. If you have some experience with C you can try to track this down a bit more by compiling (run `make` inside the `module` directory) the module yourself with some basic printk statements. It you want to do that you'll need to unload the modules via [`scripts/unload.sh`](https://github.com/linux-surface/surface-aggregator-module/blob/master/scripts/unload.sh) and then manually load the compiled module via `sudo insmod surface_sam.ko`.

qzed, 2020-03-13:
Also did previous kernels work?

Doridian, 2020-03-13:
@qzed Well, sadly I cannot tell you that sadly as I just swapped to Linux on my SB2, I ran Win10 before that.

qzed, 2020-03-13:
@Doridian you could try and manually install a release from https://github.com/linux-surface/linux-surface/releases, e.g. https://github.com/linux-surface/linux-surface/releases/tag/fedora-31-5.5.6. Although I suspect that that doesn't make a difference. There haven't been any changes in the linux-surface patches so it'd have to be something that changed in upstream.

qzed, 2020-03-13:
Regarding the signing, it's probably easier if you temporarily disable secure-boot. You also don't have to install the module and reboot every time, unloading the builtin and loading the externally compiled one should work just fine. You just have to unload and re-load the compiled module after that.

Doridian, 2020-03-13:
@qzed Honestly, I'm not too worried. I will just write a couple line bash script that unload,compile,sign,load and then I can rapidly make changes if needed.

Doridian, 2020-03-13:
After adding quite a few `printk` statements, I have narrowed the error down to the following

```
printk("3\n");

        // export GPIOs
        status = gpiod_export(gpio_dgpu_power, false);
        if (status)
                goto err_out;

printk("4\n");
```

It prints 3, but does not print 4, so the `gpiod_export` must throw `ENOSYS`

qzed, 2020-03-13:
Thanks! I've tracked it down (https://elixir.bootlin.com/linux/v5.5.8/source/include/linux/gpio/consumer.h#L715): The default Fedora config does not set `CONFIG_GPIO_SYSFS`, which is required for `gpiod_export` to actually be implemented. If not implemented, it returns `-ENOSYS`, which is what we see here. I'll update the kernel config trigger a new build.

qzed, 2020-03-13:
Also I should add that symbol to the module dependencies, so let's keep this issue open until that's done.

qzed, 2020-03-13:
New Fedora builds are in flight, version 5.5.9-2 should fix the issue.

Doridian, 2020-03-13:
Well, it seems the F31 build failed in some build step. I'll test it as soon as that's fixed. Thanks for fixing this so fast.

qzed, 2020-03-13:
Yeah that was some spurious error during upload. Happens from time to time, no idea what causes that. I've restarted the build.

Doridian, 2020-03-13:
Fix confirmed. After updating to 5.5.9-2 I can now control the dGPU power state successfully :)

qzed, 2020-03-13:
Thanks for confirming!

qzed, 2020-03-16:
Module dependencies fixed via 3013df5b6b0e9e163964c2f2dc48e9cb47fd6f59.

closed 2020-03-16 by qzed