linux-surface/surface-aggregator-module #38: surface_sam: hps: using result of pci_enable_device() ?

kitakar5525, 2020-05-13:
When building with CONFIG_ENABLE_MUST_CHECK=y (y is the default value
but some distributions don't set), the compiler gives the following
warnings:

    drivers/platform/x86/surface_sam/surface_sam_hps.c: In function ‘__shps_dgpu_rp_set_power_unlocked’:
    drivers/platform/x86/surface_sam/surface_sam_hps.c:378:4: warning: ignoring return value of ‘pci_enable_device’, declared with attribute warn_unused_result [-Wunused-result]
      378 |    pci_enable_device(rp);
          |    ^~~~~~~~~~~~~~~~~~~~~
    drivers/platform/x86/surface_sam/surface_sam_hps.c: In function ‘shps_dgpu_powered_on’:
    drivers/platform/x86/surface_sam/surface_sam_hps.c:782:3: warning: ignoring return value of ‘pci_enable_device’, declared with attribute warn_unused_result [-Wunused-result]
      782 |   pci_enable_device(rp);
          |   ^~~~~~~~~~~~~~~~~~~~~

Looking at the other usages of pci_enable_device() in the Linux kernel,
many callers use the result. So, let's use the result here, too, like this?

```diff
diff --git a/drivers/platform/x86/surface_sam/surface_sam_hps.c b/drivers/platform/x86/surface_sam/surface_sam_hps.c
index 4fba5ee75a66..b5bd933f57e1 100644
--- a/drivers/platform/x86/surface_sam/surface_sam_hps.c
+++ b/drivers/platform/x86/surface_sam/surface_sam_hps.c
@@ -374,8 +374,11 @@ static int __shps_dgpu_rp_set_power_unlocked(struct platform_device *pdev, enum
 
                pci_restore_state(rp);
 
-               if (!pci_is_enabled(rp))
-                       pci_enable_device(rp);
+               if (!pci_is_enabled(rp)) {
+                       status = pci_enable_device(rp);
+                       if (status)
+                               return status;
+               }
 
                pci_set_master(rp);
                clear_bit(SHPS_STATE_BIT_RPPWRON_SYNC, &drvdata->state);
@@ -778,8 +781,11 @@ static int shps_dgpu_powered_on(struct platform_device *pdev)
        if (drvdata->dgpu_root_port_state)
                pci_load_and_free_saved_state(rp, &drvdata->dgpu_root_port_state);
        pci_restore_state(rp);
-       if (!pci_is_enabled(rp))
-               pci_enable_device(rp);
+       if (!pci_is_enabled(rp)) {
+               status = pci_enable_device(rp);
+               if (status)
+                       return status;
+       }
        pci_set_master(rp);
        dbg_dump_drvsta(pdev, "shps_dgpu_powered_on.2");
        dbg_dump_power_states(pdev, "shps_dgpu_powered_on.2");
```

That said, I'm not sure if returning the functions without doing nothing
is OK or not (error handlings? printing messages?). So, I'm glad if you can
take a look at this.

kitakar5525, 2020-05-13:
Ah, one of the result is explicitly ignored by commit db21c72f47bb04. Is there reason for ignoring?

qzed, 2020-05-13:
Commit db21c72 was before I added the `pci_is_enabled` check. Unfortunately I don't remember exactly, but I think without the check for `pci_is_enabled`, there were some issues. I believe they should have been resolved by the check. I'll have to test this, thanks for reporting!

kitakar5525, 2020-05-14:
OK, thanks!

qzed, 2020-11-03:
Okay, so turns out I can drop most of the HPS code by fixing issues that prevent proper D3cold support and then relying on the PCI/PM core to do its job (and also runtime PM). That code is currently in the `cleanup` branch and I want to do a bit more testing on that, but ultimately it's going to resolve this issue by simply dropping the offending code.

closed 2020-11-03 by qzed

kitakar5525, 2020-12-01:
Thanks! now there are no warnings when `ONFIG_ENABLE_MUST_CHECK=y`.