linux-surface/surface-aggregator-module #41: Support for SB3 dGPU power setting.

qzed, 2020-06-11:
This seems to be fairly similar to the SB2, although it looks like there are some subtle differences.

Previous discussion: https://github.com/linux-surface/linux-surface/issues/201#issuecomment-642029552

steveniemitz, 2020-06-15:
hmm, in other news, I've somehow managed to permanently disable the dGPU, even in windows.  Is there any way from windows to re-enable it?  (It's enabled in the UEFI settings).

steveniemitz, 2020-06-15:
hard coding the PCI parameters got me much further:
```
	bus = 0;
	dev = 29; 
	fun = 4; 
```

now that I know what I'm looking for, I'm going to see if I can find this in the ACPI tables.

qzed, 2020-06-15:
Usually, the two-button reset can fix things. That should reset any firmware state.

steveniemitz, 2020-06-15:
> Usually, the two-button reset can fix things. That should reset any firmware state.

heh I didn't even know that existed, good to know now!

Random question, on the SB2, with the GPU off, does `lspci` still show it?

qzed, 2020-06-15:
> Random question, on the SB2, with the GPU off, does `lspci` still show it?

If it's completely off, no. When I re-worked the power-setting stuff, I briefly tried finding the PCIe root port via the dGPU PCIe device by first looking that and then hopping along the device hierarchy. Problem is that if it's off (via the ACPI method) then it's not visible. That was before I found the ACPI entry.

You should keep in mind when looking for that value in ACPI, that it may be encoded:
https://github.com/linux-surface/surface-aggregator-module/blob/d2f9b60a9a54f6cb02aad3f7137c3032417fdccf/module/surface_sam_hps.c#L220-L223

steveniemitz, 2020-06-15:
hmm, yeah, I guess I'm not completely shutting it down then, since I still see it in `lspci`.  Making progress though!

My current testing process to see if it's actually off is "does the base get hot or not"...

qzed, 2020-06-15:
Hmm, try going with `lspci -v`. I remember that there might be some state update issues. Also I believe dmesg should show messages when the dGPU gets turned off (something regarding hotplug?, I'm really not sure any more, I should probably check this).

steveniemitz, 2020-06-15:
ah, awesome, ok I think it is working.  I got the gpios working now as well so I can turn the GPU on/off via `dgpu_power` (or the `surface` command).

I found the PCI bus info in ACPI as well:
```
        Device (RP13)
        {
            <snip>
            Method (_ADR, 0, NotSerialized)  // _ADR: Address
            {
                If ((RPAD != Zero))
                {
                    Return (RPAD) /* \RPAD */
                }
                Else
                {
                    Return (0x001D0004)
                }
            }
```

so that should be easy to hook up as well.

qzed, 2020-06-15:
That's the basic ACPI definition for a PCI device, so we'd still have to hard-code the `RP13`, unless that's linked somewhere.

steveniemitz, 2020-06-15:
Yeah, the old driver hard coded RP5 here.  I guess we can select the right one based on the hardware info?

steveniemitz, 2020-06-15:
I've pushed my hacking so far into a fork:
https://github.com/steveniemitz/surface-aggregator-module/commits/feature/sb3-v2

I haven't gotten the IRQs working yet.  I just need those and events from SSH to work and it should be good.  Then we just need to figure out how to encapsulate the device specific differences.

qzed, 2020-06-15:
Guess we won't get around hard-coding it... although the RP5 hard-code was for a key in a map/table. This table can be obtained from the hot-plug device itself and maybe the hard-code could have even been dropped if we knew the table format better.

I wonder if it would actually make a difference if we're hard-coding `RP13` vs. the root port address directly. If there's a change to one, I'd expect that there's also a change to the other.

qzed, 2020-06-15:
The SAN notifications seem to have been replaced via a new ACPI device: https://github.com/linux-surface/acpidumps/blob/f9998ef39ec581e731d17b5cb2bfddb1d87b9ba6/surface_book_3/dsdt.dsl#L26807

qzed, 2020-06-15:
> I've pushed my hacking so far into a fork:
> https://github.com/steveniemitz/surface-aggregator-module/commits/feature/sb3-v2

Neat!

steveniemitz, 2020-06-15:
hmm, so I got the GPIO IRQs working by removing the triggering parameters from the IRQ request

```
-       const int irqf_base = IRQF_SHARED | IRQF_TRIGGER_RISING | IRQF_TRIGGER_FALLING;
+       const int irqf_base = IRQF_SHARED;
```

This should mean "as configured".  I'm a little confused why it didn't work as-is though, from what I can tell the configuration shouldn't have changed between the SB2 and 3.  Any reason the mode was explicitly set there?

qzed, 2020-06-15:
Hmm, I think setting that specifically might be remnants from me testing some things. It's probably not required. Weird that specifying those causes problems. Especially since `GpioInt (Edge, ActiveBoth, ...` (from ACPI) basically defines them as triggering on both edges.

steveniemitz, 2020-06-15:
> Especially since ...

Yeah, that's why I was confused too, it _SHOULD_ work, but doesn't (and the ACPI data hasn't changed between SB2 -> 3).  Maybe a quirk of the GPIO setup on SB3?  I was getting this error:

```
Jun 15 17:05:11 steve-work-sb kernel: [ 2136.071234] icelake-pinctrl INT3455:00: pin 115 cannot be used as IRQ
```

qzed, 2020-06-15:
Could be an issue in the pinctrl driver. The SB2 should use a different one.

steveniemitz, 2020-06-15:
hmm I guess I'm not quite there yet, the gpu driver fails with:

```
nouveau: probe of 0000:02:00.0 failed with error -17
```

on power on.  I think it's because the power-on interrupt handler isn't firing (as we discussed above).

qzed, 2020-06-15:
Hmm, to be fair, I've only really tested everything with the nvidia driver. I've always had some issues with nouveau, especially when detaching.

steveniemitz, 2020-06-15:
oh hm, interesting.  I want to try to get the ACPI notifications set up, if that still doesn't work after that I'll try the nvidia driver instead.

qzed, 2020-06-15:
No clue how to do that properly with the new setup. I think this would require an ACPI driver for the new device (MSHW0216/SGCP) which would have to talk to the HPS driver, basically calling [`shps_dgpu_powered_on`](https://github.com/linux-surface/surface-aggregator-modulehttps://github.com/linux-surface/surface-aggregator-module/blob/d2f9b60a9a54f6cb02aad3f7137c3032417fdccf/module/surface_sam_hps.c#L740-L755).

At least on the 0x81 / Information Change. There seems to be an additional 0x80 / Status Change event that doesn't have a RQSG equivalent on the SB2.

steveniemitz, 2020-06-15:
+1, that's what I was thinking too.

steveniemitz, 2020-06-15:
neat:
```
Jun 15 19:53:27 steve-work-sb kernel: [  169.996233] surface_dgpu_hps MSHW0153:00: setting dGPU power state to 'off'
Jun 15 19:53:27 steve-work-sb kernel: [  169.996294] surface_dgpu_hps MSHW0153:00: setting dGPU direct power to 'off'
Jun 15 19:53:27 steve-work-sb kernel: [  170.049359] pcieport 0000:00:1d.4: pciehp: Slot(12): Link Down
Jun 15 19:53:27 steve-work-sb kernel: [  170.049362] pcieport 0000:00:1d.4: pciehp: Slot(12): Card not present
*Jun 15 19:53:27 steve-work-sb kernel: [  170.122212] surface_sam_sgcp MSHW0216:00: sgcp notify 128
```

and 129 for on.  Now just need to hook up the cross module call.

qzed, 2020-06-15:
Looks good!

steveniemitz, 2020-06-16:
I pushed my branch w/ the new driver hooked up.  Everything seem to work now.  I'd call this a very-WIP branch still, but I was able to disable/enable the device successfully and have the nvidia driver recognize it.  I gave up on the nouveau driver though.

Thanks for all the guidance!

steveniemitz, 2020-06-16:
I did a bunch of cleanup this morning on the driver.  I removed the separate sgcp driver and instead just registered for ACPI notifications on `\_SB.SGPC` in the hps driver.  This simplifies things a lot and should also simplify merging it w/ the other SB models.

qzed, 2020-06-16:
Neat!

steveniemitz, 2020-06-18:
hmm, things have been working great so far, with the exception of one bug I just ran into.  After some combination of rebooting into windows and back, the ACPI firmware started timing out trying to turn the GPU off:

```
                                      Initialized Local Variables for Method [WPGS]:
Jun 17 20:29:02 steve-work-sb kernel:   Local0: 00000000f12005e1 <Obj>           Integer 00000000000009BB
Jun 17 20:29:02 steve-work-sb kernel: Initialized Arguments for Method [WPGS]:  (3 arguments defined for method invocation)
Jun 17 20:29:02 steve-work-sb kernel:   Arg0:   000000000c71575b <Obj>           Integer 000000000605000F
Jun 17 20:29:02 steve-work-sb kernel:   Arg1:   00000000959faebc <Obj>           Integer 0000000000000000
Jun 17 20:29:02 steve-work-sb kernel:   Arg2:   000000003263ae30 <Obj>           Integer 0000000000000C80
Jun 17 20:29:02 steve-work-sb kernel: ACPI Error: Aborting method \_SB.PCI0.RP13.WPGS due to previous error (AE_AML_LOOP_TIMEOUT) (20200326/psparse-529)
Jun 17 20:29:02 steve-work-sb kernel: ACPI Error: Aborting method \_SB.PCI0.RP13.HGOF due to previous error (AE_AML_LOOP_TIMEOUT) (20200326/psparse-529)
Jun 17 20:29:02 steve-work-sb kernel: ACPI Error: Aborting method \_SB.SHPS._DSM due to previous error (AE_AML_LOOP_TIMEOUT) (20200326/psparse-529)
Jun 17 20:29:02 steve-work-sb kernel: ACPI: \_SB_.SHPS: failed to evaluate _DSM (0x3021)
Jun 17 20:29:02 steve-work-sb kernel: surface_dgpu_hps: probe of MSHW0153:00 failed with error -5
```

Looking at the DSM, it seems to be polling a system memory address for a value:
https://github.com/linux-surface/acpidumps/blob/f9998ef39ec581e731d17b5cb2bfddb1d87b9ba6/surface_book_3/dsdt.dsl#L4769-L4783

my ACPI knowledge is pretty limited here.  I stepped through it though in `acpiexec` and can see the address evaluates to `0x006B08C0`.  Any thoughts on if I'm on the right track here?

qzed, 2020-06-18:
I tried to figure out what that thing was doing on the SB2 quite some time ago (although I never had any issue like that). IIRC, the `WPGS` function kind of looks like it's waiting for a specific GPIO pin to become set/unset. `GGIV` could mean Get GPIO Input Value. Likely those are represented through MMIO. I think those could be the same as the ones exported via ACPI, i.e. the ones the driver has access to (https://github.com/linux-surface/acpidumps/blob/f9998ef39ec581e731d17b5cb2bfddb1d87b9ba6/surface_book_3/dsdt.dsl#L26859), but I'm not completely sure about that.

steveniemitz, 2020-06-18:
oh hm interesting.  Good theory on the naming too.

I think the issue I ran into is mostly unrelated, I've had a bunch of issues rebooting from windows directly into linux unrelated to the hps also.

Anyways, I pushed a couple more cleanup fixes to my branch which should allow it to be compatible w/ non-SB3 hardware.  Want me to open a pull-request?

Also @NickyTope if you're feel adventurous, want to give it a spin too? (https://github.com/steveniemitz/surface-aggregator-module/tree/feature/sb3-v2)

qzed, 2020-06-19:
> Want me to open a pull-request?

Sure, go ahead. Looks ready for a PR to me. I have one or two small comments but that's easier to do if you open a PR.

steveniemitz, 2020-06-19:
awesome, opened #43. happy to get some comments on this, I haven't written a kernel module for awhile :D

steveniemitz, 2020-06-20:
sooooo my SB3 died yesterday :(  The logo shows up for a second and then it just turns back off.  Really weird, I must have had some bad hardware or something.

I'll be out of the game for a week or so while I RMA it, so sorry if I'm unable to test anything for a few days.

qzed, 2020-06-20:
Sorry to hear that, best of luck with the RMA.

I had some (maybe) similar issues in the past, which I think is connected to the Surface Dock (I think specifically when I shut down Linux while connected to the dock): The device sometimes wouldn't start at the first try. Keyboard would light up, but then go dark again (I don't remember the logo). Normally the second or third try works, but I think I had to two-button-reset the device once. If you haven't tried that yet, maybe that does something.

steveniemitz, 2020-06-21:
oh yeah actually I had a bunch of issues booting with the dock attached.  Sadly that's not the issue here, I tried a bunch of different button combinations, durations, etc, but it seems pretty stuck at this point.

Thanks for the code review comments, I'll go through them in the next day or so.

qzed, 2020-06-21:
> oh yeah actually I had a bunch of issues booting with the dock attached. Sadly that's not the issue here, I tried a bunch of different button combinations, durations, etc, but it seems pretty stuck at this point.

Hmm yeah, that sounds a bit different.

> Thanks for the code review comments, I'll go through them in the next day or so.

No hurry.

qzed, 2020-07-02:
I've merged the PR into the cleanup branch and subsequently merged that into master. Will update the kernel patches next.

closed 2020-07-02 by qzed

steveniemitz, 2020-07-02:
oh cool! so I should start building my modules off that (cleanup) branch now?

qzed, 2020-07-02:
Right now, all the other support stuff is still on the `feature/sb3-v2` branch, so you probably want to stick with that for now. I'll re-implement that on top of the master branch soon-ish.