linux-surface/surface-aggregator-module #42: Support for SB3 keyboard and touchpad (aka generify HID subsystem).

qzed, 2020-06-11:
See discussion on https://github.com/linux-surface/linux-surface/issues/201.

The system is largely similar to the SL3, except that the HID devices are separated via IIDs.

qzed, 2020-06-11:
Depends on #40 for events.

qzed, 2020-06-11:
Looking into the Windows driver, there are a couple of devices that it targets:

```
ACPI\\VEN_MSHW&DEV_0096                                          TC: 0x08  EV: 0x01
{B2C18583-06DF-488E-82E4-C0EF4888A401}\\SHMD_DEVICE              TC: 0x15  EV: 0x0e
{21635d77-3d7c-47b9-be18-26576667bfdc}\\Target_KIP&Category_HID  TC: 0x15  EV: 0x0e
{2DEDC554-A829-42AB-90E9-E4E4B4772981}\\Target_KIP&Category_HID  TC: 0x15  EV: 0x21
{21635d77-3d7c-47b9-be18-26576667bfdc}\\Target_SAM&Category_HID  TC: 0x15  EV: 0x01
```

`TC` is the actual device subsystem, `EV` the event registry subsystem used. The payload across those subsystems is the same. See https://github.com/linux-surface/surface-aggregator-module/issues/40#issuecomment-642880085 for more details on event registration.

qzed, 2020-06-11:
@archseer, @steveniemitz, @NickyTope, if you have some spare time and windows still installed it would be neat to find out which of those devices are actually used and present, and which we'd have to support. I think you should be able to find those via the names above as some property (maybe device name or device path) in the Windows device manager.

The `ACPI\VEN_MSHW&DEV_0096` device is the one used on the SL1/SL2. I think that `{2DEDC554-A829-42AB-90E9-E4E4B4772981}\Target_KIP&Category_HID` should be used on the SB3 based on the testing you've done. This is probably also the device used on the SL3, but that could also use `{21635d77-3d7c-47b9-be18-26576667bfdc}\Target_SAM&Category_HID`. I think it's also possible that multiple of those devices are present and some might have a different use-case (e.g. those with `EV 0x0e`).

@archseer Related to that: Can you check how the events are enabled on the SL3?

Using the script from the `feature/sb3-v2` branch, either
```
./scripts/evreg.py disable 0x02 0x15 0x00 <iid>
```
or
```
./scripts/evreg.py legacy-disable 0x01 0x15 0x00 <iid>
```
should disable (some) input events. On the SL3, `<iid>` should be `0x00`, but if that doesn't work try `0x01` or `0x03`. Replacing `disable`/`legacy-disable` with `enable`/`legacy-enable` should enable them again.

I also wonder if we can separate the events on the SL3 similar as is done on the SB3 by using the right event-enable/disable calls.

NickyTope, 2020-06-14:
There are four devices with id starting 2DED above listed in windows, all called surface Hid Mini Driver.

had to google a bit how to dump any info on these, ended up using `Get-WmiObject Win32_PNPEntity` results below..

[2ded.txt](https://github.com/linux-surface/surface-aggregator-module/files/4776251/2ded.txt)

Let me know any more info required

qzed, 2020-06-14:
Thanks! That contains everything I was looking for. Looks like those are exactly the devices for IIDs 1, 3, 5, 6 (marked as `SHP_DEVICE_<iid>`).

One thing that I'd love to know is how Windows knows of these devices, i.e. there needs to be some sort of entity that tells Windows they're there Usually that's the job of ACPI, but that isn't used here, at least in the normal way. This is the reason we use the MFD cells in the SID driver to, basically, manually add the devices (https://github.com/linux-surface/surface-aggregator-module/blob/feature/sb3-v2/module/surface_sam_sid.c#L39-L51). I just wonder if there's a better way to do that/a way to do that automatically without us having to hard-code stuff. Maybe you guys have some idea? But that's another topic...

steveniemitz, 2020-06-29:
I've got some free time now that the dGPU stuff is (mostly) working, is there anything I can work on to move this forward?

qzed, 2020-06-29:
The missing parts are the events: Right now the `feature/sb3-v2` and `master` branches don't support any event registry apart from the SAM one (i.e. legacy). I've implemented support for other event registries on the `cleanup` branch two about weeks ago or so and I'm currently working to get that branch in a state where I feel comfortable replacing the current master with it. The cleanup branch now (finally) uses the new asynchronous message/packet handling under the hood, so, right now, the major thing that's missing before I can merge it into master is some structured testing (for which I still need to implement some fault-injection capabilities so that I can reliably test communication failures, time-outs, and stuff like that; trace-points for testing have been mostly finished yesterday). That should hopefully be done by end of this week.

Apart from that, one big thing: @archseer can you check how the events are enabled on the SL3? I doubt that the SAM registry is used (as that's on channel 1 and the events come from channel 2), so either KIP or REG should enable/disable events. See second part of https://github.com/linux-surface/surface-aggregator-module/issues/42#issuecomment-642892160 for details.

Ideally we want to make sure that the SL3 also works reliably and we don't break things (as mentioned, I kind of think the event enable/disable might be not-working already and we just haven't noticed yet).

So at the moment there's not much you can help here, but for the other issues (DTX and battery, I think) we still need to figure out which event registry is actually used for those.

qzed, 2020-07-07:
@steveniemitz the `feature/sb3-v3` branch has the updated version, based on the current master. Can you test that this works as expected? If so, I'm going to merge this into master.

I'd still like to get some info on how the events are enabled/disabled on the SL3 but I think this should not block this part from being merged. I'm pretty sure that the registry currently used is for the SL3 is the wrong one, since that one is on channel 1, but events are received on channel 2. If I'm reading the Windows SL3 Surface HID driver correctly, the channel on which the event registry resides and on which the events are being received should be the same...

steveniemitz, 2020-07-07:
ooo nice! I'll give it a try today.

steveniemitz, 2020-07-07:
lgtm, I tested the dGPU, keyboard, trackpad, lid.  `surface status` and related commands still work.  I'd say merge it!

steveniemitz, 2020-07-07:
actually, `make dkms-install` failed, it looks like the makefile is missing 
`sources += surface_sam_ssh_trace.h`

qzed, 2020-07-07:
Detachment and battery should now work too, on the same branch.

> actually, `make dkms-install` failed, it looks like the makefile is missing
> `sources += surface_sam_ssh_trace.h`

Ah right, I forgot to add that. I'll fix this in a sec.

> lgtm, I tested the dGPU, keyboard, trackpad, lid. `surface status` and related commands still work. I'd say merge it!

Thanks! I'll add the fix for the dkms issue on `master` and rebase the `feature/sb3-v3` branch onto it. I won't merge the detachment and battery stuff just yet as I still want to figure out the registries being used for them.

Also let's keep this issue open until the SL3 keyboard event stuff has been tested.

steveniemitz, 2020-07-07:
nice!  I actually haven't tested detatch/reattach recently, last time I tried I think multi-touch was disabled on the trackpad after re-attaching, is that still the case or should I re-test?

qzed, 2020-07-07:
Oh right, I totally forgot that part... That's still an issue.

qzed, 2020-07-07:
I wonder how Windows detects that kind of device change. I mean we could go via the events/commands of the DTX subsystem, but that would kind of require hard-coding that either all HID devices get removed when the base is removed, or at least which of them get removed.

qzed, 2020-07-07:
Could you try detaching and re-attaching and see if there are any unhandled event warnings in the log?

steveniemitz, 2020-07-07:
will do.  Also I just noticed that I think I lost battery info, lemme try rebooting and double check.

steveniemitz, 2020-07-07:
I ended up with a few events:
```

Jul  7 16:00:31 steve-work-sb3 kernel: [24083.185252] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x0e, tc: 0x0e, cid: 0x1d, iid: 0x00, chn: 0x01)
Jul  7 16:00:45 steve-work-sb3 kernel: [24097.577958] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, cid: 0x53, iid: 0x01, chn: 0x01)
Jul  7 16:00:47 steve-work-sb3 kernel: [24099.573609] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, cid: 0x53, iid: 0x01, chn: 0x01)
Jul  7 16:00:49 steve-work-sb3 kernel: [24100.969777] surface_dgpu_hps MSHW0153:00: dGPU physically detached
Jul  7 16:00:49 steve-work-sb3 kernel: [24100.970265] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x0e, tc: 0x0e, cid: 0x2c, iid: 0x00, chn: 0x01)
Jul  7 16:00:49 steve-work-sb3 kernel: [24100.971649] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x0e, tc: 0x0e, cid: 0x1d, iid: 0x00, chn: 0x01)
Jul  7 16:00:49 steve-work-sb3 kernel: [24100.974783] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, cid: 0x17, iid: 0x01, chn: 0x01)
Jul  7 16:01:05 steve-work-sb3 kernel: [24116.840168] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x0e, tc: 0x0e, cid: 0x2c, iid: 0x00, chn: 0x01)
Jul  7 16:01:05 steve-work-sb3 kernel: [24116.840971] surface_dgpu_hps MSHW0153:00: dGPU physically attached
Jul  7 16:01:05 steve-work-sb3 kernel: [24116.843892] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x0e, tc: 0x0e, cid: 0x1d, iid: 0x00, chn: 0x01)
Jul  7 16:01:05 steve-work-sb3 kernel: [24117.095495] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, cid: 0x17, iid: 0x01, chn: 0x01)
Jul  7 16:01:05 steve-work-sb3 kernel: [24117.097167] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, cid: 0x53, iid: 0x01, chn: 0x01)
Jul  7 16:01:06 steve-work-sb3 kernel: [24118.327054] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x0e, tc: 0x0e, cid: 0x2c, iid: 0x00, chn: 0x01)
Jul  7 16:01:06 steve-work-sb3 kernel: [24118.328614] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x0e, tc: 0x0e, cid: 0x1d, iid: 0x00, chn: 0x01)
Jul  7 16:01:08 steve-work-sb3 kernel: [24120.088023] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, cid: 0x53, iid: 0x01, chn: 0x01)
Jul  7 16:01:28 steve-work-sb3 kernel: [24139.860534] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x0e, tc: 0x0e, cid: 0x1d, iid: 0x00, chn: 0x01)

```

steveniemitz, 2020-07-07:
yeah, looks like I lost most of the power stuff (AC, battery).  I'll try to checkout what changed and see if I can get them working again.

steveniemitz, 2020-07-07:
oh ha, the battery/power lines seem like they fell out of `sid_devs_sb3`:

```
diff --git a/module/surface_sam_sid.c b/module/surface_sam_sid.c
index f30053c..22ae4e3 100644
--- a/module/surface_sam_sid.c
+++ b/module/surface_sam_sid.c
@@ -50,6 +50,9 @@ static const struct mfd_cell sid_devs_sb3[] = {
        { .name = "surface_sam_sid_vhf",      .id = 5 },
        { .name = "surface_sam_sid_vhf",      .id = 6 },
        { .name = "surface_sam_dtx",          .id = -1 },
+      { .name = "surface_sam_sid_ac",       .id = -1 },
+      { .name = "surface_sam_sid_battery",  .id =  1 },
+      { .name = "surface_sam_sid_battery",  .id =  2 },
        { },
 };
```

qzed, 2020-07-07:
> oh ha, the battery/power lines seem like they fell out of `sid_devs_sb3`:

Oh right, I forgot to actually add that part to the `feature/sb3-v3` branch. Oops. Fixed that.

> edit: I'm seeing:
> 
> ```
> [    4.634500] surface_sam_ssh_rqst: unexpected result while enabling event source: 0x01
> ```
> 
> which my guess is what's causing the missing power/ac devices.

Hmm, that should not be connected as the devices were missing completely. We should try to figure out which event enable call fails. Can you enable debug output and try to reproduce this? Alternatively print out `params.target_category` here:
https://github.com/linux-surface/surface-aggregator-module/blob/57f60766d9965ce8f22a1d43c7379052d8b6d608/module/surface_sam_ssh.c#L4217-L4221

steveniemitz, 2020-07-07:
sure thing, lemme add that in.

Also
> but that would kind of require hard-coding that either all HID devices get removed when the base is removed, or at least which of them get removed.

It does seem like the windows driver removes the appropriate HID devices when you detach.

qzed, 2020-07-07:
> It does seem like the windows driver removes the appropriate HID devices when you detach.

Yeah, the question is how we detect/decide what devices are supposed to be removed. I've had a closer look at the detachattach.log from https://github.com/linux-surface/linux-surface/issues/201#issuecomment-655171238 and there are the same events that you mentioned above (currently unhandled), but there doesn't seem to be something explicit like "HID device 3 removed". From what I can tell, all devices connected to the base are on channel 2 while devices connected to the clipboard are on channel 1, so maybe all devices on channel 2 get removed automatically (would make sense why they put the second battery there), but that could also be a coincidence and I'm not quite confident enough in that theory yet.

Also I have no clue what the unhandled events and payloads specifically indicate. It looks like it would have something to do with devices being removed or added due to the correlation with detachment and attachment, but no clue what exactly.

steveniemitz, 2020-07-08:
> oh ha, the battery/power lines seem like they fell out of sid_devs_sb3

Just noticed that one of the batteries is missing as well, I think the base one is missing still.  I'll poke around more.

steveniemitz, 2020-07-08:
> Can you enable debug output and try to reproduce this? Alternatively print out params.target_category here

Looks like its category 0x15.
```
[13582.276429] surface_sam_ssh_rqst: unexpected result while enabling event source: 0x01 cat: 0x15, channel: 0x02, instance_id: 0x00
```

qzed, 2020-07-08:
The `unexpected result while enabling event source` should be fixed now, I've rebased the `feature/sb3-v3` branch again. I forgot to actually set the correct instance ID and left it zero, which likely caused the error.

>  The error seems pretty benign since events are still enabled anyways.

Yeah, the events are a bit weird. As far as I can tell they are enabled by default, so the enable call is not strictly required. On one side this makes it a bit easier developing the event handling logic as we more or less always get informed about which events we need to handle (also the reason the driver works at all on the `features/sb3-v2` branch), on the other hand it complicates figuring out event enabling/disabling commands and could lead to (more or less random) issues if the event enabling/disabling calls are wrong, e.g. when events for some reason got disabled and we're unable to reliably enable them again (something that should be fixed for the keyboard/touchpad now, still have to figure this out for the battery and DTX).

steveniemitz, 2020-07-08:
> should be fixed now

Yep, looks good!

qzed, 2020-07-08:
Neat!

qzed, 2020-07-08:
Made things a bit more generic yet again and rebased the `feature/sb3-v3` branch. Let me know if there are any issues.

steveniemitz, 2020-07-08:
ooo it didn't like that one:

```
[    4.346914] sysfs: cannot create duplicate filename '/devices/platform/MSHW0117:00/surface_sam_sid_vhf.5'
[    4.346916] CPU: 7 PID: 192 Comm: kworker/u16:2 Tainted: G           OE     5.7.7-surface #2
[    4.346917] Hardware name: Microsoft Corporation Surface Book 3/Surface Book 3, BIOS 6.19.140 04/13/2020
[    4.346922] Workqueue: events_unbound async_run_entry_fn
[    4.346923] Call Trace:
[    4.346928]  dump_stack+0x6d/0x9a
[    4.346931]  sysfs_warn_dup.cold+0x17/0x35
[    4.346932]  sysfs_create_dir_ns+0xb8/0xd0
[    4.346934]  kobject_add_internal+0xbd/0x2b0
[    4.346936]  kobject_add+0x7e/0xb0
[    4.346939]  ? _cond_resched+0x19/0x30
[    4.346942]  ? kmem_cache_alloc_trace+0x16c/0x240
[    4.346944]  device_add+0x132/0x900
[    4.346946]  platform_device_add+0xf9/0x240
[    4.346948]  mfd_add_devices+0x46e/0x5b0
[    4.346952]  ? __acpi_match_device.part.0.constprop.0+0x11f/0x150
[    4.346955]  surface_sam_sid_probe+0x5d/0x6f [surface_sam_sid]
[    4.346956]  platform_drv_probe+0x3b/0x80
[    4.346957]  really_probe+0x2b3/0x3e0
[    4.346958]  driver_probe_device+0xbc/0x100
[    4.346959]  __driver_attach_async_helper+0x91/0xa0
[    4.346960]  async_run_entry_fn+0x3c/0x150
[    4.346963]  process_one_work+0x1e8/0x3b0
[    4.346964]  worker_thread+0x4d/0x400
[    4.346965]  kthread+0x104/0x140
[    4.346967]  ? process_one_work+0x3b0/0x3b0
[    4.346968]  ? kthread_park+0x90/0x90
[    4.346969]  ret_from_fork+0x1f/0x40
[    4.346972] kobject_add_internal failed for surface_sam_sid_vhf.5 with -EEXIST, don't try to register things with the same name in the same directory.
[    4.347420] surface_sam_sid: probe of MSHW0117:00 failed with error -17
```

steveniemitz, 2020-07-08:
```
diff --git a/module/surface_sam_sid.c b/module/surface_sam_sid.c
index a053564..8795dba 100644
--- a/module/surface_sam_sid.c
+++ b/module/surface_sam_sid.c
@@ -128,7 +128,7 @@ static const struct mfd_cell sid_devs_sb3[] = {
        },
        {
                .name = "surface_sam_sid_vhf",
-               .id = 5,
+               .id = 6,
```

;)

qzed, 2020-07-09:
Ah, thanks for spotting that... silly copy-paste mistake. Fixed.

qzed, 2020-07-13:
I've merged the current state into the linux-surface repo, so there should now be basic battery, keyboard, and touchpad support for the SB3 built into the kernel packages.

qzed, 2020-07-24:
Main support for this has been implemented. The remaining issues regarding detaching/attaching the clipboard will be handled in #46.

closed 2020-07-24 by qzed