linux-surface/surface-aggregator-module #45: SL3 No touchpad and keyboard

tmillich, 2020-07-15:
Since the last update my keyboard and touchpad is not working anymore

### Environment

- Hardware model: Surface Laptop 3 (Intel)
- Kernel version: 5.7.8-surface x86_64 x86_64 x86_64 GNU/Linux
- Distribution: Ubuntu 20.04

[  531.202929] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)
[  531.306455] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)
[  531.409167] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)
[  544.910620] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)
[  545.014129] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)
[  545.018374] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)
[  545.122055] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)
[  545.224990] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)
[  545.328613] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)
[  545.431379] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)
[  545.534103] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)
[  545.637091] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x15, tc: 0x15, cid: 0x00, iid: 0x01, chn: 0x02)

Any ideas?

qzed, 2020-07-15:
Moved to the SAM repo, as that's directly caused by some changes in the latest commits here.

qzed, 2020-07-15:
Can you try and test the module form the [`bugfix/sl3-hid`](https://github.com/linux-surface/surface-aggregator-module/tree/bugfix/sl3-hid) branch? Just `git clone` this repo and follow the instructions here: https://github.com/linux-surface/surface-aggregator-module/wiki/Testing-and-Installing#buildtest-the-modules.

tmillich, 2020-07-15:
I tried it. I get the error:

'Failed to start surface-dtx-daemon.service: Unit surface-dtx-deamon.service not found'

qzed, 2020-07-15:
Ah, sorry. The script loading the modules failed to start the DTX daemon because it's not available (and the script shouldn't assume that it's available). That was a bug in the script and it should be fixed in the latest commit.

Regardless of the error, the modules should have been loaded correctly.

> [UPDATE]
> Okey i figured out that keyboard and touchpad is working!

Can you confirm that the media keys (volume, brightness, etc.) and the caps-lock light also work?

> But my touchscreen doesnt work anymore.

Hmm, that should not be connected to the Surface Aggregator Module. Can you check `dmesg` and see if there are any error messages?

qzed, 2020-07-15:
If you don't mind I'd love to test a couple of things before I merge this into the kernel/patches. I will update the [announcement issue](https://github.com/linux-surface/linux-surface/issues/96) once new releases are on its way. Until then, you can install the module or fall back to an earlier version of the kernel, whatever you prefer.

tmillich, 2020-07-15:
Alright :) thanks

qzed, 2020-07-15:
Regarding testing: Can you run the `evreg.py` script from the `scripts/` directory (via sudo) with the following arguments (you'll need an external keyboard for this, as this is likely going to mess with the input again) and confirm some things?

- Running `./evreg.py reg-disable 0x15 0x00 0x01` should disable keyboard events. Running `./evreg.py reg-enable 0x15 0x00 0x01` should enable them again.
- Running `./evreg.py reg-disable 0x15 0x00 0x03` should disable touchpad events. Running `./evreg.py reg-enable 0x15 0x00 0x03` should enable them again.

Next, run
```
./evreg.py command 0x15 0x04 <iid> 0x02 0x01  0x02  0x00 0x00 0x00 0x00  0x10 0x00 0x00 0x00  0x00
```
and replace `<iid>` with 1 to 6 (inclusively), i.e. run the command once for each number. For which numbers to you get any output (1 and 3 should definitely produce output, but there may be others)? A short explanation to this command: This basically queries metadata from the connected HID devices (or time out if no device is available for that slot). We know (so far) that 1 should be the keyboard and 3 the touchpad (should hopefully be confirmed by the enable/disable calls above). On the SB3, there's also 5 and 6, which are for other (currently unknown firmware/MS-proprietary?) things. Would be good to export all devices, just so they're available in case they're ever needed.

Also: Are there any other `unhandled event` messages in the dmesg log?

tmillich, 2020-07-15:
- `./evreg.py reg-disable 0x15 0x00 0x01` --> disables keyboard events
- `./evreg.py reg-disable 0x15 0x00 0x01` --> disables touchpad events

i guess for 'command' you want to insert 'enable'|'disable', right?

qzed, 2020-07-15:
>   * `./evreg.py reg-disable 0x15 0x00 0x01` --> disables keyboard events
> 
>   * `./evreg.py reg-disable 0x15 0x00 0x01` --> disables touchpad events

Neat, thanks!

> i guess for 'command' you want to insert 'enable'|'disable', right?

Oh, I messed up the documentation of the script. This should be `simple` instead of `command` (no `enable`/`disable`).

qzed, 2020-07-15:
The script should be fixed now and the `command` sub-command should work (without having to replace it with `simple`).

tmillich, 2020-07-15:
1. `02 00 00 00 00 10 00 00 00 00 20 00 00 00 5e 04 ae 09 34 12 00 00 00 00 00 00`
2. `02 00 00 00 00 00 00 00 00 ff`
3. `02 00 00 00 00 10 00 00 00 00 20 00 00 00 5e 04 af 09 34 12 00 00 00 00 00 00`
4. `02 00 00 00 00 00 00 00 00 ff`
5. `02 00 00 00 00 10 00 00 00 00 20 00 00 00 5e 04 22 09 34 12 00 00 00 00 00 00`
6. `02 00 00 00 00 00 00 00 00 ff`

qzed, 2020-07-15:
Neat, thanks! So that indicates that there are devices in slots 1, 3, and 5. I'll update the driver accordingly and then start merging this into the kernel. Looks like the other commands don't actually time out, but they also don't return any buffer data (the `ff` at the end indicates the end, data up to and including the `ff` is just meta-data for buffer retrieval, so there's no actual content there, on the other entries there's the `10` indicating 0x10 = 16 bytes of data have been returned).

tmillich, 2020-07-15:
Oh okey :) about other unhandeled events... for now i dont find any :) but i keep a eye on it

qzed, 2020-07-15:
Perfect, thanks!

qzed, 2020-07-15:
Just a heads-up: I've merged this into the `master` branch of this repo and deleted the `bugfix/sl3-hid` branch.

tmillich, 2020-07-15:
Got those unhandeled events... but cant reproduce it :(

[  585.428468] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, cid: 0x53, iid: 0x01, chn: 0x01)
[ 1219.821216] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, cid: 0x53, iid: 0x01, chn: 0x01)
[ 1442.985926] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, cid: 0x53, iid: 0x01, chn: 0x01)
[ 1566.076875] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, cid: 0x53, iid: 0x01, chn: 0x01)
[ 1587.092893] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, cid: 0x53, iid: 0x01, chn: 0x01)

qzed, 2020-07-15:
Ah, that's the DPTF event for the battery/AC (so doesn't have anything to do with the keyboard/touchpad), we currently don't know the payload for that (https://github.com/linux-surface/linux-surface/issues/136). Should occur fairly randomly and be triggered when something in power delivery changes or there's some update regarding that.

A small explanation to the events: TC is target category, CID command ID (or here rather the type of event for the given TC), (IID instance ID, CHN channel, RQID request ID which should be equal to TC). TC=2 is battery/AC, TC=0x15 is HID/keyboard/touchpad ([here](https://github.com/linux-surface/surface-aggregator-module/blob/99b024313615009b2f940e9d166e78e81ab39704/doc/requests.txt#L31-L66)'s a longer list if you're interested).

qzed, 2020-07-15:
Alright, new kernel source and patches have been updated and kernels are building. Should be available later today or at latest tomorrow.

tmillich, 2020-07-15:
Thanks. I really appreciate what you guys are doing! The Open Source community in computer science is just nice.

qzed, 2020-07-16:
Thanks! It sure is.

I'm going to close this as it seems to be fixed. Feel free to comment or re-open if there are any remaining issues.

closed 2020-07-16 by qzed