linux-surface/surface-aggregator-module #47: SB3 dGPU issues when booting between windows and linux

steveniemitz, 2020-07-29:
Sometimes, when booting between windows and linux, the hps system seems like it can get stuck in a "weird" state.  Attempting to power the dGPU on/off via ACPI will hang, and eventually time out:

```
kernel: ACPI Error: Aborting method \_SB.PCI0.RP13.WPGS due to previous error (AE_AML_LOOP_TIMEOUT) (20200528/psparse-529)
kernel: ACPI Error: Aborting method \_SB.PCI0.RP13.HGOF due to previous error (AE_AML_LOOP_TIMEOUT) (20200528/psparse-529)
kernel: ACPI Error: Aborting method \_SB.SHPS._DSM due to previous error (AE_AML_LOOP_TIMEOUT) (20200528/psparse-529)
```

once it gets in this state, I've found one workaround to get things working again:
- boot back into windows
- detach the clipboard (via the button or DTX app).  This will take a couple tries, the first attempt or two will hang, but eventually will succeed.
- reattach
- reboot back into linux

We should figure out what the windows driver is doing here to "reset" the state correctly.  There was some discussion here https://github.com/linux-surface/surface-aggregator-module/issues/41#issuecomment-645704764 wrt what the firmware is waiting for.

archseer, 2020-08-01:
Semi-relatedly, I've had similar issues with the wifi firmware (https://github.com/linux-surface/linux-surface/issues/30), it will retain some of the Windows state and then fail to initialise on Linux. A quick workaround is that instead of rebooting, you fully shut down the device, then start it again. That way the machine power cycles and loses any sort of retained state.

steveniemitz, 2020-08-04:
agreed, I've gotten into the habit of never actually rebooting, and only shutting down/powering up. I have noticed though that the dGPU state seems to persist even across a full power down (sadly :( )

qzed, 2020-08-04:
To be honest, I don't even have a good clue on how to start debugging this. You could probably extend [`dbg_dump_drvsta`](https://github.com/linux-surface/surface-aggregator-module/blob/b9fe8f28b70b8a399c43177629a6a73641a87808/module/surface_sam_hps.c#L656) to also go and read the current GPIO values (`gpio_dgpu_power`, `gpio_dgpu_presence`, `gpio_base_presence`), which might give us a bit more info (if necessary insert that function in a couple of places), but I doubt that's going to be much help.

steveniemitz, 2020-08-04:
that seems like a good starting point to me. I mostly just made this issue to track it in case other people ran into the same. Hopefully I'll have a good chunk of time at some point to sit down and try to debug it.

steveniemitz, 2020-08-08:
looking through the ACPI firmware, I think this is the process HGON/HGOF uses to change the power mode (using ON for the example here)

- set GPIO `0x0605000E` to 1, it looks like this is referred to as `PWR RQST` (power request)
- wait for GPIO `0x0605000F` to go to 1, this is referred to as `PWR`. (power state)

I'm curious if we're missing handling the `PWR RQST` somehow?  It seems like its mapped to system memory but I don't really have a lot of experience with this, so I can't see how the GPIOs are actually set up in the kernel.

qzed, 2020-08-09:
> I'm curious if we're missing handling the `PWR RQST` somehow?

I have no clue.

Some thoughts on what Windows _might_ do: I somewhat doubt that Windows is calling the HGON/HGOF directly (or even via _DSM). Note that the `_ON` and `_OFF` methods of the PowerRegion of the Root Port call HGON/HGOF, so I think that Windows (at least when the driver is not in fallback/error-handling mode) uses those by switching the Root Port between D0 and D3 states. I've tried something similar in our Linux driver, but I still seem to need the _DSM call for a [workaround](https://github.com/linux-surface/surface-aggregator-module/blob/deb213a39a585cfded4ba26b5f67e2ec88795997/module/src/clients/surface_sam_hps.c#L446-L451). I also have some suspicions that we may need some changes in the PCIe hot-plug core to properly handle devices like this; I'm not really happy with the current driver and to me it feels a bit like a tacked-on thing that _should_ work fairly out-of-the-box just with the in-kernel PCIe hot-plug, Root Port, and power-state handling.

Now that I think about it: Does this happen when you blacklist the module? Could be that there's some race condition going on, i.e. PCIe core turns on dGPU, waits for GPIO, while simultaneously our driver turns off the dGPU and the GPIO never gets set.