linux-surface/surface-aggregator-module #48: Surface Laptop 2 keyboard not working after resuming from hibernate

raphtlw, 2020-08-20:
After waking up from hibernate mode, the keyboard does not work and I cannot log back into the system.

### Environment

- Hardware model: Surface Laptop 2
- Kernel version: Linux surface 5.8.1-arch1-3-surface linux-surface/linux-surface#1 SMP PREEMPT Tue, 18 Aug 2020 04:24:22 +0000 x86_64 GNU/Linux
- Distribution: Manjaro Linux

<details><summary>`dmesg` output</summary>

```
[    0.000000] microcode: microcode updated early to revision 0xd6, date = 2020-04-27
[    0.000000] Linux version 5.8.1-arch1-3-surface (linux-surface@archlinux) (gcc (GCC) 10.1.0, GNU ld (GNU Binutils) 2.34.0) linux-surface/linux-surface#1 SMP PREEMPT Tue, 18 Aug 2020 04:24:22 +0000
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-linux-surface root=UUID=36f84b4f-b2de-40e8-8bd7-4c7b9484c5d0 rw quiet resume=UUID=25eaab8e-6b26-4e45-8b29-a20df17ff842 udev.log_priority=3
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040400000-0x000000008aa0bfff] usable
[    0.000000] BIOS-e820: [mem 0x000000008aa0c000-0x000000008aa0cfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008aa0d000-0x000000008aa0dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008aa0e000-0x000000008bdcdfff] usable
[    0.000000] BIOS-e820: [mem 0x000000008bdce000-0x000000008be40fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000008be41000-0x000000008be8afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008be8b000-0x000000008beb0fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008beb1000-0x000000008befefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000008beff000-0x000000008befffff] usable
[    0.000000] BIOS-e820: [mem 0x000000008bf00000-0x000000008f7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000026f7fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by MSFT
[    0.000000] efi: TPMFinalLog=0x8bea7000 ACPI=0x8befe000 ACPI 2.0=0x8befe014 SMBIOS=0x8be4c000 SMBIOS 3.0=0x8be4a000 MEMATTR=0x88cee018 ESRT=0x88d07198 
[    0.000000] SMBIOS 3.2.0 present.
[    0.000000] DMI: Microsoft Corporation Surface Laptop 2/Surface Laptop 2, BIOS 138.3261.768 05.29.2020
[    0.000000] tsc: Detected 1800.000 MHz processor
[    0.001115] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001118] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001127] last_pfn = 0x26f800 max_arch_pfn = 0x400000000
[    0.001132] MTRR default type: write-back
[    0.001134] MTRR fixed ranges enabled:
[    0.001136]   00000-9FFFF write-back
[    0.001137]   A0000-BFFFF uncachable
[    0.001139]   C0000-FFFFF write-protect
[    0.001140] MTRR variable ranges enabled:
[    0.001142]   0 base 00C0000000 mask 7FC0000000 uncachable
[    0.001144]   1 base 00A0000000 mask 7FE0000000 uncachable
[    0.001145]   2 base 0090000000 mask 7FF0000000 uncachable
[    0.001146]   3 base 008E000000 mask 7FFE000000 uncachable
[    0.001147]   4 base 008D000000 mask 7FFF000000 uncachable
[    0.001148]   5 disabled
[    0.001149]   6 disabled
[    0.001150]   7 disabled
[... 798 lines cut ...]
[   11.002825] audit: type=1130 audit(1597908272.735:94): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=user-runtime-dir@1000 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   11.016369] mwifiex_pcie 0000:02:00.0: Request to disable ps_mode received. Disabling it.
[   11.020358] audit: type=1101 audit(1597908272.752:95): pid=1151 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:accounting grantors=pam_tally2,pam_access,pam_unix,pam_permit,pam_time acct="raphael" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   11.020826] audit: type=1103 audit(1597908272.752:96): pid=1151 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:setcred grantors=? acct="raphael" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
[   11.020975] audit: type=1006 audit(1597908272.752:97): pid=1151 uid=0 old-auid=4294967295 auid=1000 tty=(none) old-ses=4294967295 ses=3 res=1
[   11.024139] audit: type=1105 audit(1597908272.755:98): pid=1151 uid=0 auid=1000 ses=3 msg='op=PAM:session_open grantors=pam_loginuid,pam_loginuid,pam_keyinit,pam_limits,pam_unix,pam_permit,pam_mail,pam_systemd,pam_env acct="raphael" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   13.732103] Bluetooth: RFCOMM TTY layer initialized
[   13.732111] Bluetooth: RFCOMM socket layer initialized
[   13.732119] Bluetooth: RFCOMM ver 1.11
[   15.766087] mwifiex_pcie 0000:02:00.0: Request to disable ps_mode received. Disabling it.
[   17.569700] mwifiex_pcie 0000:02:00.0: info: trying to associate to 'flow_5G' bssid 04:d9:f5:a1:b8:0c
[   17.672887] mwifiex_pcie 0000:02:00.0: info: associated to bssid 04:d9:f5:a1:b8:0c successfully
[   18.636905] IPv6: ADDRCONF(NETDEV_CHANGE): wlp2s0: link becomes ready
[   19.011929] kauditd_printk_skb: 11 callbacks suppressed
[   19.011932] audit: type=1131 audit(1597908280.742:110): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   20.673372] audit: type=1130 audit(1597908282.405:111): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   21.181101] audit: type=1131 audit(1597908282.912:112): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=user@968 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   21.195974] audit: type=1131 audit(1597908282.929:113): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=user-runtime-dir@968 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   22.681235] fuse: init (API version 7.31)
[   32.017077] audit: type=1131 audit(1597908293.749:114): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   35.674991] audit: type=1131 audit(1597908298.318:115): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-hostnamed comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   35.968907] audit: type=1334 audit(1597908298.615:116): prog-id=10 op=UNLOAD
[   35.968919] audit: type=1334 audit(1597908298.615:117): prog-id=9 op=UNLOAD
```
</details>

qzed, 2020-08-20:
As far as I can tell, the controller (hardware) disables the events. Should be fairly straightforward to enable them again on resume, but I'll have to make a couple of changes for that.

archseer, 2020-08-21:
Hmm I use hibernation but things seem to function fine on SL3. I'll double check later

qzed, 2020-08-26:
@raphtlw Okay, events should now get re-enabled when resuming from hibernation. Can you test the module in the master branch? To do that, clone the repo, `cd` into the `module/` directory, then run `sudo make dkms-install`, reboot and start testing. After you've done, you can uninstall the module again via `sudo make dkms-uninstall` (again running that from the `module/` directory).

qzed, 2020-08-26:
HID devices should now also be properly suspended and resumed via the appropriate callbacks.

raphtlw, 2020-08-28:
@qzed unfortunately running `sudo make dkms-install` gives me an error.

> `Error! Bad return status for module build on kernel: 5.8.3-arch1-1-surface (x86_64)`

qzed, 2020-08-28:
Ah, looks like you're using a different GCC version than the one with which the kernel has been compiled (10.2.0). I believe you'll need to install that from the testing repositories on Manjaro.

qzed, 2020-08-28:
A new kernel release containing the changes is currently building. Should be live later, I'll ping this issue once it is.

qzed, 2020-08-28:
New releases will be live in a couple of minutes. Can you confirm that the issue has been fixed?

qzed, 2020-08-28:
Seems to be working according to https://github.com/linux-surface/linux-surface/issues/268.

closed 2020-08-28 by qzed