linux-surface/surface-aggregator-module #49: SB3: Unexpected result while enabling event source

NickyTope, 2020-08-30:
Since upgrading my kernel to 5.8.5 I could no longer run make in the sb3-v3 branch and looked at latest commits, I realised that cleanup had been merged to master so have done a `dkms-install` on master branch.

I now get the following errors on boot and my battery status is no longer being reported, keyboard and trackpad work fine, clipboard detaches very slow.

```~  dmesg | grep surface_sam
[    2.409294] surface_sam_ssh serial0-0: SAM controller version: 10.11.139
[    2.588543] surface_sam_ssh serial0-0: unexpected result while enabling event source: 0x01 (tc: 0x02, iid: 0x00, reg: 0x21)
[    2.588747] surface_sam_sid_battery: probe of 02:02:01:00 failed with error -71
[   32.470015] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, tid: 0x01, cid: 0x53, iid: 0x01)
[  175.327393] surface_sam_ssh serial0-0: event: unhandled event (rqid: 0x02, tc: 0x02, tid: 0x01, cid: 0x53, iid: 0x01)```

```~  acpi
Battery 0: Discharging, 0%, rate information unavailable
Battery 1: Full, 100%```

qzed, 2020-08-31:
First regarding clipboard detachment: I've not yet integrated the driver for this into the master and cleanup branches due to restructuring of the client device setup, so that's the expected behavior for now. Not sure when I get to this.

Okay, so the error indicates a failure when enabling events, specifically the EC returns (what I assume is) a non-zero status code (0x01). This should only affect the base-battery, the clipboard-battery should work fine. Looks like I mixed up the event registries. According to [this](https://github.com/linux-surface/surface-aggregator-module/issues/44#issuecomment-656685901) the registry should be KIP, but was set to REG. I've pushed a fix for that.

Can you try again with the latest version?

NickyTope, 2020-08-31:
Thanks again for your extremely quick response!!

Unfortunately `sudo make` is failing for me on both master and cleanup now, details below:
```
  cleanup  sudo make
make -C /lib/modules/"5.8.5-arch1-1-surface"/build M=/home/nt/apps/surface-aggregator-module/module/src __KERNEL_HAS_SSAM_MODALIAS_SUPPORT__=y modules
make[1]: Entering directory '/usr/lib/modules/5.8.5-arch1-1-surface/build'
  CC [M]  /home/nt/apps/surface-aggregator-module/module/src/core.o
  CC [M]  /home/nt/apps/surface-aggregator-module/module/src/ssh_parser.o
  CC [M]  /home/nt/apps/surface-aggregator-module/module/src/ssh_packet_layer.o
/home/nt/apps/surface-aggregator-module/module/src/ssh_packet_layer.c:482:20: error: redefinition of ‘ssh_packet_get’
  482 | struct ssh_packet *ssh_packet_get(struct ssh_packet *packet)
      |                    ^~~~~~~~~~~~~~
In file included from /home/nt/apps/surface-aggregator-module/module/src/ssh_packet_layer.c:17:
./include/linux/surface_aggregator_module.h:325:34: note: previous definition of ‘ssh_packet_get’ was here
  325 | static inline struct ssh_packet *ssh_packet_get(struct ssh_packet *packet)
      |                                  ^~~~~~~~~~~~~~
/home/nt/apps/surface-aggregator-module/module/src/ssh_packet_layer.c: In function ‘ssh_packet_next_try’:
/home/nt/apps/surface-aggregator-module/module/src/ssh_packet_layer.c:621:12: error: implicit declaration of function ‘ssh_packet_priority_get_base’; did you mean ‘ssh_packet_priority_get_try’? [-Werror=implicit-function-declaration]
  621 |  u8 base = ssh_packet_priority_get_base(priority);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |            ssh_packet_priority_get_try
In file included from ./include/linux/export.h:43,
                 from ./include/linux/linkage.h:7,
                 from ./include/linux/kernel.h:8,
                 from ./include/linux/unaligned/access_ok.h:5,
                 from ./arch/x86/include/asm/unaligned.h:9,
                 from /home/nt/apps/surface-aggregator-module/module/src/ssh_packet_layer.c:3:
/home/nt/apps/surface-aggregator-module/module/src/ssh_packet_layer.c:624:26: error: implicit declaration of function ‘__SSH_PACKET_PRIORITY’; did you mean ‘SSH_PACKET_PRIORITY’? [-Werror=implicit-function-declaration]
  624 |  WRITE_ONCE(p->priority, __SSH_PACKET_PRIORITY(base, try + 1));
      |                          ^~~~~~~~~~~~~~~~~~~~~
./include/linux/compiler.h:297:33: note: in definition of macro ‘__WRITE_ONCE’
  297 |  *(volatile typeof(x) *)&(x) = (val);    \
      |                                 ^~~
/home/nt/apps/surface-aggregator-module/module/src/ssh_packet_layer.c:624:2: note: in expansion of macro ‘WRITE_ONCE’
  624 |  WRITE_ONCE(p->priority, __SSH_PACKET_PRIORITY(base, try + 1));
      |  ^~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281: /home/nt/apps/surface-aggregator-module/module/src/ssh_packet_layer.o] Error 1
make[1]: *** [Makefile:1756: /home/nt/apps/surface-aggregator-module/module/src] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.8.5-arch1-1-surface/build'
make: *** [Makefile:31: all] Error 2
```

qzed, 2020-08-31:
Ah yeah, the problem is that the surface kernels already contain the main header file of the module and I've made some changes to it. I've tried to set up the module so that I don't have to change (almost) any code when integrating it into the kernel and can basically just copy it. Unfortunately I found out that that's not working out so well when I make changes to the header file, and that's what's causing the issue. It should work on any non-surface kernel.

TL;DR: Can you test it on a non-surface kernel?

NickyTope, 2020-09-01:
thanks, that makes sense.. will look at this tonight, need my sb for work at the moment

NickyTope, 2020-09-01:
Installed and booted linux 5.8.5 and headers, can confirm make ran with no errors.. 

"sudo make insmod" got the keyboard and trackpad working..

I thought I could just "make dkms-install" then but ran into issues with existing module. I did a "dkms remove surface_sam/0.1" then rebooted back into surface kernel, looks like I still can't "make dkms-install" though, is that right?

NickyTope, 2020-09-01:
My apologies, I did not check the battery status in the middle of all that... Can look again tomorrow

qzed, 2020-09-01:
No worries. DKMS install should work on non-surface kernels, but not on surface kernels at the moment. Still haven't fixed that problem. That being said on the latest surface kernels you shouldn't need to install the module manually (except for bugfixes maybe) as I'm usually updating it in the kernel once a week or so.

qzed, 2020-09-01:
Builds on surface kernels should now work again.

NickyTope, 2020-09-01:
Thank you @qzed can confirm the above issue is fixed and that battery 0 is now being reported.. still doesn't work with my polybar but I'll look more into what they are using to check the levels.

closed 2020-09-01 by NickyTope

qzed, 2020-09-02:
Thanks for the feedback!

> still doesn't work with my polybar

Did this work before? There haven't been any changes to the kernel-to-userspace interface that I'm aware of.

NickyTope, 2020-09-03:
Yea it did work before, not sure what changed but I just told it to use BAT1 and it started working again, I can use BAT1 and BAT2 in polybar and they line up with the values from acpi - Battery 0 and Battery 2...

```
acpi                             
Battery 0: Full, 100%
Battery 1: Discharging, 0%, rate information unavailable
Battery 2: Full, 100%
```
[image]

qzed, 2020-09-03:
Ahhh, right sorry I have changed the names from BAT0 and BAT1 to BAT1 and BAT2 to be consistent with the Surface Book 2, but that was quite a while ago.

qzed, 2020-09-08:
The fix is now also included in the latest kernel releases.