linux-surface/surface-aggregator-module #51: Help Wanted: Improvements for the Surface Laptop Keyboard Driver

qzed, 2020-10-02:
I'm currently cleaning up the Surface Aggregator Module drivers to eventually get them integrated into the upstream kernel. The HID/keyboard/touchpad driver needs a bit of re-work and I'd appreciate if any Surface Laptop 1, 2, or 3 or Surface Book 3 owners could run a couple of tests for me to figure out more details on the commands/protocol. This should also help finally getting the caps-lock led to work on the Surface Laptops 1 and 2.

Due to the interface being different, the commands I want you to run are separated in two categories: First, Surface Laptop 1 and 2, and second, Surface Laptop 3 and Surface Book 3. They are described below.

**Before you can start**, you'll need to
1. Update to the latest linux-surface kernel (or if you're using this module via DKMS, update to the latest version in the `master` branch). Don't forget to reboot afterwards.
2. Clone this repository (run `git clone  https://github.com/linux-surface/surface-aggregator-module.git`) and change into its directory (`cd surface-aggregator-module`). If you've already cloned it, make sure it's up to date (run `git pull` inside the directory).
3. Load the debug-interface by running `modprobe surface_aggregator_cdev`.
4. Run the commands below, choosing the ones for your device. You'll need to run them as root (e.g. via `sudo`). If not stated otherwise, the commands should have an output. Please provide that (in association with the command you ran) or mention if it fails.

If you have a **Surface Laptop 1 or 2**, please run:

- `./scripts/ssam/hid.py legacy-get-descriptor 0`

  This should return the HID descriptor of the keyboard.

- `./scripts/ssam/hid.py legacy-get-descriptor 1`

  This should return the HID report descriptor of the keyboard.

- `./scripts/ssam/hid.py legacy-get-descriptor 2`

  This should return the HID device attributes of the keyboard.

- `./scripts/ssam/hid.py legacy-get-feature-report 0`

  This should return the main feature report of the keyboard.

- `./scripts/ssam/hid.py legacy-set-capslock-led 1` and `./scripts/ssam/hid.py legacy-set-capslock-led 0`

  This does not have any output. The first command should turn on the caps-lock led, the second command should turn it off. Confirmation on this would be great.

If you have a **Surface Laptop 3 or Surface Book 3**, please run:

- `./scripts/ssam/hid.py hid-get-descriptor 1 0`

  This should return the HID descriptor of the keyboard.

- `./scripts/ssam/hid.py hid-get-descriptor 1 1`

  This should return the HID report descriptor of the keyboard.

- `./scripts/ssam/hid.py hid-get-descriptor 1 2`

  This should return the HID device attributes of the keyboard.

Thanks!

Cc: @NickyTope, @archseer, @shieldwed, @fxgsell, @lecram1989, @sudosubin, @raphtlw, @cvandal, @steveniemitz

shieldwed, 2020-10-02:
Hi @qzed

Here you are the descriptors for my Surface Book 3:
```bash
# sudo ./scripts/ssam/hid.py hid-get-descriptor 1 0
[9, 33, 17, 1, 0, 1, 34, 128, 0]
# sudo ./scripts/ssam/hid.py hid-get-descriptor 1 1
[5, 1, 9, 6, 161, 1, 133, 1, 21, 0, 37, 1, 117, 1, 149, 8, 5, 7, 25, 224, 41, 231, 129, 2, 117, 8, 149, 10, 25, 0, 41, 145, 38, 255, 0, 129, 0, 5, 12, 10, 192, 2, 161, 2, 26, 193, 2, 42, 198, 2, 149, 6, 177, 3, 192, 5, 8, 25, 1, 41, 3, 117, 1, 149, 3, 37, 1, 145, 2, 149, 5, 145, 1, 192, 5, 1, 9, 128, 161, 1, 133, 91, 21, 0, 37, 1, 117, 1, 149, 3, 9, 151, 9, 152, 9, 153, 129, 2, 149, 5, 129, 1, 192, 5, 12, 9, 1, 161, 1, 133, 3, 149, 1, 117, 16, 21, 0, 38, 255, 3, 25, 0, 42, 255, 3, 129, 0, 192]
# sudo ./scripts/ssam/hid.py hid-get-descriptor 1 2
[32, 0, 0, 0, 94, 4, 174, 9, 2, 16, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```
I hope this helps?
Cheers!

raphtlw, 2020-10-02:
I tried running `sudo modprobe surface_aggrergator_cdev` but it failed.

`modprobe: FATAL: Module surface_aggrergator_cdev not found in directory /lib/modules/5.8.12-arch1-2-surface`

qzed, 2020-10-02:
@raphtlw Oh, sorry, looks like I made a typo. This should be `modprobe surface_aggregator_cdev` (there was an `r` that should not have been there). Can you try again?

qzed, 2020-10-02:
@shieldwed Thanks, that's perfect!

raphtlw, 2020-10-02:
> @raphtlw Oh, sorry, looks like I made a typo. This should be `modprobe surface_aggregator_cdev` (there was an `r` that should not have been there). Can you try again?

Ah thanks, how did I not notice that 😅

raphtlw, 2020-10-02:
Hey @qzed, here are the descriptors for my Surface Laptop 2

```zsh
# ./scripts/ssam/hid.py legacy-get-descriptor 0
[9, 33, 0, 1, 51, 1, 34, 99, 0]
# ./scripts/ssam/hid.py legacy-get-descriptor 1
[5, 1, 9, 6, 161, 1, 133, 1, 21, 0, 37, 1, 117, 1, 149, 8, 5, 7, 25, 224, 41, 231, 129, 2, 117, 8, 149, 10, 25, 0, 41, 145, 38, 255, 0, 129, 0, 5, 12, 10, 192, 2, 161, 2, 26, 193, 2, 42, 198, 2, 149, 6, 177, 3, 192, 5, 8, 25, 1, 41, 3, 117, 1, 149, 3, 37, 1, 145, 2, 149, 5, 145, 1, 192, 5, 12, 9, 1, 161, 1, 133, 3, 117, 16, 149, 1, 21, 0, 38, 255, 3, 25, 0, 42, 255, 3, 129, 0, 192]
# ./scripts/ssam/hid.py legacy-get-descriptor 2
[32, 0, 0, 0, 94, 4, 28, 9, 52, 18, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
# ./scripts/ssam/hid.py legacy-get-feature-report 0
[1, 2, 1, 1, 0, 0, 0]
```

For the caps lock indicator commands, I am getting a python error: `TypeError: 'NoneType' object is not iterable`

qzed, 2020-10-02:
Awesome, thanks!

> For the caps lock indicator commands, I am getting a python error: `TypeError: 'NoneType' object is not iterable`. However, the indicator light turns on +1

Argh, somehow I added the code for printing any result returned by it, while the command itself doesn't return anything. Should be fixed now.

Does sending `0` as parameter turn it off?

cvandal, 2020-10-02:
Hey @qzed :wave:

Similar to @raphtlw, I'm getting the error bellow:

```
➜  surface-aggregator-module (master) ✔ modprobe surface_aggregator_cdev
modprobe: FATAL: Module surface_aggregator_cdev not found in directory /lib/modules/5.8.12-1.surface.fc32.x86_64
```

qzed, 2020-10-03:
@cvandal Looks like you don't have the latest version of the linux-surface kernel installed (the module has changed in that version). You have 5.8.12-1, the latest is 5.8.12-2.

cvandal, 2020-10-03:
> @cvandal Looks like you don't have the latest version of the linux-surface kernel installed (the module has changed in that version). You have 5.8.12-1, the latest is 5.8.12-2.

Interesting. I'm running the latest via DNF. Do I need to compile and install from source?

```
➜  ~ sudo dnf update --disableexcludes=all
[sudo] password for chris: 
Last metadata expiration check: 0:43:11 ago on Sat 03 Oct 2020 09:42:44 AM AEST.
Dependencies resolved.
Nothing to do.
Complete!
```

qzed, 2020-10-03:
Hmm, I can see version 5.8.12-2 in the repo... can you try running dnf with `--recheck`?

cvandal, 2020-10-03:
Thanks @qzed that did the trick :bow: 

```
➜  surface-aggregator-module (master) ✔ sudo ./scripts/ssam/hid.py hid-get-descriptor 1 0
[9, 33, 17, 1, 0, 1, 34, 128, 0]
➜  surface-aggregator-module (master) ✔ sudo ./scripts/ssam/hid.py hid-get-descriptor 1 1
[5, 1, 9, 6, 161, 1, 133, 1, 21, 0, 37, 1, 117, 1, 149, 8, 5, 7, 25, 224, 41, 231, 129, 2, 117, 8, 149, 10, 25, 0, 41, 145, 38, 255, 0, 129, 0, 5, 12, 10, 192, 2, 161, 2, 26, 193, 2, 42, 198, 2, 149, 6, 177, 3, 192, 5, 8, 25, 1, 41, 3, 117, 1, 149, 3, 37, 1, 145, 2, 149, 5, 145, 1, 192, 5, 1, 9, 128, 161, 1, 133, 91, 21, 0, 37, 1, 117, 1, 149, 3, 9, 151, 9, 152, 9, 153, 129, 2, 149, 5, 129, 1, 192, 5, 12, 9, 1, 161, 1, 133, 3, 149, 1, 117, 16, 21, 0, 38, 255, 3, 25, 0, 42, 255, 3, 129, 0, 192]
➜  surface-aggregator-module (master) ✔ sudo ./scripts/ssam/hid.py hid-get-descriptor 1 2
[32, 0, 0, 0, 94, 4, 174, 9, 52, 18, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```

qzed, 2020-10-03:
@cvandal Is that a Surface Laptop 3?

cvandal, 2020-10-03:
> @cvandal Is that a Surface Laptop 3?

Yes, sorry, I updated my comment to include that it's from a Surface Laptop 3.

qzed, 2020-10-03:
Awesome thanks! I think I have everything I need now!

raphtlw, 2020-10-03:
> Awesome, thanks!
>
>
> Argh, somehow I added the code for printing any result returned by it, while the command itself doesn't return anything. Should be fixed now.
>
> Does sending `0` as parameter turn [...]

Yep, it works now, and sending `0` turned it off.

qzed, 2020-10-04:
@raphtlw (or anyone with a Surface Laptop 1 or 2): Can you test the latest module version and make sure the keyboard still works? Note that you'll need an external keyboard as you'll have to unload the modules. To test this module:

1. Update to the latest linux-surface kernel. Don't forget to reboot afterwards.
2. Clone this repository (run `git clone  https://github.com/linux-surface/surface-aggregator-module.git`) and change into the `module/` directory (e.g. `cd surface-aggregator-module/module`). If you've already cloned it, make sure it's up to date (run `git pull` inside the directory).
3. Build the module by running `make clean all`.
4. Unload the old/in-kernel modules by running `make modprobe-unload`.
5. Load the new modules by running `make insmod`.

You can then test if the keyboard still works. If it doesn't, please check `dmesg` for errors (and post that if there are any). If the keyboard still works, can you toggle the caps-lock key a couple of times? This should cause some lines in the dmesg log to pop up, beginning with `surface_hid_raw_request` and subsequent lines beginning with `report: `, followed by a hex dump. Could you post this output (from a couple of caps-lock key presses)?

Once you're done, you can either run `make rmmod` and `make modprobe-load` to restore the old modules or simply reboot.

Thanks!

Cc: @lecram1989, @sudosubin

qzed, 2020-10-05:
The caps-lock light should now work. Can someone confirm this?

raphtlw, 2020-10-06:
Just tested the module on my Surface Laptop 2. The keyboard unfortunately does not work after doing `make insmod`.

<details>
<summary>
`dmesg` output
</summary>
[    0.000000] microcode: microcode updated early to revision 0xd6, date = 2020-04-27
[    0.000000] Linux version 5.8.12-arch1-2-surface (linux-surface@archlinux) (gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35) #1 SMP PREEMPT Thu, 01 Oct 2020 16:21:31 +0000
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-linux-surface root=UUID=36f84b4f-b2de-40e8-8bd7-4c7b9484c5d0 rw quiet resume=UUID=25eaab8e-6b26-4e45-8b29-a20df17ff842 udev.log_priority=3
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040400000-0x000000008aa0bfff] usable
[    0.000000] BIOS-e820: [mem 0x000000008aa0c000-0x000000008aa0cfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008aa0d000-0x000000008aa0dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008aa0e000-0x000000008bdcdfff] usable
[    0.000000] BIOS-e820: [mem 0x000000008bdce000-0x000000008be40fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000008be41000-0x000000008be8afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008be8b000-0x000000008beb0fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008beb1000-0x000000008befefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000008beff000-0x000000008befffff] usable
[    0.000000] BIOS-e820: [mem 0x000000008bf00000-0x000000008f7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000026f7fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by MSFT
[    0.000000] efi: TPMFinalLog=0x8bea7000 ACPI=0x8befe000 ACPI 2.0=0x8befe014 SMBIOS=0x8be4c000 SMBIOS 3.0=0x8be4a000 MEMATTR=0x88cee018 ESRT=0x88d07198 
[    0.000000] SMBIOS 3.2.0 present.
[    0.000000] DMI: Microsoft Corporation Surface Laptop 2/Surface Laptop 2, BIOS 138.3261.768 05.29.2020
[    0.000000] tsc: Detected 1800.000 MHz processor
[    0.001153] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001155] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001164] last_pfn = 0x26f800 max_arch_pfn = 0x400000000
[    0.001170] MTRR default type: write-back
[    0.001171] MTRR fixed ranges enabled:
[    0.001173]   00000-9FFFF write-back
[    0.001174]   A0000-BFFFF uncachable
[    0.001176]   C0000-FFFFF write-protect
[    0.001177] MTRR variable ranges enabled:
[    0.001179]   0 base 00C0000000 mask 7FC0000000 uncachable
[    0.001181]   1 base 00A0000000 mask 7FE0000000 uncachable
[    0.001182]   2 base 0090000000 mask 7FF0000000 uncachable
[    0.001183]   3 base 008E000000 mask 7FFE000000 uncachable
[    0.001185]   4 base 008D000000 mask 7FFF000000 uncachable
[    0.001186]   5 disabled
[    0.001186]   6 disabled
[    0.001187]   7 disabled
[    0.001188]   8 disabled
[    0.001189]   9 disabled
[    0.001948] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002330] last_pfn = 0x8bf00 max_arch_pfn = 0x400000000
[    0.022197] esrt: Reserving ESRT space from 0x0000000088d07198 to 0x0000000088d07298.
[... 974 lines cut ...]
[ 2768.736493] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
[ 2768.791148] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
[ 2768.868769] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
[ 2768.961178] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
[ 2769.066188] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
[ 2769.170992] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
[ 2769.276069] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
[ 2769.381107] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
[... cut ...]
[ 2813.536406] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
[ 2813.641265] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
[ 2813.746055] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
[ 2813.850941] surface_serial_hub serial0-0: event: unhandled event (rqid: 0x08, tc: 0x08, tid: 0x02, cid: 0x03, iid: 0x00)
</details>

qzed, 2020-10-06:
Okay, I've spotted and fixed a bug. Can you update and try again?

raphtlw, 2020-10-07:
@qzed the keyboard and the caps lock indicator still isn't working after a git pull :(

<details>
<summary><code>dmesg</code> output</summary>

[    0.000000] microcode: microcode updated early to revision 0xd6, date = 2020-04-27
[    0.000000] Linux version 5.8.12-arch1-2-surface (linux-surface@archlinux) (gcc (GCC) 10.2.0, GNU ld (GNU Binutils) 2.35) #1 SMP PREEMPT Thu, 01 Oct 2020 16:21:31 +0000
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-linux-surface root=UUID=36f84b4f-b2de-40e8-8bd7-4c7b9484c5d0 rw quiet resume=UUID=25eaab8e-6b26-4e45-8b29-a20df17ff842 udev.log_priority=3
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040400000-0x000000008aa0bfff] usable
[    0.000000] BIOS-e820: [mem 0x000000008aa0c000-0x000000008aa0cfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008aa0d000-0x000000008aa0dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008aa0e000-0x000000008bdcdfff] usable
[    0.000000] BIOS-e820: [mem 0x000000008bdce000-0x000000008be40fff] type 20
[    0.000000] BIOS-e820: [mem 0x000000008be41000-0x000000008be8afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008be8b000-0x000000008beb0fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008beb1000-0x000000008befefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000008beff000-0x000000008befffff] usable
[    0.000000] BIOS-e820: [mem 0x000000008bf00000-0x000000008f7fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000026f7fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by MSFT
[    0.000000] efi: TPMFinalLog=0x8bea7000 ACPI=0x8befe000 ACPI 2.0=0x8befe014 SMBIOS=0x8be4c000 SMBIOS 3.0=0x8be4a000 MEMATTR=0x88cee018 ESRT=0x88d07198 
[    0.000000] SMBIOS 3.2.0 present.
[    0.000000] DMI: Microsoft Corporation Surface Laptop 2/Surface Laptop 2, BIOS 138.3261.768 05.29.2020
[    0.000000] tsc: Detected 1800.000 MHz processor
[    0.001101] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001104] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001113] last_pfn = 0x26f800 max_arch_pfn = 0x400000000
[    0.001119] MTRR default type: write-back
[    0.001120] MTRR fixed ranges enabled:
[    0.001122]   00000-9FFFF write-back
[    0.001123]   A0000-BFFFF uncachable
[    0.001125]   C0000-FFFFF write-protect
[    0.001126] MTRR variable ranges enabled:
[    0.001128]   0 base 00C0000000 mask 7FC0000000 uncachable
[    0.001130]   1 base 00A0000000 mask 7FE0000000 uncachable
[    0.001131]   2 base 0090000000 mask 7FF0000000 uncachable
[    0.001132]   3 base 008E000000 mask 7FFE000000 uncachable
[    0.001133]   4 base 008D000000 mask 7FFF000000 uncachable
[    0.001134]   5 disabled
[    0.001135]   6 disabled
[    0.001136]   7 disabled
[    0.001137]   8 disabled
[    0.001138]   9 disabled
[    0.001874] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002282] last_pfn = 0x8bf00 max_arch_pfn = 0x400000000
[    0.022160] esrt: Reserving ESRT space from 0x0000000088d07198 to 0x0000000088d07298.
[    0.022170] e820: update [mem 0x88d07000-0x88d07fff] usable ==> reserved
[... 874 lines cut ...]
[  667.121286] surface_keyboard: probe of MSHW0096:00 failed with error -71
[  667.156255] battery: ACPI: Battery Slot [BAT1] (battery present)
[  667.164147] ACPI: AC Adapter [ADP1] (off-line)
[  675.430338] kauditd_printk_skb: 10 callbacks suppressed
[  675.430342] audit: type=1106 audit(1602049333.481:110): pid=8186 uid=0 auid=1000 ses=2 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[  675.430619] audit: type=1104 audit(1602049333.481:111): pid=8186 uid=0 auid=1000 ses=2 msg='op=PAM:setcred grantors=pam_faillock,pam_permit,pam_env,pam_faillock acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
[... cut ...]
[  750.953389] audit: type=1334 audit(1602049409.004:121): prog-id=15 op=UNLOAD
[  751.642451] audit: type=1130 audit(1602049409.691:122): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=upower comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  751.647435] audit: type=1106 audit(1602049409.697:123): pid=8515 uid=0 auid=1000 ses=2 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/2 res=success'
```bash

</details>

raphtlw, 2020-10-07:
I'm not very sure, but this line seems to be the error: 
```
[  667.121220] surface_keyboard MSHW0096:00: unexpected hid descriptor length: got 8457, expected 9
[  667.121286] surface_keyboard: probe of MSHW0096:00 failed with error -71
```

qzed, 2020-10-07:
Yeah, that's the error. I messed up the length access and read the field as 16 bit value instead of 8 bit... and that part worked before... Fixed that now. Could you try again?

raphtlw, 2020-10-07:
@qzed Yay! It works now, I am typing this using the loaded modules, and the caps lock is working perfectly too! Here's the output of the caps lock indicators:

<details>
<summary>caps lock logs from `dmesg`</summary>

```
[27909.129065] hid-generic 0006:045E:091C.0007: surface_hid_raw_request: reportnum=1, rtype=1, reqtype=9
[27909.129071] report: 00000000: 01 02
[27909.594459] hid-generic 0006:045E:091C.0007: surface_hid_raw_request: reportnum=1, rtype=1, reqtype=9
[27909.594467] report: 00000000: 01 00
[27909.792217] hid-generic 0006:045E:091C.0007: surface_hid_raw_request: reportnum=1, rtype=1, reqtype=9
[27909.792223] report: 00000000: 01 02
[27910.234929] hid-generic 0006:045E:091C.0007: surface_hid_raw_request: reportnum=1, rtype=1, reqtype=9
[27910.234935] report: 00000000: 01 00
[27910.898403] hid-generic 0006:045E:091C.0007: surface_hid_raw_request: reportnum=1, rtype=1, reqtype=9
[27910.898409] report: 00000000: 01 02
[27911.275887] hid-generic 0006:045E:091C.0007: surface_hid_raw_request: reportnum=1, rtype=1, reqtype=9
[27911.275893] report: 00000000: 01 00
```
</details>

qzed, 2020-10-07:
Awesome! Thank you very much for testing!

closed 2020-10-07 by qzed