linux-surface/surface-aggregator-module #53: People wanting to test Surface Pro X

Dmole, 2020-10-28:
https://github.com/Sonicadvance1/linux/issues/27

thmichel, 2020-11-06:
Hin, would be happy to test Linux on my Pro X. Are there any build instructions?

qzed, 2020-11-06:
@thmichel You might get more help for that on https://github.com/Sonicadvance1/linux/issues/27. There are also a [link](https://build.opensuse.org/package/show/home:lkocman:aarch64-laptops/kernel-source) to an openSUSE build service repo. Building the kernel should be fairly standard, but you'll probably need the appropriate patches and config options.

thmichel, 2020-11-06:
I built the kernel now using WSL on Windows. Problem is I cannot get the Pro X to boot the Linux image provided in your link, so I cannot install Linux in the first place.

qzed, 2020-11-06:
Can't really help with that as I've never tried linux on Android, sorry.

mirh, 2021-02-19:
Progresses have been made, thanks also to linaro.

qzed, 2022-06-11:
For Surface Pro X support, see https://github.com/Sonicadvance1/linux/issues/27 and https://github.com/linux-surface/surface-pro-x. While the changes haven't been included in this repository yet, we now do have SAM support for most parts (keyboard, touchpad, battery stats).

closed 2022-06-11 by qzed