linux-surface/surface-aggregator-module #55: module not working on opensuse ? (module verification failed)

GoldenHashtag, 2021-02-08:
surface linux kernel has no opensuse version, so instead of compiling the whole linux kernel i decided to install all the components individually. `ipts` and `surface-gpe` did work though but not sure why aggregator won't.
My device is surface pro 6. secure boot is enabled using microsoft & third party key. opensuse by default uses `shim`.
```
> uname -r
5.10.12-1-default
```
installed it via `$ make dkms-install` then manually loaded via
```
$ modprobe surface_perfmode; \
modprobe surface_aggregator; \
modprobe surface_acpi_notify; \
modprobe surface_aggregator_registry; \
modprobe surface_hid; \
modprobe surface_battery; \
modprobe surface_dtx; \
modprobe surface_aggregator_cdev
```
however battery status still won't display, even after rebooting.
dmesg shows "module verification failed", not sure if that caused the issue.
```
$ dmesg | grep surface
[    2.384087] integrity: Loaded X.509 cert 'Default Company Ltd: linux-surface: 24ef5c444499ba0aa6f2a8bea4258f065fea4ec6'
[   16.142523] surface_aggregator: loading out-of-tree module taints kernel.
[   16.142524] surface_aggregator: loading out-of-tree module taints kernel.
[   16.142617] surface_aggregator: module verification failed: signature and/or required key missing - tainting kernel
```

```
$ dmesg | grep acpi
[    0.028996] ACPI: LAPIC_NMI (acpi_id[0x01] high edge lint[0x1])
[    0.028997] ACPI: LAPIC_NMI (acpi_id[0x02] high edge lint[0x1])
[    0.028998] ACPI: LAPIC_NMI (acpi_id[0x03] high edge lint[0x1])
[    0.028999] ACPI: LAPIC_NMI (acpi_id[0x04] high edge lint[0x1])
[    0.029000] ACPI: LAPIC_NMI (acpi_id[0x05] high edge lint[0x1])
[    0.029001] ACPI: LAPIC_NMI (acpi_id[0x06] high edge lint[0x1])
[    0.029002] ACPI: LAPIC_NMI (acpi_id[0x07] high edge lint[0x1])
[    0.029003] ACPI: LAPIC_NMI (acpi_id[0x08] high edge lint[0x1])
[    0.029004] ACPI: LAPIC_NMI (acpi_id[0x09] high edge lint[0x1])
[    0.029004] ACPI: LAPIC_NMI (acpi_id[0x0a] high edge lint[0x1])
[    0.029005] ACPI: LAPIC_NMI (acpi_id[0x0b] high edge lint[0x1])
[    0.029006] ACPI: LAPIC_NMI (acpi_id[0x0c] high edge lint[0x1])
[    0.029007] ACPI: LAPIC_NMI (acpi_id[0x0d] high edge lint[0x1])
[    0.029008] ACPI: LAPIC_NMI (acpi_id[0x0e] high edge lint[0x1])
[    0.029009] ACPI: LAPIC_NMI (acpi_id[0x0f] high edge lint[0x1])
[    0.029010] ACPI: LAPIC_NMI (acpi_id[0x10] high edge lint[0x1])
[    0.163732] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
[    0.296344] acpi PNP0A08:00: _OSC: OS supports [ExtendedConfig ASPM ClockPM Segments MSI EDR HPX-Type3]
[    0.299662] acpi PNP0A08:00: _OSC: OS now controls [PCIeHotplug SHPCHotplug PME AER PCIeCapability LTR DPC]
[    0.413437] clocksource: acpi_pm: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 2085701024 ns
```

```
$ ls /sys/bus/acpi/devices/
ACPI0003:00  INT3472:00   LNXPOWER:0d  PNP0C0F:01  device:1c  device:3f  device:62
ACPI000C:00  INT3472:01   LNXSYBUS:00  PNP0C0F:02  device:1d  device:40  device:63
ACPI000E:00  INT3472:02   LNXSYBUS:01  PNP0C0F:03  device:1e  device:41  device:64
INT0800:00   INT347A:00   LNXSYSTM:00  PNP0C0F:04  device:1f  device:42  device:65
INT0E0C:00   INT347E:00   LNXVIDEO:00  PNP0C0F:05  device:20  device:43  device:66
INT33A1:00   INT3F0D:00   MSHW0005:00  PNP0C0F:06  device:21  device:44  device:67
INT33BE:00   LNXCPU:00    MSHW0006:00  PNP0C0F:07  device:22  device:45  device:68
INT3400:00   LNXCPU:01    MSHW0008:00  device:00   device:23  device:46  device:69
INT3403:00   LNXCPU:02    MSHW0036:00  device:01   device:24  device:47  device:6a
INT3403:01   LNXCPU:03    MSHW0040:00  device:02   device:25  device:48  device:6b
INT3403:02   LNXCPU:04    MSHW0045:00  device:03   device:26  device:49  device:6c
INT3403:03   LNXCPU:05    MSHW0048:00  device:04   device:27  device:4a  device:6d
INT3403:04   LNXCPU:06    MSHW0084:00  device:05   device:28  device:4b  device:6e
INT3403:05   LNXCPU:07    MSHW0091:00  device:06   device:29  device:4c  device:6f
INT3403:06   LNXCPU:08    MSHW0102:00  device:07   device:2a  device:4d  device:70
INT3403:07   LNXCPU:09    MSHW0111:00  device:08   device:2b  device:4e  device:71
INT3403:08   LNXCPU:0a    MSHW0125:00  device:09   device:2c  device:4f  device:72
INT3403:09   LNXCPU:0b    MSHW0131:00  device:0a   device:2d  device:50  device:73
INT3403:0a   LNXCPU:0c    NTZ0412:00   device:0b   device:2e  device:51  device:74
INT3403:0b   LNXCPU:0d    PNP0000:00   device:0c   device:2f  device:52  device:75
INT3403:0c   LNXCPU:0e    PNP0100:00   device:0d   device:30  device:53  device:76
INT3403:0d   LNXCPU:0f    PNP0103:00   device:0e   device:31  device:54  device:77
INT3403:0e   LNXPOWER:00  PNP0A08:00   device:0f   device:32  device:55  device:78
INT3403:0f   LNXPOWER:01  PNP0B00:00   device:10   device:33  device:56  device:79
INT3403:10   LNXPOWER:02  PNP0C02:00   device:11   device:34  device:57  device:7a
INT3407:00   LNXPOWER:03  PNP0C02:01   device:12   device:35  device:58  device:7b
INT343D:00   LNXPOWER:04  PNP0C02:02   device:13   device:36  device:59  device:7c
INT3440:00   LNXPOWER:05  PNP0C02:03   device:14   device:37  device:5a  device:7d
INT3441:00   LNXPOWER:06  PNP0C02:04   device:15   device:38  device:5b  device:7e
INT3446:00   LNXPOWER:07  PNP0C02:05   device:16   device:39  device:5c
INT3447:00   LNXPOWER:08  PNP0C02:06   device:17   device:3a  device:5d
INT3449:00   LNXPOWER:09  PNP0C04:00   device:18   device:3b  device:5e
INT344A:00   LNXPOWER:0a  PNP0C0A:00   device:19   device:3c  device:5f
INT344B:00   LNXPOWER:0b  PNP0C0D:00   device:1a   device:3d  device:60
INT3470:00   LNXPOWER:0c  PNP0C0F:00   device:1b   device:3e  device:61
```

also, surface-control won't work.
```
$ surface status
Error: Cannot access device.
       No surface control device not found.
```

qzed, 2021-02-08:
Your secureboot configuration expects all kernel modules to be signed. Installation via DKMS does not sign them for you. So you have two options:
1. Disable secureboot.
2. Sign the modules manually. See e.g. [here](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/kernel_administration_guide/chap-documentation-kernel_administration_guide-working_with_kernel_modules#sect-signing-kernel-modules-for-secure-boot) for some documentation on that. The gist is: Create a new key/MOK (if you don't have any), sign the modules with that key, then enroll that key.

StollD, 2021-02-08:
I am not sure if this is due to secureboot. @GoldenHashtag mentioned that `ipts` and `surface-gpe` work, so it wouldnt make sense for the SAM modules to be rejected. The warning above is also just the typical "The kernel wont get support from upstream" message, and not that the module wasnt loaded.

@GoldenHashtag could you post the full dmesg? errors rarely conform to grep expressions, so maybe that reveals something.

My guess would be that some other option that is required by SAM is not enabled in the SUSE kernel. So the module would load but then be unable to do anything because of the missing dependencies.

qzed, 2021-02-08:
Oh, right. Missed that other modules are working... doesn't look  like a secureboot problem then. @StollD is probably correct with missing kernel options.  In addition to the dmesg log can you post the output of `ls -l /sys/bus/serial/devices`?

GoldenHashtag, 2021-02-08:
> Your secureboot configuration expects all kernel modules to be signed. Installation via DKMS does not sign them for you. So you have two options:
>
>     1. Disable secureboot.
>
>     2. Sign the modules [...]

Tried disabling secure boot, doesn't help.

> I am not sure if this is due to secureboot. @GoldenHashtag mentioned that `ipts` and `surface-gpe` work, so it wouldnt make sense for the SAM modules to be rejected. The warning above is also just [...]

```
$ modprobe -r surface_perfmode; \
> modprobe -r surface_aggregator; \
> modprobe -r surface_acpi_notify; \
> modprobe -r surface_aggregator_registry; \
> modprobe -r surface_hid; \
> modprobe -r surface_battery; \
> modprobe -r surface_dtx; \
> modprobe -r surface_aggregator_cdev
modprobe: FATAL: Module surface_aggregator is in use.
$ modprobe -r surface_perfmode; modprobe -r surface_aggregator; modprobe -r surface_acpi_notify; modprobe -r surface_aggregator_registry; modprobe -r surface_hid; modprobe -r surface_battery; modprobe -r surface_dtx; modprobe -r surface_aggregator_cdev
$ lsmod | grep surface
surfacepro3_button     20480  0
surface_gpe            20480  0
$ dmesg --clear
$ modprobe surface_perfmode; \
> modprobe surface_aggregator; \
> modprobe surface_acpi_notify; \
> modprobe surface_aggregator_registry; \
> modprobe surface_hid; \
> modprobe surface_battery; \
> modprobe surface_dtx; \
> modprobe surface_aggregator_cdev
$ dmesg
$ lsmod | grep surface
surface_aggregator_cdev    16384  0
surface_dtx            24576  0
surface_battery        28672  0
surface_hid            16384  0
surface_aggregator_registry    16384  0
surface_acpi_notify    20480  0
surface_perfmode       16384  0
surface_aggregator    143360  7 surface_acpi_notify,surface_perfmode,surface_battery,surface_aggregator_cdev,surface_dtx,surface_hid,surface_aggregator_registry
serdev                 24576  1 surface_aggregator
surfacepro3_button     20480  0
surface_gpe            20480  0
```
when starting the modules manually there was no output in `dmesg`, yet `lsmod` still indicates they are loaded.
executed this right after restarted pc and logged in:
```
$ dmesg > log.txt ###because it's so long that exceeded terminal scroll back buffer
$ mv log.txt /home/golden
$ chmod 777 /home/golden/log.txt
```
[full log](https://pastebin.com/7eEkQJvQ)
password: az24BSsRd5

> Oh, right. Missed that other modules are working... doesn't look like a secureboot problem then. @StollD is probably correct with missing kernel options. In addition to the dmesg log can you post the output of `ls -l /sys/bus/serial/devices`?

```
$ ls -l /sys/bus/serial/devices
total 0
```

GoldenHashtag, 2021-02-08:
I have no idea about kernel options as this is a very new installation (just yesterday) and the only thing i've been working on was surface related drivers on linux.
Looking up the web and found `intel_lpss_pci` is required for this to be functional but it's already loaded.

qzed, 2021-02-08:
Can you upload the kernel config? It's normally located in one of `/proc/config.gz`, `/boot/config`, or `/boot/config-<kernel-version>`.

Based on this

> > `ls -l /sys/bus/serial/devices`?
> 
> ```
> $ ls -l /sys/bus/serial/devices
> total 0
> ```

 you're at least missing `SERIAL_DEV_BUS=y` `SERIAL_DEV_CTRL_TTYPORT=y`, possibly more. You should probably petition the opensuse team to enable those.

StollD, 2021-02-08:
This seems to be the config they use: https://github.com/openSUSE/kernel-source/blob/stable/config/x86_64/default

- `CONFIG_SERIAL_DEV_BUS` is set to `m`
- `CONFIG_SERIAL_DEV_CTRL_TTYPORT` doesnt exist

GoldenHashtag, 2021-02-10:
yes, my config was just the same as @StollD said.
I've been looking into this for two days, the only relevant informantion i could find was [this](https://www.reddit.com/r/openSUSE/comments/72wz1l/best_way_to_compile_a_kernel_with_custom_options/). but I really can't figure out how to use this open build service. also from my guess the custom compiled kernel requires signing to avoid issues with secure boot then it's rather complicated as tumbleweed is a rolling release and kernel update are frequent, and that you have to sign for each updates. 
opensuse uses rpm which is same as fedora. it also had dnf in main repository which enables users to add fedora repositories. from my past experiences fedora and opensuse usually are binary compatible (lots of software without opensuse support can be installed using their fedora build). I'm currently looking into this way. however then again tumbleweed is rolling release and uses alternative distro version number similar to fedora silverblue and standard dnf repository can't be used.

GoldenHashtag, 2021-02-10:
> Can you upload the kernel config? It's normally located in one of `/proc/config.gz`, `/boot/config`, or `/boot/config-<kernel-version>`.
>
> Based on this
>
>
> you're at least missing `SERIAL_DEV_BUS=y` [...]

I do want to petition opensuse to enable those as it would be the best solution (up to date kernel and battery status the same time), but i'm not sure if they will listen (they may just tell me to use obs), nor do i know where to petition them.

or should i just learn how to use obs and automate everything.

GoldenHashtag, 2021-02-10:
here, I filed a feature request.
https://bugzilla.opensuse.org/show_bug.cgi?id=1182035
let's hope they will change it.

qzed, 2021-02-10:
I think those options should be general enough that they have a good chance to be enabled. Pretty much any other big distribution has those set (IIRC NixOS doesn't have them set either, but that's the only other distro I'm aware of and that's a bit special anyways).

Also with v5.12 parts of the Surface Aggregator Module will land upstream (basically the core and some stuff for the 5th generation devices), so any kernel wanting to support Surface devices with that will have to enable them. You can expect more things to land in v5.13.

GoldenHashtag, 2021-02-20:
Issue resolved as of version 20210218.

closed 2021-02-20 by GoldenHashtag