linux-surface/surface-aggregator-module #56: Surface 3 support?

deliciouslytyped, 2021-04-30:
I see the Surface 3 isn't listed in the supported devices table, and I'm having some issues, namely `acpi -V` doesn't list any thermal info.
```
# nix-shell -p acpi --run "acpi -V"
Battery 0: Full, 100%
Battery 0: design capacity 7683 mAh, last full capacity 6404 mAh = 83%
Adapter 0: off-line
Cooling 0: Processor 0 of 10
Cooling 1: STR0 no state information available
Cooling 2: LCD 0 of 99
Cooling 3: LCD 0 of 99
Cooling 4: LCD 0 of 99
Cooling 5: LCD 0 of 99
Cooling 6: Processor 0 of 10
Cooling 7: soc_dts0 no state information available
Cooling 8: LCD 0 of 99
Cooling 9: Processor 0 of 10
Cooling 10: INT3400 Thermal no state information available
Cooling 11: TCHG 0 of 4
Cooling 12: STR2 no state information available
Cooling 13: LCD 0 of 99
Cooling 14: LCD 0 of 99
Cooling 15: PNIT no state information available
Cooling 16: LCD 0 of 99
Cooling 17: Processor 0 of 10
Cooling 18: soc_dts1 no state information available
Cooling 19: intel_powerclamp no state information available
```

Devices in sysfs seem to exist though, but I don't know if any of this is sensible:
```
# cat /sys/class/thermal/thermal_zone*/{type,temp}
PNIT
soc_dts0
soc_dts1
INT3400 Thermal
STR0
STR2
54000
50000
57000
20000
-273150
-273150
```

deliciouslytyped, 2021-04-30:
I forgot to mention, I don't seem to have surface_aggregator sysfs path
```
# find /sys -iname "*surface*"
/sys/devices/platform/surface3-wmi
/sys/bus/platform/devices/surface3-wmi
/sys/bus/platform/drivers/surface3-wmi
/sys/bus/platform/drivers/surface3-wmi/surface3-wmi
/sys/bus/spi/drivers/Surface3-spi
/sys/bus/acpi/drivers/surface_pro3_button
/sys/bus/i2c/drivers/surface3
/sys/module/i2c_core/holders/surface3_button
/sys/module/i2c_core/holders/surface3_power
/sys/module/surface3_spi
/sys/module/surface3_spi/drivers/spi:Surface3-spi
/sys/module/surfacepro3_button
/sys/module/surface3_button
/sys/module/surface3_button/drivers/i2c:surface3
/sys/module/surface3_wmi
/sys/module/surface3_wmi/drivers/platform:surface3-wmi
/sys/module/surface3_power
/sys/module/wmi/holders/surface3_wmi
```

qzed, 2021-04-30:
The Surface 3 doesn't have a Surface Aggregator Module (SAM) connected via the Surface Serial Hub (SSH), i.e. SAM-over-SSH. Instead it uses a connetion via HID-over-I2C (at least as far as I can tell from ACPI), similar to the Surface Pro 4 (and probably other earlier generation devices). The driver in this repo only targets SAM-over-SSH devices, meaning 5th generation (i.e. Surface Pro 5/2017, Surface Book 2, Surface Laptop 1) and later. Previous generations are unfortunately not supported by this.

Supporting SAM-over-HID/I2C would probably require a significant amount of reverse engineering, as we currently don't know much about it. I don't even know if it provides a thermal subsystem at all (there's no hint of that in ACPI like on the later generations, as far as I can tell). There's been an attempt centered around the discrete GPU of the Surface Book 1 (which is controlled via SAM) at https://github.com/linux-surface/surface-aggregator-module-gen4, but again, full support would require someone with such a device to reverse engineer the communication protocol (either by looking at the Windows drivers or by listening in on the communication on Windows, as we have done for SAM-over-SSH) as well as figuring out its capabilities/what it's used for on those devices in the first place (all functionality that's provided by it on later generation seems to be available directly via ACPI on the Surface 3 and doesn't seem to require a SAM/SAN driver).

closed 2021-04-30 by qzed