linux-surface/surface-aggregator-module #58: Surface Control unable to set or get performance - 'device does not exist'

JPDucky, 2021-05-30:
I have been unable to get the control utility to set or get performance as the device does not exist. I have both re-installed and un-installed-then-installed-again to no avail. The same thing happened under the fedora 5.12.6 kernel as well, this is a new install as of a week ago. 

These are the contents of /sys/bus/surface_aggregator/devices/"01\:03\:01\:00\:01/":
        driver modalias power software_node subsystem uevent

I have attempted to touch/nano a 'perf_mode' file but cannot, even as root. This is what happens when I try and set performance:
``` 
sudo surface performance set 4
Error: Failed to get current performance mode

Caused by:
    0: Could not access device "/sys/bus/surface_aggregator/devices/01:03:01:00:01/perf_mode"
    1: No such file or directory (os error 2)
```

I'm sure I am missing something simple 

Kernel: Linux fedora 5.12.8-1.surface.fc34.x86_64 #1 SMP Sat May 29 15:37:39 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

qzed, 2021-05-31:
That interface has been replaced with the ACPI platform profile (`/sys/firmware/acpi/platform_profile`). See https://github.com/linux-surface/linux-surface/issues/96#issuecomment-838706300 for more details.

JPDucky, 2021-05-31:
> That interface has been replaced with the ACPI platform profile (`/sys/firmware/acpi/platform_profile`). See [linux-surface/linux-surface#96 (comment)](https://github.com/linux-surface/linux-surface/issues/96#issuecomment-838706300) for more details. There's also a new subcommand to go with that: `surface profile`.

Can confirm that the `surface profile set xyz` is working. Closing this out.

closed 2021-05-31 by JPDucky

gamer191, 2021-11-07:
@qzed I might be missing something, but ``surface profile --help`` doesn't show the available performance modes. They don't seem to be the same as the surface performance ones, since ``surface profile set 2`` isn't supported and nor are the others. What are the possible profiles? Also, what is the difference between ``surface profile`` and ``surface performance``?

qzed, 2021-11-07:
Try `surface profile list` instead for listing them, then use the names listed there.

gamer191, 2021-11-08:
> Try `surface profile list` instead for listing them, then use the names listed there.

Thanks
> The kernel interface.

I'm not quite sure what that means (I'm a bit of a noob). Is it something important?

qzed, 2021-11-08:
Newer kernels have a different interface, the new command (`surface profile`) uses that interface whereas the old command (`surface performance`) uses the old interface. So depending on the kernel you either have to use the new or the old command, but both essentially do the same.

gamer191, 2021-11-09:
@qzed Thanks heaps for explaining that 😀🙏