linux-surface/surface-aggregator-module #62: folders inside surface_aggregator are empty

ECO1AI, 2022-06-11:
after installing surface control and running `surface status` i get : `Error: No devices found`
also i tried running `surface performance get` and got this error : 
```
Error: Failed to open performance mode device

Caused by:
    0: Could not access device "/sys/bus/surface_aggregator/devices/01:03:01:00:01"
    1: No such file or directory
```
after checking "/sys/bus/surface_aggregator" i found : 
2 binary files ( uevent , drivers_probe )
2 empty folders ( devices , drivers  )
1 empty text file ( drivers_autoprobe  )

my device is surface pro 4 running manjaro gnome with the latest surface kernel ( 5.18  )
if there is anything missing or you have any quastion about how well my device is running linux, feel free to ask
thank you for your hard work in advance

qzed, 2022-06-11:
The Pro 4 doesn't use the Surface Serial Hub interface, so it's not supported by this module and the surface-control utility.

ECO1AI, 2022-06-11:
i see
thanks

closed 2022-06-11 by ECO1AI

qzed, 2022-06-11:
There's a discussion mostly about the SB1 which seems to use the same EC/interface at https://github.com/linux-surface/linux-surface/issues/93, but I'm not sure how much of that applies to the SP4.