linux-surface/surface-aggregator-module #63: USB-C Power Delivery Events

qzed, 2022-06-22:
The `USC` subsystem is used for USB-C power delivery management on the Pro X via a `SurfaceUsbCMuxAcpiFilterDriver.sys`. It's possible that other (newer) devices use that as well.

Quoting myself from the issue where this was first discovered (https://github.com/linux-surface/surface-pro-x/issues/33):

> There is indeed some SAM stuff going on in the SurfaceUsbCMuxAcpiFilterDriver.sys using the USC subsystem. There seem to be two methods, specifically. A FilterGetPortNumbers and a FilterSendUSBCAck. In addition to that, there's a PdEventInfoUpdateFromSamToSoc function, which might be some part of a PD event handler by its name.

> In fact all three functions are called in the same ProcessUsbCResponse function, first getting the port number(s?), then updating the PD event info, then sending the ACK back. I think what's happening is that SAM might send a PD event when a USB-C source is plugged in. This is then handled by that process-response function.

And from the related issue for the Pro X (the device on which this was first discovered; https://github.com/linux-surface/surface-pro-x/issues/35):

> Turns out we can enable and receive events on Linux (using the standard SAM event registry) when I plug in my phone for charging, but I should probably get some USB-C power meter and a powerful enough USB-C charger to see if there are any changes. Also I don't know yet what to send back to SAM via the ACK command.

> Interestingly, when I enable SAM events, the port seems to do something (at very least cut power for a brief moment) as my phone screen turns on. Phone says "fast charging" regardless of events being enabled/disabled though...

Events can be enabled with
```
./scripts/ssam/events.py enable 0x01 0x01 0x0b 0x0c 0x1b 0x00 0x01
```
and listened to with
```
./scripts/ssam/events.py listen 0x1b
```

It would be interesting to know whether that's also a thing on other devices.

NP-chaonay, 2022-06-23:
SLS confirmed nothing happens when run above events.py

Both switching from AC > AC+USB Type C > USB Type C >> no events displayed
and USB Type C > AC+USB Type C > AC >> no events displayed

qzed, 2022-06-23:
Thanks. It looks like this is specific to the Pro X.