linux-surface/surface-aggregator-module #64: SAM firmware reverse engineering

quo, 2022-10-31:
First of all, @qzed thanks for all the work you've already done reverse engineering and documenting a lot of the SAM stuff. It's been quite helpful!

I've been reverse engineering the SP7 SAM firmware (specifically SurfaceSAM_14.312.139.bin) in an attempt to debug something. Haven't found anything particularly useful so far, but figured I would share regardless. Let me know if you'd like me to look at anything in particular.

<details>
<summary>Click here for info dump</summary>

## Firmware structure

As can be seen on SP7 teardown photos, the SAM microcontroller is an NXP LPC54S001J (Cortex-M4, 360KB RAM), with a separate Winbond 16MB flash chip.
- Datasheet: https://www.nxp.com/docs/en/data-sheet/LPC540xx.pdf
- User manual: https://www.mouser.com/pdfdocs/NXP_LPC540_UM.pdf
- SVD: https://github.com/NXPmicro/mcux-soc-svd/tree/main/LPC54S005

The SVD is for a slightly different part number, but it does the job. There's a script to import SVDs into Ghidra, but it fails due to some overlapping address ranges. You can either remove these from the SVD or hack the script a bit.

The bin file consists of a signature and two firmware images. The images are encoded as arrays of `{ u32 offset, u8 len = 16, u8[16] data }`, so can be extracted fairly easily. The two images are identical, except one is meant to be flashed at 0x10004000 and the other at 0x10084000 (standard A/B update handling), so some addresses differ by one bit. The images have a header and end with a CRC16, which are used by the SAM when flashing. The actual raw firmware image starts 0x66C.

The raw firmware images start with a standard ARM vector table, and contain an NXP image header which tells the NXP bootloader to load the first 0x29484 bytes into SRAM at address 0. For reverse engineering, you can just split the raw image and load the first 0x30000 bytes at 0, and the remaining 0x50000 bytes at 0x10034000 or 0x100B4000.

The firmware contains an RTOS which is internally referred to as Kaos. I can't find anything about it on Google, so I assume it was created by MS. Kaos appears somewhat inspired by FreeRTOS and offers the same basic primitives: tasks, timers, events, semaphores, and message queues. There are dozens of tasks and timers, which communicate through dozens of message queues, which leads to a ton of indirection (and memory overhead) so it can be very difficult to follow what's going on. Some parts of the firmware use vtable-like constructions for some additional indirection. Lots of fun.

## SAM protocol

(I'll try to use the terminology from https://github.com/linux-surface/surface-aggregator-module/tree/master/doc.)

The following target(/source) IDs are implemented:
- 0 = Host
- 1 = SAM
- 2 = KIP
- 3 = Debug
- 4 = Surflink

I believe the SAM just forwards any messages with TID != 1 to different serial connections. (But I have not yet traced the entire message path.)

For TID == 1, only the following TCs are handled by the SAM on the SP7:
- 01 = SAM: SAM
- 02 = BAT: Battery
- 03 = TMP: Thermal
- 04 = PMC: Power
- 05 = FAN: Fan
- 07 = DBG: Debug
- 09 = FWU: Firmware update
- 0c = TCL: (Trace/crash logs?)
- 0d = SFL: Surflink
- 10 = BLD: Surface Blades
- 12 = SEN: Sensors
- 13 = SRQ: (?)
- 15 = HID: HID
- 17 = BKL: Backlight
- 1b = USC: USB-C

The firmware does not explicitly name the TCs in any way (not even using the abbreviated names), so the above names are based on names of related tasks, message queues, etc. Of course, most of these names were already known.

FWU/TCL/SRQ commands are handled together, so their functionality is presumably related in some way.

## Debug mode

The SAM has a debug mode variable (default 0) and a "safe mode" flag (default 1).

The safe mode flag is read with TC 1, CID 0x27, and written by TC 7, CID 0x5F.
When the safe mode flag is true, all CIDs >= 0x80 are disabled (for all TCs), and various other functionality is disabled.

The debug mode is read with TC 1, CID 0x29, and written by TC 7, CID 0x4E.
The debug mode values are:
- 0 = disabled
- 1 = basic without logging
- 2 = basic
- 3 = possibly related to firmware update?
- 4 = full
[... 373 lines cut ...]
00
01  (see existing doc)
02  (see existing doc)
03  (see existing doc)
04  (see existing doc)
```

### TC 17: Backlight
```
02
03
04,87
05,88
```

### TC 1B: USB-C
```
00
06,80
81
82
83
```

</details>

qzed, 2022-10-31:
Oh, very nice work!

Regarding debug logs: How are they sent to the host? Standard events? I think it would be nice if we could provide some way to enable them via sysfs and dump them to the kernel logs.

qzed, 2022-10-31:
Regarding terminology: I think SurfLink is the charging/dock connector. Blade (at least on the SPX) is the keyboard connector, which also uses a UART. I don't know the specifics of that on earlier generations like the SP7.

Also there's an older talk from Alex Ionescu about the SP4 firmware (including KAOS, IIRC). Might interest you: https://recon.cx/media-archive/2017/mtl/recon2017-mtl-04-alex-ionescu-Fun-with-Sam-Inside-the-Surface-Aggregator-Module.mp4

quo, 2022-11-01:
> Regarding debug logs: How are they sent to the host? Standard events?

Not events I think, just regular messages. If you do:
```
./ctrl.py request 7 1 0x4b 0 1 1 # set log target = host
./ctrl.py request 7 1 0x4e 0 1 2 # set debug mode = 2
```
Then you'll start seeing "dropping unexpected command message" errors in dmesg.
I haven't actually looked at the log data yet, but I'm pretty sure it's almost entirely numbers, with very few strings, so fairly opaque.

> Blade (at least on the SPX) is the keyboard connector, which also uses a UART.

"Blades" was MS's term for Surface accessories, but all the news articles about it are from 2013, so I figured it was dead technology. Makes sense that the pins are still there on the keyboard connector, but is the Blade stuff actually used for the keyboard? I thought the keyboard used the KIP or HID messages?

> Also there's an older talk from Alex Ionescu about the SP4 firmware (including KaOS, IIRC). Might interest you.

Definitely very interesting! Skimmed it quickly. His description of Kaos seems very similar to what I've seen, but most of the other stuff he mentions sounds different. He says he spent months reverse engineering almost everything, I wonder why he never published anything beyond this talk.

qzed, 2022-11-01:
> Then you'll start seeing "dropping unexpected command message" errors in dmesg.

Ah, I think that's what I meant with "events" (messages sent by the EC that are not a response to a direct previous command. Might have a look at that later today.

> I haven't actually looked at the log data yet, but I'm pretty sure it's almost entirely numbers, with very few strings, so fairly opaque.

Ah, I kind of feared that. But makes sense if they have some tracing infrastructure set up. Might be worth a try checking how that windows kernel driver trace stuff works and see if there are similarities.

> Makes sense that the pins are still there on the keyboard connector, but is the Blade stuff actually used for the keyboard? I thought the keyboard used the KIP or HID messages?

So I'm not sure how the SP4 to SP7+ handle the keyboard stuff since the kernel sees that as USB, but on the SPX, SP8, and SP9, the connector thing for the keyboard is some custom serial/UART thing, which I think is what they call the blade interface. That then gets handled/translated into HID messages by SAM, which it presents via the HID interface. I think the KIP subsystem (likely keyboard and integrated peripherals or something like that) is also involved (also that might be some separate processor due to extra firmware I think), so maybe that translates it instead of SAM.

So in essence: `Type cover <--(blade interface)--> SAM <--(HID interface)--> kernel`, at least for SPX and everything with the new typecover.

I'm wondering whether the blade thing is relevant for the gens before that at all... maybe SAM somehow has a USB interface on those gens and the blade interface is still used? Or maybe in some reduced capacity? Otherwise I'd have expected it only on the newer Pros and maybe on the Surface Books.

>  He says he spent months reverse engineering almost everything, I wonder why he never published anything beyond this talk.

Yeah, he has the presentation somewhere as PDF but that's unfortunately all I could find. Also a lot has probably changed since the SP4 days. Especially due to the new interface and added components.

qzed, 2022-11-01:
> > Then you'll start seeing "dropping unexpected command message" errors in dmesg.
> 
> Ah, I think that's what I meant with "events" (messages sent by the EC that are not a response to a direct previous command. Might have a look at that later today.

Yeah, normal events. I also had to enable them via
```
./events.py enable 0x01 0x01 0x0b 0x0c 0x07 0x00 0x01
```

qzed, 2022-11-01:
So CID=0x48 seems to send some null-terminated strings, but the majority of data seems to be in some binary format.

quo, 2022-11-01:
> I'm wondering whether the blade thing is relevant for the gens before that at all... maybe SAM somehow has a USB interface on those gens and the blade interface is still used? Or maybe in some reduced capacity? Otherwise I'd have expected it only on the newer Pros and maybe on the Surface Books.

Well, Alex also discusses the Blade stuff he found in the old firmware, which includes communication and authentication. And the SP7 firmware also has various Blade related tasks/etc.

There's some old info here which suggests the old type cover also uses a serial protocol: http://edwardsh.in/keyboard%20cover/2015/08/13/applying-logic-to-the-surface-touch-cover

So my guess would be that all the type covers use the "Blade" protocol, and they've just changed where/how the translation to HID takes place.

> Yeah, normal events. I also had to enable them

Ah, ok. From a quick look at your code I figured events had small request ids whereas the ids in the errors looked more random. I need to have a closer look at the event stuff.

> Also my guess is that there's some timestamp and other header stuff before each record.

Yeah, it should be similar/identical to the TCL data. I'll see if I can figure out the format.

qzed, 2022-11-01:
> So my guess would be that all the type covers use the "Blade" protocol, and they've just changed where/how the translation to HID takes place.

I think that makes sense.

> Ah, ok. From a quick look at your code I figured events had small request ids whereas the ids in the errors looked more random. I need to have a closer look at the event stuff.

On my SB2, the request ID seems to always be 0x0007 (normally request ID should match the target category). But other values (command ID and especially instance ID) seem all over the place. I assume instance ID is some subsystem ID.

Haven't checked on the SPX yet though, so maybe things are a bit different on newer devices.

qzed, 2022-11-01:
And you are absolutely correct with the request IDs on the newer devices... they're all over the place. With the new format it looks like
```
80 07 03 01
```
and I don't have to enable any events (in fact trying to do that will return some error code).

So, there's the `03` that is normally only set for host-to-EC messages.

quo, 2022-11-05:
I've added some info about the debug log data format. And I was wrong about it being related to TCL, it doesn't look like the log data ends up in the TCL buffers. There's a fault handler function that seems to fill TCL buffer 1. It's not yet clear to me how the other TCL buffers are filled.

> So, there's the 03 that is normally only set for host-to-EC messages

Yeah, AFAICT that just means the message was intended to go to TID 3 = Debug. When you override the log target, it doesn't bother to set the "correct" TID.

qzed, 2022-11-05:
> > So, there's the 03 that is normally only set for host-to-EC messages
> 
> Yeah, AFAICT that just means the message was intended to go to TID 3 = Debug. When you override the log target, it doesn't bother to set the "correct" TID.

If I haven't messed anything up it's actually a bit weirder than that: For "normal" messages you have e.g.:
- `[header] 80 TC 00 01 [...]`: SAM to host
- `[header] 80 TC 01 00 [...]`: host to SAM
- `[header] 80 TC 00 02 [...]`: KIP to host
- `[header] 80 TC 02 00 [...]`: host to KIP

But we have `[header] 80 07 03 01 [...]` which would be _SAM to host_ (so far pretty standard) but also _host to debug_? Point is, there's a byte set that I normally would have expected to be zero.

qzed, 2022-11-05:
I think it's possible this could also be interpreted as _debug to SAM_ and _SAM to host_, but I'm not sure if that would fit into the whole KIP perspective and I'd also have thought that debug messages originate from SAM itself.

Any chance you can find out more about the two target ID bytes on how they seem to be used?

quo, 2022-11-05:
The bytes are just target ID and source ID.
So `03 01` means from SAM to Debug (except the override causes the message to be sent to the Host instead).
And like I said, I think the SAM just forwards everything with TID!=1.
So if you send something with `02 00`, the SAM forwards it to the KIP. The KIP then sends a reply to the SAM with `00 02`, and since the TID of that reply is 0, the SAM forwards it to the Host.

qzed, 2022-11-05:
Ah, got it. That actually makes much more sense, thanks. I'll update the docs accordingly.

qzed, 2022-11-05:
Alright, I've improved the handling for unknown/unsupported TIDs a bit: https://github.com/linux-surface/kernel/compare/32815a5c269b33541e58315fbc9cc866a4cb309f...351805f561beb9ff9c95b6c53338210632703ec6. Mostly just https://github.com/linux-surface/kernel/commit/f1b2c939ca190ac678ea03e9fc490323f7a84f3f, which means that instead of trying to match up the request ID to something that doesn't exist it ignores and drops anything that isn't addressed to the host directly.

I guess if we want to properly handle the debug messages, we'll have to handle them separately from regular messages anyways, meaning we'll need a `command->tid == SSAM_SSH_TID_DEBUG` there.

quo, 2022-11-13:
Looks good!

Meanwhile I've managed to flash modified firmware to the SAM. Interestingly there seems to be some code in the firmware update logic to do something with hashes and (maybe) signatures, but it's not actually used for the firmware images MS provides. So all you need to do is extract the firmware image, modify it, update the CRC16 at the end of the file, and then upload it using TC 9 CID 3 and CID 4.

So now I've added a command to the firmware to write arbitrary memory. And it works. :)

I've published some scripts here: https://github.com/quo/sam-fw-tools

qzed, 2022-11-13:
Nice work!

I'm kind of surprised that it's not signed. Is there any other protection against that or could just any random user with admin permissions on Windows upload some firmware?

quo, 2022-11-13:
My guess is that anyone who can communicate with the SAM can flash new firmware. You might even be able to do it via the Surflink connection (ie. without being admin, and even when the device is turned off).

There are some conditions that seem to relate to whether firmware updates are allowed that I haven't really figured out yet, but that's bypassed when the safe mode is disabled.

qzed, 2022-11-13:
> My guess is that anyone who can communicate with the SAM can flash new firmware. You might even be able to do it via the Surflink connection (ie. without being admin, and even when the device is turned off).

If you manage to do that without any sort of authentication (I'd hope there is some), I think you should ask MS for a sizeable bug bounty xD

I honestly have no idea how locked down the driver interface is on Windows, so chances are that that's limited to some kernel stuff and user-space might be blocked somehow... might be worth checking if the Windows API allows arbitrary commands. Some SAM stuff should definitely be available like DTX (on the SB2 and SB3) or I think the TCL stuff. But if an attacker has admin rights, you've probably lost anyways. Anyways... awesome work!

quo, 2022-11-13:
I actually wonder if the Surflink UART connection from the SAM is directly exposed on the Surflink connector, or if there is another controller in between. Might be fun to try to probe the connector, send some commands to TID 4, and see if anything shows up.

I'm not really sure what the worst thing is that you could actually do with modified SAM firmware though, in terms of security. Keylogging and then somehow exfiltrating via synthesized HID events maybe?

qzed, 2022-11-13:
Yeah, HID interface is probably the most dangerous thing. Keylogging, full keyboard and touchpad control... I'd guess you could find ways to use that as a sort of basic rootkit to pull more advanced stuff into the OS (like having it write commands into a terminal or something, use that to disable some protection stuff and download OS-level malware). Doesn't really have to be that dangerous by itself, the problem is you can't really detect it from the OS until it starts to act.

qzed, 2022-11-13:
> I actually wonder if the Surflink UART connection from the SAM is directly exposed on the Surflink connector, or if there is another controller in between. Might be fun to try to probe the connector, send some commands to TID 4, and see if anything shows up.

I'm not sure if the information that I have is correct or how reliable it is, but there are a SAM_DEBUG_RX and a SAM_DEBUG_TX line connected directly from SAM to SurfLink. Now I'm not sure if that is only for pre-production / debug models or if that is also on the final ones. In fact, there even seems to be a debug mux that can multiplex those lines to lines of one of the USB-C ports on the SPX (again, no idea if that's still present on the final models). The USB-C mux apparently also allows access to the blade UART and the SAM-to-SoC/Host TX/RX lines.

qzed, 2023-01-05:
Finally got around to unpack the SPX firmware and load it into Ghidra. Had to specify 0x67c as offset (not 0x66c), but with that, everything seems to work.

quo, 2023-01-06:
Nice! The offset is 0x67c because the unpack script prepends the 0x10 byte setup data, since this is needed by the upload script. So the file consists of the 0x10 byte setup header + 0x66c bytes of headers (see parse_fw()) + the actual ARM binary.

A couple pointers to get you started (assuming the firmware is very similar to the SP7 firmware):

- If you follow the SVCall vector, you'll get to a function that contains a switch (or if/else chain) that dispatches various Kaos system calls based on syscall number. These numbers are:
  ```
  0x11 Sleep
  0x12 TaskDestroy
  0x13 TaskSetPriority?
  0x20 MessageQueueCreate
  0x21 MessageQueueSend
  0x22 MessageQueueRead
  0x23 MessageQueueDestroy
  0x24 MessageQueuePeek
  0x30 SemaphoreCreate
  0x31 SemaphoreClaim
  0x32 SemaphoreRelease
  0x33 SemaphoreReleaseAll?
  0x34 SemaphoreDestroy
  0x40 EventCreate
  0x41 EventWait
  0x42 EventSignal
  0x43 EventDestroy
  ```

  Look for `svc` instructions to find the wrapper functions which are used to invoke these syscalls from "user" mode. The creation functions are especially helpful, because the first arg is a string naming the thing being created.

- I've also found it to be extremely helpful to create a RAM dump using the 0x4d debug command, and loading that into Ghidra as well. Trying to follow some of the indirection and figuring out the data structures using just the ROM data is almost impossible.

- To get to the SAM command handling, look for the function that calls `MessageQueueCreate("SamHostTx",...)` and `MessageQueueCreate("SamHostRx",...)`. At the end of this function, a struct is filled and passed to another function.
  The layout of the struct is:
  ```c
  struct cmd_queue {
  	struct cmd_queue *next;
  	u32 *msg_queue_handle_ptr;
  	u16 msg_queue_block_size;
  	u16 _padding0;
  	u32 *event_handle_ptr;
  	u32 event_flags;
  	u8 target_category;
  	u8 _padding1[3];
  }
  ```

  The function that is called registers the TC command queue by adding it to the linked list of all queues. You can search for invocations of this function, or inspect the linked list in RAM to find the message queue for each TC. And if you look for other references to the linked list, you'll find the SAM message dispatching function. This function checks if the target ID == 1, then loops over the list to dispatch the message to the correct message queue based on the TC (or if the target ID != 1, it forwards the message to a different target).

  Then, you can look for calls to MessageQueueRead with the messages queues from the list. These will usually be found in loops, inside top level task functions. The usage is a little ad-hoc, so each call site is structured differently, but usually the read is followed by a call to a function which parses the SAM message buffer into a struct, which is then passed to a TC specific function containing a big switch on the command ID.

qzed, 2023-01-06:
Thanks! That is quite helpful!