linux-surface/surface-aggregator-module #65: `container_of_const` undefined

L1ghtmann, 2023-06-10:
First, I just want to say a huge thank you for your work on linux-surface and this module. 

I have been using this in a project of mine for a few months now and it seems a recent commit (a3c6f36 specifically) broke building on Debian and Ubuntu:
```
In file included from /sam/module/src/clients/surface_aggregator_hub.c:18:
/sam/module/src/clients/../../include/linux/surface_aggregator/device.h: In function ‘ssam_device_get’:
/sam/module/src/clients/../../include/linux/surface_aggregator/device.h:221:33: error: implicit declaration of function ‘container_of_const’; did you mean ‘container_of_safe’? [-Werror=implicit-function-declaration]
  221 | #define to_ssam_device(d)       container_of_const(d, struct ssam_device, dev)
      |                                 ^~~~~~~~~~~~~~~~~~
/sam/module/src/clients/../../include/linux/surface_aggregator/device.h:293:23: note: in expansion of macro ‘to_ssam_device’
  293 |         return sdev ? to_ssam_device(get_device(&sdev->dev)) : NULL;
      |                       ^~~~~~~~~~~~~~
/sam/module/src/clients/../../include/linux/surface_aggregator/device.h:221:55: error: expected expression before ‘struct’
  221 | #define to_ssam_device(d)       container_of_const(d, struct ssam_device, dev)
      |                                                       ^~~~~~
/sam/module/src/clients/../../include/linux/surface_aggregator/device.h:293:23: note: in expansion of macro ‘to_ssam_device’
  293 |         return sdev ? to_ssam_device(get_device(&sdev->dev)) : NULL;
      |                       ^~~~~~~~~~~~~~
/sam/module/src/clients/surface_aggregator_hub.c: In function ‘ssam_hub_mark_hot_removed’:
/sam/module/src/clients/../../include/linux/surface_aggregator/device.h:221:55: error: expected expression before ‘struct’
  221 | #define to_ssam_device(d)       container_of_const(d, struct ssam_device, dev)
      |                                                       ^~~~~~
/sam/module/src/clients/surface_aggregator_hub.c:119:36: note: in expansion of macro ‘to_ssam_device’
  119 |         struct ssam_device *sdev = to_ssam_device(dev);
      |                                    ^~~~~~~~~~~~~~
```

I'm not too familiar with kernel modules, but I did a bit of searching and figured `device.h` may have been missing a `container_of.h` include, but it doesn't seem like said header actually provides a definition of `container_of_const`:
```
$ cat usr/src/linux-headers-6.1.0-9-common/include/linux/container_of.h 
/* SPDX-License-Identifier: GPL-2.0 */
#ifndef _LINUX_CONTAINER_OF_H
#define _LINUX_CONTAINER_OF_H

#include <linux/build_bug.h>
#include <linux/err.h>

#define typeof_member(T, m)     typeof(((T*)0)->m)

/**
 * container_of - cast a member of a structure out to the containing structure
 * @ptr:        the pointer to the member.
 * @type:       the type of the container struct this is embedded in.
 * @member:     the name of the member within the struct.
 *
 */
#define container_of(ptr, type, member) ({                              \
        void *__mptr = (void *)(ptr);                                   \
        static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
                      __same_type(*(ptr), void),                        \
                      "pointer type mismatch in container_of()");       \
        ((type *)(__mptr - offsetof(type, member))); })

/**
 * container_of_safe - cast a member of a structure out to the containing structure
 * @ptr:        the pointer to the member.
 * @type:       the type of the container struct this is embedded in.
 * @member:     the name of the member within the struct.
 *
 * If IS_ERR_OR_NULL(ptr), ptr is returned unchanged.
 */
#define container_of_safe(ptr, type, member) ({                         \
        void *__mptr = (void *)(ptr);                                   \
        static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
                      __same_type(*(ptr), void),                        \
                      "pointer type mismatch in container_of_safe()");  \
        IS_ERR_OR_NULL(__mptr) ? ERR_CAST(__mptr) :                     \
                ((type *)(__mptr - offsetof(type, member))); })

#endif  /* _LINUX_CONTAINER_OF_H */
```

Do you happen to know of a dev package that provides this definition? I apologize in advance if I am missing something obvious. 

Thank you again!

qzed, 2023-06-10:
That macro has been introduced somewhat recently ( either 6.2 or 6.3), see
***@***.***/.
If you don't want to compile a custom kernel, you can probably get away
with patching it in before those other macros.

qzed, 2023-06-10:
Ah crap. Looks like the mail response x'ed the link. https://lore.kernel.org/lkml/CAJZ5v0gJuE0SOqErrZPSHKjgNuVqk0gtaQpPztGBqP1zqn9AEQ@mail.gmail.com/

L1ghtmann, 2023-06-10:
Managed a successful build with two patches (due to additional changes made in 30790b0):
1) `linux/container_of.` -- 0001-container_of-add-container_of_const-that-preserves-const-ness-of-the-pointer.patch
- https://lore.kernel.org/lkml/20221205121206.166576-1-gregkh@linuxfoundation.org
2) `linux/device.h` -- 0005-misc-driver-core-make-struct-device_type-uevent-take-a-const.patch
- https://lore.kernel.org/lkml/20230111113018.459199-6-gregkh@linuxfoundation.org

Will give the resulting build a go and edit my message with the result. Thanks again!

closed 2023-06-10 by L1ghtmann